esp-rs/esp-idf-svc #221: Regarding File System and WebSockets

akashlama1998-icloud, 2023-02-01:
Are there any examples for working with file system? Is there even anything available like SPIFFS or LittleFS or anything? And http server works great had no issues yet.. but are WebSockets supported right now or any examples on how I could implement it?

ivmarkov, 2023-02-01:
All file systems that work with ESP IDF (spiffs and fatfs) do work with this crate. Please use the forum to ask for examples. I'll close this now, but if you have other questions/concerns - reopen.

ivmarkov, 2023-02-01:
Websockets are supported, if you mean server-side (HTTP) support. Look at [this](https://github.com/ivmarkov/ruwm/blob/master/ruwm-esp32/src/services.rs#L349).

closed 2023-02-01 by ivmarkov