esp-rs/esp-idf-svc #223: Run ETH w5500 + another device same SPI

reinismu, 2023-02-06:
I will need to run Eth w5500 and SD card on the same SPI lane. Seems like current logic of ETH will take control of whole SPI line. I wonder if lib API must change or it is hardware limitation?

ivmarkov, 2023-03-26:
@reinismu I've now committed your PR, and then on top my changes that allow passing the SPI driver either by reference, or by moving it.

Would *really* appreciate if you can check on your end if it still works after my changes. I - ahem - just disconnected my w5500 from the esp32 board it was attached to last moth. :(

reinismu, 2023-03-26:
Will try to set it up tomorrow :)

ivmarkov, 2023-03-27:
Well... you can always
```rust
pub type SpiEspEth<'d> = EspEth<'d, SpiEth<&'d SpiDriver<'d>>>
```

No?

ivmarkov, 2023-03-27:
We can also define, in the `eth.rs` module:
```rust
pub type EspSpiEth<'d, T> = EspEth<'d, SpiEth<T>>;
pub type EspRmiiEth<'d> = EspEth<'d, RmiiEtnh>;
pub type EspOpenEth<'d> = EspEth<'d, OpenEth>;
```