esp-rs/esp-idf-svc #224: provide a `Send + 'static` HTTP request type

hawkw, 2023-02-11:
## Motivation

I'd like to be able to perform async operations in response to an HTTP request, and send the response when those asynchronous operations complete. Since the `handler`s passed to `EspHttpServer` are synchronous, running async code in response to HTTP requests currently necessitates having the handler function send the request to a background task over a channel or something. 

Unfortunately, this is not currently possible, as the `EspHttpRequest` and `EspHttpConnection` types are not `Send`, and have a non-`'static` lifetime. Since the request type is not `Send`, a channel of requests is also not `Send`, and therefore any handler or `handler_fn` which holds the sender side of a channel of requests is also not `Send`. However, the `Handler` trait requires `Send`.

## Proposed Feature

It would be really nice if there was a type which represents a HTTP request and the capacity to respond to it, but which is `Send` and is valid for the `'static` lifetime? I notice that the `EspHttpRequest` type contains a borrowed reference to an ESP-IDF `httpd_req_t`, and a map of headers inside an `UnsafeCell`. Is it possible to introduce an additional request type where the `httpd_req_t` is allocated on the heap, and the header map is stored in a `Mutex` or something? Since this would, of course, introduce additional overhead, I'd expect there to be an explicit conversion into this type which is only performed when the user requires it. 

I'm not particularly familiar with the ESP-IDF C API, so I'm not sure whether it's possible to heap-allocate the `httpd_req_t`, or whether this would violate ESP-IDF's invariants, so perhaps the proposed solution is not possible. Perhaps another solution would be to add some kind of higher-level API for request callbacks that can be completed in a background task or on another thread, where the `httpd_request_t` itself remains in the handler? 

I'm certainly willing to work on a PR implementing some kind of solution for this issue. However, I'd need guidance from maintainers to understand what the correct approach is, as I'm not particularly familiar with ESP-IDF's C API, or with the internal implementation details of this crate.. Thanks in advance!

ivmarkov, 2023-02-12:
@hawkw the HTTP server in ESP IDF is an interesting thing. :) It is single threaded, yet blocking in its request-response metaphor.

In terms of its capabilities I believe we are reaching its limits.

Rather than trying to tweak it any further, how about looking at my `edge-net` project? Amongst other things, it contains a `no-std`, no-alloc pure async HTTP server based on the `httparse` crate. Sure, the server is not tested yet, and surely there are bugs in there, especially in the chunked decoder, but I think this is a much more promising avenue of exploration. 

I would've fixed all the issues in it myself, but I simply do not have the physical time to work on it ATM. But I would gladly accept PRs!

ivmarkov, 2023-02-12:
One more thing: having `Send` + `'static` on the `HttpConnection` trait impl is not strictly necessary for an async HTTP server, as you only need that if you want to spawn each HTTP handler, which - as you'll see in the `edge-net` code - is not strictly necessary. At the price of a slight performance degradation (probably negligible on embedded) you can do looping select instead.

hawkw, 2023-02-18:
> Rather than trying to tweak it any further, how about looking at my `edge-net` project? Amongst other things, it contains a `no-std`, no-alloc pure async HTTP server based on the `httparse` crate. [...]

Thanks for the suggestion! I was considering trying to just run a `hyper` server on the board, but I wanted to see how far I could get with the ESP-IDF HTTP server. I'll check out `edge-net` as well --- might be a better option than `hyper` for embedded use-cases! And I'd be happy to contribute back upstream.

ivmarkov, 2023-02-19:
Based on my very superficial look at `hyper`, it is a STD-only solution which does do and need heap allocations. In contrast, `edge-net` is `no_std` and does not need a heap allocator at all, hence the ambition of it is to also be usable on other bare-metal platforms, that do not have STD impl like the ESP32 and that do avoid heap allocations by all means.

`edge-net` will definitely not scale as much as `hyper`, but for embedded environments that's a non-goal anyway, as these can open only a few sockets (as in 10 to 16 max, on the ESP IDF).

The common thing between `edge-net` and `hyper` is the utilization of the `no_std` `httparse` crate which does the heavy lifting of parsing the HTTP headers.

Other than that, I doubt you'll be able to run `hyper` on the ESP IDF anytime soon, as it seems to have a hard requirement on `tokio`. `tokio` does not run on the  ESP IDF (only the `smol` ecosystem does, with my custom patches, and `async-std` might be easy to do as well, due to their common heritage which is the `polling` library). That is, [unless this PR gets merged at some point in time](https://github.com/tokio-rs/mio/pull/1602).

closed 2023-02-19 by ivmarkov

hawkw, 2023-02-19:
Just for the record, `hyper` does not actually have a mandatory dependency on `tokio`; it has built in support for tokio but can be used with other async executors. You're correct that it does require `std` and performs dynamic heap allocations, though.

ivmarkov, 2023-02-19:
To me, [this](https://github.com/hyperium/hyper/blob/master/Cargo.toml#L35) looks like a non-optional dependency.

ivmarkov, 2023-02-19:
And to elaborate further, the problem is not the "executor", the problem is the Reactor. Can `hyper` be used with other reactors besides `tokio`'s `mio` based one? I don't think so.