esp-rs/esp-idf-svc #229: Netif PPP support not being built

codertao, 2023-02-15:
Not sure if this should be here or esp-idf-sys. I'm using `esp-idf-sys=0.32`, `esp-idf-svc=0.45`, `esp-idf release/v4.4`

I'm trying to experiment with the Netif PPP stack. 
In code it's gated behind `#[cfg(esp_idf_ppp_support)]`;
I've modified the sdkconfig I'm using to include PPP_SUPPORT:
```
CONFIG_LWIP_PPP_SUPPORT=y
CONFIG_LWIP_PPP_ENABLE_IPV6=y
CONFIG_LWIP_PPP_DEBUG_ON=y
CONFIG_PPP_SUPPORT=y
CONFIG_PPP_DEBUG_ON=y
```

However the the `cfg(esp_idf_ppp_support)` isn't being hit. The `cargo build -v` shows the rustc invocation includes `--cfg esp_idf_lwip_ppp_enable_ipv6  --cfg esp_idf_lwip_ppp_debug_on  --cfg esp_idf_lwip_ppp_support `- it's only picking up the LWIP configs.

```
[dependencies]
esp-idf-hal = { version = "0" }
esp-idf-sys = { version = "0.32", features = ["binstart"] }
esp-idf-svc = { version = "0.45", features = ["experimental"] }
embedded-svc = { version = "0.24", features = ["experimental"] }
...

[build-dependencies]
embuild = "0.31"
anyhow = "1"
```

codertao, 2023-02-15:
Related, after switching code to use lwip_ppp_support, I ran into
```
    fn default_mac_raw_type(&self) -> Option<esp_mac_type_t> {
        match self {
            Self::Sta => Some(esp_mac_type_t_ESP_MAC_WIFI_STA),
            Self::Ap => Some(esp_mac_type_t_ESP_MAC_WIFI_SOFTAP),
            Self::Eth => Some(esp_mac_type_t_ESP_MAC_ETH),
            #[cfg(esp_idf_slip_support)]
            #[cfg(esp_idf_lwip_ppp_support)]
            _ => None,
        }
    }
```
Requires both SLIP and PPP support for the fall through.

ivmarkov, 2023-05-11:
Better late than never, I guess. Should be fixed with https://github.com/esp-rs/esp-idf-svc/commit/22a8be64292d55e79a7ca468f23c756c953d82fe

closed 2023-05-11 by ivmarkov