esp-rs/esp-idf-svc #232: esp_websocket_event_data_t corrupted when receiving if EspWebSocketClient initialized with new_with_conn

evoggy, 2023-02-19:
Hi!

I've been using the websockets implementation for sending and receiving binary data on an ESP32 (ESP32-WROOM-32E). Sending and receiving seems to be working fine if the ```EspWebSocketClient::new``` is used for initialization, but if ```EspWebSocketClient::new_with_conn``` is used there seems to be an issue with the received data being corrupted.

From what I've observed the data passed from the SDK looks fine on [client.rs#L356](https://github.com/esp-rs/esp-idf-svc/blob/master/src/ws/client.rs#L356). But when the data in message (```esp_websocket_event_data_t```) is accessed in ```EspWebSocketConnection::next``` it seems corrupted. Some things, like ```op_code``` and ```client``` seems fine, but other like ```data_ptr``` and ```data_len``` are corrupted.

I'm new to rust, so I might be mistaken, but I can't find anything wrong in the implementation. Could it be that the underlying data in ```esp_websocket_event_data_t``` is modified in the IDF after the callback returns ?

Vollbrecht, 2023-03-29:
Did you get closer to the problem, or did you move on? Do you have an simplified project setup to help reproduce the problem? Anyway thank you for reporting.

evoggy, 2023-10-23:
No, I never got this working and instead just used ```EspWebSocketClient::new```.