esp-rs/esp-idf-svc #233: How to write to NVS?

tadejgasparovic, 2023-02-24:
This may end up being inexperience with Rust on my end, but I'm struggling to find a way to write to NVS. I see there's a [`set_raw()`](https://esp-rs.github.io/esp-idf-svc/esp_idf_svc/nvs/struct.EspNvs.html#method.set_raw) function, but unlike `get_raw()`, it is private.

Here's what I have so far:
```rust
let partition = EspDefaultNvsPartition::take()?;
let mut nvs = EspDefaultNvs::new(partition, "ns", true)?;
```

I can call `get_raw`,  `contains` , or `remove`:
```rust
let mut buf: [u8; 256] = [0; 256];
nvs.get_raw("test", &mut buf)?;
nvs.contains("test")?;
nvs.remove("test")?;
```

But I wasn't able to find a way of calling `set_raw` without doing some unsafe memory manipulation

ivmarkov, 2023-02-24:
That's a bug indeed, so I'll keep that open. A workaround is to use the `set_raw` fn from the `embedded-svc` `RawStorage` trait that is implemented on the same struct.

tadejgasparovic, 2023-02-24:
Thanks for the workaround. Works as expected.
```
RawStorage::set_raw(&mut nvs, "test", &buf)?;
```

closed 2023-02-27 by ivmarkov