esp-rs/esp-idf-svc #235: EspHttpConnection::read can never return an error (incomplete response)

schurwanzn-stabl, 2023-03-08:
## Problem
`EspHttpConnection::read` returns a Result, suggesting (since it is not documented) that it would return an error when reading the response body fails.

However the ESP-IDF function it calls in it's implementation, `esp_http_client_read_response`, is not documented to return any errors and is implemented to silently ignore any errors that might occur.

## Impact
We have implemented a simple HTTP GET request by calling `EspHttpConnection::new`, then `initiate_request`, followed by `initiate_response`. After which we just `.read()` from EspHttpConnection until it returns Ok(0), imitating the http_request example.

Testing this approach under adverse network conditions [while reading the data in small chunks (4KiB) and simulatenously writing the read data into flash (EspOtaUpdate)] has ended with the code regarding EspHttpConnection terminating it's read normally, even though the read data was incomplete.

ivmarkov, 2023-03-08:
Unfortunately, without providing a reproducible code snippet, I can only guess what is the problem, and whether the problem is in the Rust HTTP client wrapper, or in the ESP IDF HTTP client itself, or in your code, or in some in-between proxy, as another poster suggested.

I get it that you don't have time (as per the forum message) to implement a reproducible code snippet, but I wonder how you expect to get something actionable from the above problem description?

schurwanzn-stabl, 2023-03-09:
I can also provide a simple code example to demonstrate how this discrepancy impacts real code:
```rust
// [... initialize logging, connect to network etc. ...]

  let mut http_client = EspHttpConnection::new(&Configuration {
      crt_bundle_attach: Some(esp_idf_sys::esp_crt_bundle_attach),
      ..Default::default()
  })?;

  log::info!("Doing HTTP request...");
  http_client.initiate_request(Method::Get, "https://httpbin.org/drip?duration=30&numbytes=40&code=200&delay=2", &[])?;
  http_client.initiate_response()?;

  if http_client.status() != 200 {
      return Err(anyhow::anyhow!(
          "request to {} returned HTTP status code {}",
          OTA_API_DOWNLOAD_URL,
          http_client.status()
      ));
  }

  log::info!("Content length: {:?}", http_client.content_len());
  log::info!("Reading response body...");

  let mut total_bytes_read = 0;
  let mut buffer = [0u8; 4];

  loop {
      let bytes_read = match http_client.read(&mut buffer) {
          Ok(bytes_read) => bytes_read,
          Err(err) => {
              log::error!("http_client.read() failed: {}", err);
              return Err(err.into())
          },
      };

      if bytes_read == 0 {
          // read is complete
          break;
      }
      
      log::info!("Read {} bytes", bytes_read);
      total_bytes_read += bytes_read;
  }
  
  log::info!("Download completed successfully. Received {} bytes", total_bytes_read);
```

When executed this code downloads 40 bytes of random data over a period of about 30 seconds. When you interrupt the connection of the ESP to the internet (e.g. by closing the wifi AP) while it is downloading, the code will still complete "successfully" even though it did not read all 40 bytes and ESP-IDF logged errors about the connection failure.

ivmarkov, 2023-03-09:
Got it. Basically, you are saying that in its current shape, the `esp_http_client_read_response` is useless. Perhaps then you can open a bug at the ESP IDF repo?

What we can do as a workaround:
* Switch to using `esp_http_client_read` instead. The "fill-in-the-buffer-fully" functionality is anyway we do not need, as long as the buffer contains at least a single byte which was read
*  The docu of `esp_http_client_read` is also confusing. First, the func docu says it will return "-1" on any error (which is quite terrible, as you can't do error diagnostics, but oh well) and then at the end it mentions it might return EAGAIN. So it seems we have to cycle in a loop on EAGAIN. Fine

I can do the above changes, but given that (a) you are experiencing the issue (b) they are relatively trivial, you can perhaps devise a PR, test it at your side, and submit it if it fixes the issue?

schurwanzn-stabl, 2023-03-10:
> Basically, you are saying that in its current shape, the `esp_http_client_read_response` is useless.

At least if you want more than some quick and dirty code that just assumes everything will always work fine. `esp_http_client_read_response` looks like some quick and dirty convenience function. I won't open an issue for it since I don't think it would bear any fruits. It would be a breaking API change for something that can be easily worked around by just using a different function. The best I would hope for would be a documentation improvement.

> The docu of esp_http_client_read is also confusing. First, the func docu says it will return "-1" on any error (which is quite terrible, as you can't do error diagnostics, but oh well) and then at the end it mentions it might return -EAGAIN. So it seems we have to cycle in a loop on -EAGAIN. Fine

It seems like the special case for EAGAIN handling has been added only in ESP-IDF 5. In v4.4.4 it will completely ignore EAGAIN errors and just return early with however many bytes were read (which might be 0 again!).

Thus I have implemented an additional workaround for ESP-IDF 4 by checking esp_http_client_is_complete_data_received if 0 bytes were read to determine whether the body is actually EOF or it is just EAGAIN.

I decided to forward EAGAIN to the user instead of looping until data is available to (a) imitate to the behaviour of std::io::Read [which explicitely states that it might return a EAGAIN equivalent error if data will be available later] and (b) match the behaviour the code will exert in ESP-IDF 5.

See PR #237.

---

> which is quite terrible, as you can't do error diagnostics, but oh well

At times I feel like the authors of some of the code in ESP-IDF don't believe that errors actually occur.

closed 2023-03-26 by ivmarkov