esp-rs/esp-idf-svc #242: EspHttpServer panic on handler error

dragazo, 2023-03-24:
I have an http(s) server running on an esp32-s3 board, and make frequent requests to it (~2Hz) for status info. Every once it a while (around every 4 minutes), I get a panic that resets the board. From further investigation, it appears to come from within the error handler for a request endpoint. Specifically, if a handler returns `Err(_)`, the `EspHttpConnection::render_error` function is called, which renders a 500 error and message body. However, at the time when my error is returned, the connection is not in the request state, so the `self.assert_request()` triggers an immediate panic and board reset.

As far as I can tell, the only safe way to avoid this is to never return `Err(_)` from a handler and always do error handling manually. This is especially the case since there is no mechanism to add custom headers to any auto-generated responses.

ivmarkov, 2023-03-26:
The functions panic and (IMO) should panic, as these panics indicate a bug, rater than a normal behavior.
Specifically to your problem, `EspHttpConnection::render_error` should not try to render a 500 error and a message body if the user has already driven the connection into a response state, which is a bug.

ivmarkov, 2023-03-26:
Hopefully, [this](https://github.com/esp-rs/esp-idf-svc/commit/9741d9a9270e69b9449d53b98ee05ab698d996ba) should take care of it. Please close this issue if it fixes the problem for you.

closed 2023-04-03 by dragazo