esp-rs/esp-idf-svc #248: EspHttpConnection body received in fetch header state

wouterdebie, 2023-04-09:
I'm pretty much using the http client in the example, however, I need to post some data. When doing this, I get a debug message:

```
I (891184) esp-x509-crt-bundle: Certificate validated
I (892194) HTTP_CLIENT: Body received in fetch header state, 0x3fcb969b, 16
0x3fcb969b - _heap_start
    at ??:??
```

The request seems to work, but I'm not sure everything is completely correct.

My code:

```rust
let mut client = embedded_svc::http::client::Client::wrap(
    esp_idf_svc::http::client::EspHttpConnection::new(
        &esp_idf_svc::http::client::Configuration {
            crt_bundle_attach: Some(esp_idf_sys::esp_crt_bundle_attach),

            ..Default::default()
        },
    )?,
);

let headers = [
    ("Content-Type", "application/json"),
    ("Content-Length", &format!("{}", data.len())),
];

let mut request = client.post(&uri, &headers)?;
request.connection().write(data)?;
let mut response = request.submit()?;
```

Any thoughts?

Vollbrecht, 2023-04-09:
I am also not 100% sure why it happen, glancing over the code i thought that it might be something with our simplified state handling compared to the underlying http-client code. They split header and body inspection separately? Every time the response  is not only composed of an header - have an body that is not 0 - this gets triggered. You can test this when sending an Head request instead. Then you dont get this info. Though as you say it seems to work fine as is.

Vollbrecht, 2024-06-21:
could be potentially fixed by #404 . Closing this and if anything related creeps up feel free reopen it.