esp-rs/esp-idf-svc #252: Current nightly rust compiler breakes build

Vollbrecht, 2023-04-14:
If you encounter the problem - mostly for esp32c* users - please use no nightly compiler newer than 10-04-23. If you already upgraded you can set your compiler version like this :
```
$ rustup install nightly-2023-04-10
$ rustup component add rust-src --toolchain nightly-2023-04-10-x86_64-unknown-linux-gnu #last part depending on platform 
```
```
$ rustup default nightly-2023-04-10
``` 
the default compiler option is by default overridden in your local rust-toolchain.toml file. You might need to change it to something like this
```
[toolchain]
channel = "nightly-2023-04-10"
```

ivmarkov, 2023-04-15:
Fixed in `master`, but leaving it pinned, as the workaround is still necessary for the published crate.

closed 2023-04-15 by ivmarkov

Vollbrecht, 2023-04-27:
if you are using the current master branch, and using the nightly feature flag, please also use the master branch of `embedded-svc` by patching it in your Cargo.toml

```toml
[patch.crates-io]
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc" }
embedded-svc = { git = "https://github.com/esp-rs/embedded-svc" }
```