esp-rs/esp-idf-svc #253: EspMdns::add_service fails if the hostname is not set

LowderPlay, 2023-04-17:
When trying to create a new mDNS service using following code, add_service method fails with EspError(0x102) - ESP_ERR_INVALID_ARG

```rust
let mut mdns = EspMdns::take().unwrap();
mdns.add_service(Some("test"), "_http", "_tcp", 80, &[("hello", "world")]).unwrap();
```

However, if you set the hostname before adding a service, the code proceeds without a panic, yielding the wanted result:

```rust
let mut mdns = EspMdns::take().unwrap();
mdns.set_hostname("hostname").unwrap();
mdns.add_service(Some("test"), "_http", "_tcp", 80, &[("hello", "world")]).unwrap();
```

Could you add a check for that or at least put that into a documentation?
Thanks

Vollbrecht, 2024-06-21:
@ivmarkov that probably still an issue?

ivmarkov, 2024-06-21:
> @ivmarkov that probably still an issue?

Yes. Though I'm not really sure what we are supposed to do here? It is just how it works - and - it fails at runtime anyway, and does not panic.

The poster is wrong that it panics - it does so only only because he `unwrap`s.

I would close this.

ivmarkov, 2024-06-21:
> When trying to create a new mDNS service using following code, add_service method fails with EspError(0x102) - ESP_ERR_INVALID_ARG
>
> ```rust
> let mut mdns = EspMdns::take().unwrap(); [...]

There is already a check. Why should we add a check on top of the check coming from ESP IDF and resulting in an `EspError` anyway?

ivmarkov, 2024-06-21:
If you feel like, you can add a short docu string that the host name should be set first, though that would be a dip in the sea of undocumented APIs we are currently having
 :)

closed 2024-06-21 by ivmarkov