esp-rs/esp-idf-svc #254: Rework README.md to contain better information for new users / simple how to get started

Vollbrecht, 2023-04-19:


ivmarkov, 2023-04-20:
Easiest is to just put there references to the `esp-idf-template` project, which already does have a how-to-get-started info, if by that you mean how to assemble a minimal project that depends on `esp-idf-svc`. We should probably put the same link in `esp-idf-hal` and `esp-idf-sys` readmes.

ivmarkov, 2023-04-20:
We can also put mini.al info how to build the crate with examples (at least one more is coming next to the existing one). Ditto for `esp-idf-hal`.

Vollbrecht, 2023-04-20:
yeah, i was thinking something along this line - maybe also add a link to the wokwi hello-world for a zero setup example. i will come up with something and make a PR

ivmarkov, 2023-06-27:
Hopefully addressed now.

closed 2023-06-27 by ivmarkov