esp-rs/esp-idf-svc #257: Unable to compile esp-idf-svc-0.45.0 for esp-idf v5.0

katyo, 2023-04-22:
```
error[E0412]: cannot find type `eth_esp32_emac_config_t` in this scope
     --> /home/kayo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/eth.rs:750:62
      |
750   |     fn eth_esp32_emac_default_config(mdc: i32, mdio: i32) -> eth_esp32_emac_config_t {
      |                                                              ^^^^^^^^^^^^^^^^^^^^^^^

      |
     ::: /home/kayo/devel/hardware/water-thermostat/rust-esp32-std-demo/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-9eff167dc7f6d435/out/bindings.rs:23408:1
      |
23408 | pub struct eth_mac_config_t {
      | --------------------------- similarly named struct `eth_mac_config_t` defined here

error[E0422]: cannot find struct, variant or union type `eth_esp32_emac_config_t` in this scope
     --> /home/kayo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/eth.rs:751:9
      |
751   |         eth_esp32_emac_config_t {
      |         ^^^^^^^^^^^^^^^^^^^^^^^
      |
     ::: /home/kayo/devel/hardware/water-thermostat/rust-esp32-std-demo/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-9eff167dc7f6d435/out/bindings.rs:23408:1
      |
23408 | pub struct eth_mac_config_t {
      | --------------------------- similarly named struct `eth_mac_config_t` defined here

error[E0308]: mismatched types
   --> /home/kayo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/espnow.rs:126:53
    |
126 |         esp!(unsafe { esp_now_register_recv_cb(Some(Self::recv_callback)) })?;
    |                                                ---- ^^^^^^^^^^^^^^^^^^^
    |                                                |
    |                                                arguments to this enum variant are incorrect
    |
    = note: expected fn pointer `unsafe extern "C" fn(*const esp_now_recv_info, _, _)`
                  found fn item `extern "C" fn(*const u8, _, _) {EspNow::recv_callback}`
help: the type constructed contains `extern "C" fn(*const u8, *const u8, i32) {EspNow::recv_callback}` due to the type of the argument passed
   --> /home/kayo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.45.0/src/espnow.rs:126:48
    |
126 |         esp!(unsafe { esp_now_register_recv_cb(Some(Self::recv_callback)) })?;
    |                                                ^^^^^-------------------^
    |                                                     |
    |                                                     this argument influences the type of `Some`
note: tuple variant defined here
   --> /home/kayo/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/option.rs:571:5
    |
571 |     Some(#[stable(feature = "rust1", since = "1.0.0")] T),
    |     ^^^^

Some errors have detailed explanations: E0308, E0412, E0422.
For more information about an error, try `rustc --explain E0308`.
error: could not compile `esp-idf-svc` (lib) due to 3 previous errors
```

Vollbrecht, 2023-04-22:
esp-idf v5 currently only works with the git master branch. till we are finished towards the next release, please use 
```toml
[dependencies]
esp-idf-hal = { git="https://github.com/esp-rs/esp-idf-hal"}
esp-idf-svc = { git="https://github.com/esp-rs/esp-idf-svc"}

[patch.crates-io]
esp-idf-hal = { git="https://github.com/esp-rs/esp-idf-hal"}
esp-idf-svc = { git="https://github.com/esp-rs/esp-idf-svc"}
```
inside your cargo.toml

katyo, 2023-04-22:
@Vollbrecht I tested master branch, but error still happens.

ivmarkov, 2023-04-22:
Which master branch? Hopefully NOT the ESP-IDF master branch. Also from above you only need the `[patch.crates-io]` section.

katyo, 2023-04-22:
Oh, seems it happened due to misconfiguration.

closed 2023-04-22 by katyo