esp-rs/esp-idf-svc #258: Missing documentation?

VorpalBlade, 2023-04-26:
There doesn't seem to be much documentation at all in this project. I'm new to embedded rust, and seeing https://esp-rs.github.io/esp-idf-svc/esp_idf_svc/index.html is disheartening.

In addition: what do the various features do? That would also be documented in an ideal world. How does alloc here differ from alloc_handler in esp-idf-sys for example?

Vollbrecht, 2023-04-27:
Related to:  https://github.com/esp-rs/esp-idf-sys/issues/194  https://github.com/esp-rs/esp-idf-hal/issues/242

MihaelBercic, 2024-04-14:
Documentation still sadly fails to build...

Vollbrecht, 2024-04-14:
> Documentation still sadly fails to build...

The issue is about lag of content of documentation not about documentation that is not build-able . Please don't hijack the issue.

Since for `bindgen` generation of the C api we probably will never have working `docs.rs` docs. Because of that we host the docu ourself [here](https://docs.esp-rs.org/esp-idf-svc/esp_idf_svc/index.html). You are also able to build the docs with `cargo docs` locally