esp-rs/esp-idf-svc #263: "connection is not in initial phase" error after failed request

JuliDi, 2023-06-03:
Using the EspHttpConnection like in this example here https://github.com/esp-rs/esp-idf-svc/blob/master/examples/http\_request.rs#L58-L60 I often get `connection is not in initial phase` errors from here https://github.com/esp-rs/esp-idf-svc/blob/b97f7a8d3eb513fcee5cad9d90bf8f0a22a675a1/src/http/client.rs#L421 when calling `submit()` if the previous call of `submit()` failed due to a connection reset (404 does not seem to do any harm).

The code is essentially this snippet here and I would expect that a failed call to `submit()` allows for the client to be reused afterwards.

```rust
fn request(
    url: &str,
    http_client: &mut Client<EspHttpConnection>,
) -> Result<MyData> {
    let mut response = http_client
        .get(url)
        .map_err(|e| anyhow!("Error creating HTTP Request: {e}"))?
        .submit()
        .map_err(|e| anyhow!("Error submitting HTTP Request: {e}"))?;
    // do stuff
}
```

My current workaround is to create a new http client for each request. This is somewhat annoying and probably also very wasteful.

ivmarkov, 2023-06-03:
Not as wasteful as you might think, because dropping a connection where it is in a bad state ("connection reset") means that the whole TCP (and thus SSL, if we are talking HTTPS) connection has to be re-initialized again.

Yet, and as we discussed, I'll try to fix.

Jesse-Cameron, 2024-02-07:
Hi @ivmarkov , I am also experiencing this issue. Is there any other fix I can use other than creating an entirely new connection for each request?

ivmarkov, 2024-02-07:
What is the error you are receiving?

Jesse-Cameron, 2024-02-07:
Hi @ivmarkov,

So the issue is that sometimes a connection times out. And because I am re-using the client (and it's connection) it will throw an error `connection is not in initial phase` then crash/core dump.

I understand that in the case where a connection/request is in a bad state it might tricky to continue to reuse the client. But the current handling isn't ideal.

Will update you with some logs later tonight.

Cheers!

Jesse-Cameron, 2024-02-09:
Hi @ivmarkov , here is a log from my board (I'm using an esp32-c3)

```
W (82513) wifi:[ADDBA]rx delba, code:37, delete tid:7
I (82773) trainer::http: status: 200
I (82773) trainer: first to departure: 28
I (93373) trainer::http: status: 200
I (93383) trainer: first to departure: 28
I (104143) trainer::http: status: 200
I (104153) trainer: first to departure: 28
I (119133) wifi:bcn_timeout,ap_probe_send_start
W (119163) HTTP_CLIENT: Connection timed out before data was ready!
thread 'main' panicked at /home/jesse/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.47.3/src/http/client.rs:477:13:
connection is not in initial phase
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

abort() was called at PC 0x42038b3f on core 0
0x42038b3f - panic_abort::__rust_start_panic::abort
    at /home/jesse/.rustup/toolchains/nightly-2023-10-15-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/panic_abort/src/lib.rs:43
Core  0 register dump:
MEPC    : 0x40381c00  RA      : 0x40386540  SP      : 0x3fc9e790  GP      : 0x3fc91400
0x40381c00 - panic_abort
    at /home/jesse/esp/esp-idf/components/esp_system/panic.c:472
0x40386540 - __ubsan_include
    at /home/jesse/esp/esp-idf/components/esp_system/ubsan.c:313
0x3fc9e790 - _bss_end
    at ??:??
0x3fc91400 - __global_pointer$
    at ??:??
TP      : 0x3fc48764  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
0x3fc48764 - __eh_frame_hdr_end
    at ??:??
0x37363534 -
    at ??:??
0x7271706f - SYSTEM
    at ??:??
0x33323130 -
    at ??:??
S0/FP   : 0x3fc9e7cc  S1      : 0x3fc9e7cc  A0      : 0x3fc9e7cc  A1      : 0x3fc9e7ae
0x3fc9e7cc - _bss_end
    at ??:??
0x3fc9e7cc - _bss_end
    at ??:??
0x3fc9e7cc - _bss_end
    at ??:??
0x3fc9e7ae - _bss_end
    at ??:??
A2      : 0x00000000  A3      : 0x3fc9e7f9  A4      : 0x3fc97bf0  A5      : 0x3fc97bf8
0x00000000 -
    at ??:??
0x3fc9e7f9 - _bss_end
    at ??:??
0x3fc97bf0 - __atexit0
    at ??:??
0x3fc97bf8 - __atexit0
    at ??:??
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fc9e7b0  S3      : 0x3fc9e8b8
0x7a797877 - SYSTEM
    at ??:??
0x76757473 - SYSTEM
    at ??:??
0x3fc9e7b0 - _bss_end
    at ??:??
0x3fc9e8b8 - _bss_end
    at ??:??
S4      : 0x00000001  S5      : 0x00000000  S6      : 0x00000001  S7      : 0x000f4240
0x00000001 -
    at ??:??
0x00000000 -
    at ??:??
[... 584 lines cut ...]
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381bbe
0x40381bbe - esp_restart_noos
    at /home/jesse/esp/esp-idf/components/esp_system/port/soc/esp32c3/system_internal.c:108
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x171c
0x3fcd5820 - _bss_end
    at ??:??
load:0x403cc710,len:0x968
0x403cc710 - _coredump_iram_end
    at ??:??
load:0x403ce710,len:0x2f68
0x403ce710 - _coredump_iram_end
    at ??:??
SHA-256 comparison failed:
Calculated: 1d06b938c0222bf626e0bdf46178b1b37ab24d03f0360fc8fcf7153c2571deaf
Expected: 68d7bdf643ba446b8ed7ae8423241d442fd052b2bc77091100ba06fd65dcf8d5
Attempting to boot anyway...
entry 0x403cc710
0x403cc710 - _coredump_iram_end
    at ??:??
```

ivmarkov, 2024-02-09:
@Jesse-Cameron Can you post a short snippet of your code as well? I don't understand what exactly is going on: for example, are you receiving an error when you try to read/write from/to the connection - yet - you are just logging the error and then trying to re-use the connection?

Jesse-Cameron, 2024-02-09:
Hi @ivmarkov , thanks for getting back to me. 

That's pretty much correct.  You can see an example of my [code here](https://github.com/Jesse-Cameron/trainer/blob/4d2978d97407192eedd3e74c4b929028d0c4f597/src/main.rs#L81-L88)

ivmarkov, 2024-02-09:
OK. So what is happening [here](https://github.com/Jesse-Cameron/trainer/blob/main/src/http.rs#L12) is that the client is returning an error, yet you are swallowing it and re-trying. The problem is, if you currently receive an error from the `EspHttpConnection`, you pretty much have to abandon it and start everything over without swallowing the error. Sure - I can make the `panic!`s into normal errors, but that won't solve anything for you, as starting-over would *still* be required. 

The question is, should the start-over happen automatically in the connection itself (not sure how exactly btw, but we might find a way), or should we leave the problem as-is in that on error condition it should be the user who starts over with a new connection... since I'm not sure having automatic connection recovery is going to buy us anything in terms of performance. If we have received an error from the underlying TCL/TLS stack (and I think that's what's happening to you), then I think the TCP socket and the TLS layer should be re-initiated anyway (as in re-establishing both the TCP connection, and the TLS hanbdshake).

Now, I can do this in `EspHttpConnection`, but you can do it just as well, by not swallowing the error and starting over with a new connection. So if you could elaborate what is the benefit of doing this in the connection itself?

ivmarkov, 2024-02-09:
> The question is, should the start-over happen automatically in the connection itself (not sure how exactly btw, but we might find a way)

To answer myself, the `esp_http_client_cancel_request` API might be a way to do this. But yeah, performance-wise there would be no gain.

ivmarkov, 2024-02-09:
> > The question is, should the start-over happen automatically in the connection itself (not sure how exactly btw, but we might find a way)
> 
> To answer myself, the `esp_http_client_cancel_request` API might be a way to do this. But yeah, performance-wise there would be no gain.

Hmmm, rather `esp_http_client_close`...

Jesse-Cameron, 2024-02-12:
> So what is happening [here](https://github.com/Jesse-Cameron/trainer/blob/main/src/http.rs#L12) is that the client is returning an error, yet you are swallowing it and re-trying. The problem is, if [...]

Ah, yes. I see the issue with my error handling now. I'm only returning the error after the 3rd iteration. So if it fails at any point it'll reuse the connection (which as we discussed isn't ideal).

>  I can make the panic!s into normal errors, but that won't solve anything for you, as starting-over would still be required.

Agreed :+1: 

> The question is, should the start-over happen automatically in the connection itself (not sure how exactly btw, but we might find a way), or should we leave the problem as-is in that on error condition it should be the user who starts over with a new connection

yeah, I guess that's the question. I think that it is possible for the user to re-create the connection.

However, I wonder it might be a little more ergonomic for users of the library if they needn't worry about this? There isn't anything that specifically calls out that connection re-use could result in a `panic!`.

These are just my thoughts though!

enelson1001, 2024-02-22:
I am having a similar problem.  I am getting "Host is unreachable" and don't know what needs to be done to reset this condition as the host really is reachable.
I am calling http://api.openweather.org to collect weather data every 10 minutes. The program runs from1hr up to 12 hours without any problems but then the host unreachable shows up.
```
The error message
I (6656037) wifi:bcn_timeout,ap_probe_send_start␍␊
I (6956587) wifi:state: run -> init (ec0)␍␊
I (6956587) wifi:pm stop, total sleep time: 6243383171 us / 6952598718 us␍␊
␍␊
I (6956587) wifi:<ba-del>idx:0, tid:0␍␊
I (6956597) wifi:<ba-del>idx:1, tid:1␍␊
I (6956597) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1␍␊
<0x1b>[0;31mE (7235977) esp-tls: [sock=54] connect() error: Host is unreachable<0x1b>[0m␍␊
<0x1b>[0;31mE (7235977) transport_base: Failed to open a new connection: 32772<0x1b>[0m␍␊
<0x1b>[0;31mE (7235977) HTTP_CLIENT: Connection failed, sock < 0<0x1b>[0m␍␊
<0x1b>[0;33mW (7235987) weather_forecast: ============ The error is Error creating HTTP Request: ESP_ERR_HTTP_CONNECT<0x1b>[0m␍␊
thread 'main' panicked at src/main.rs:312:80:␍␊
called `Result::unwrap()` on an `Err` value: EspIOError(EspError(-1))␍␊
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace␍␊
```
I tried to unsafe methods but they in turn produce -> called `Result::unwrap()` on an `Err` value: EspIOError(EspError(-1))
1st unsafe method
```
1st unsafe method
unsafe { esp_http_client_cancel_request(httpclient.raw_connection().unwrap().handle());  }

2nd unsafe method
unsafe { esp_http_client_close(httpclient.raw_connection().unwrap().handle());  }
```
If I remove the unsafe method the I get the following errors that never go away unless I reboot the esp32.
```
E (35622345) esp-tls: couldn't get hostname for :api.openweathermap.org: getaddrinfo() returns 202, addrinfo=0x0
E (35622345) transport_base: Failed to open a new connection: 32769
E (35622345) HTTP_CLIENT: Connection failed, sock < 0
```

The code I am using
```
// Do the HTTP request up to three times, return None if request was not successful
fn do_http_request(url: &str) -> Option<String> {
    let mut retries: u16 = 0;
    for _i in 0..3 {
        let httpconnection = EspHttpConnection::new(&HttpConfig {
            use_global_ca_store: true,
            crt_bundle_attach: Some(esp_idf_svc::sys::esp_crt_bundle_attach),
            timeout: Some(Duration::from_secs(10)),
            ..Default::default()
        })
        .expect("Failed to create http connection");

        let mut httpclient = Client::wrap(httpconnection);

        match request(&mut httpclient, url) {
            Ok(json_str) => return Some(json_str),
            Err(e) => {
                warn!("============ The error is {:?}", e);
                unsafe {
                    esp_http_client_cancel_request(httpclient.raw_connection().unwrap().handle());
                }

                // Exponential backoff (e.g., wait 1s, then 2s, then 4s)
                FreeRtos::delay_ms((1 << retries) * 1000);
                retries += 1;
            }
        }
    }
    None
}

// HTTP request
fn request(http_client: &mut Client<EspHttpConnection>, url: &str) -> anyhow::Result<String> {
    let mut response = http_client
        .get(url)
        .map_err(|e| anyhow!("Error creating HTTP Request: {e}"))?
        .submit()
        .map_err(|e| anyhow!("Error submitting HTTP Request: {e}"))?;

    let status_code = response.status();
    match status_code {
        200 => {
            let mut data: Vec<u8> = vec![0; 1024];

            let _data_read = response
                .read(data.as_mut_slice())
                .map_err(|e| anyhow!("Error failed to read response {e}"))?;

            let json_str = std::str::from_utf8(&data)
                .map_err(|e| anyhow!("Error failed to convert http data to utf8: {e}"))?
                .trim_matches(char::from(0))
                .to_string();

            Ok(json_str)
        }
        _ => Err(anyhow!("Error bad status code :{}", status_code)),
    }
}
```

Is there a way to fix this without having to reset the ESP32?

Vollbrecht, 2024-06-21:
should be addressed with #404 if the error still persist with the latest git master feel free to reopen it. Thanks!

closed 2024-06-21 by Vollbrecht

gustavowd, 2025-04-04:
Hi. I'm facing a similar issue with the 0.51 version of esp-idf-svc.
After several https requests, the esp32 simply panic and resets.

The full log is:
`I (342013) rust_esp32_c3_demo: Checking for Demand Response Programs...
thread 'main' panicked at /home/gustavo/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/esp-idf-svc-0.51.0/src/http/client.rs:488:13:
connection is not in initial phase
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

abort() was called at PC 0x42077683 on core 0
0x42077683 - panic_abort::__rust_start_panic::abort
    at /home/gustavo/.rustup/toolchains/nightly-2025-01-03-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/panic_abort/src/lib.rs:48
Core  0 register dump:
MEPC    : 0x40381ce6  RA      : 0x40386c96  SP      : 0x3fca35e0  GP      : 0x3fc93400  
0x40381ce6 - panic_abort
    at /home/gustavo/embedded_rust/rust-esp32-c3-demo/.embuild/espressif/esp-idf/v5.2.3/components/esp_system/panic.c:466
0x40386c96 - __ubsan_include
    at /home/gustavo/embedded_rust/rust-esp32-c3-demo/.embuild/espressif/esp-idf/v5.2.3/components/esp_system/ubsan.c:313
0x3fc93400 - g_timer_info
    at ??:??
TP      : 0x3fc4832c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x3fca361c  S1      : 0x3fca361c  A0      : 0x3fca361c  A1      : 0x3fca35fe  
A2      : 0x00000000  A3      : 0x3fca3649  A4      : 0x00000001  A5      : 0x3fc9a000  
0x3fc9a000 - pxOverflowDelayedTaskList
    at ??:??
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fca3600  S3      : 0x00000000  
S4      : 0x3fca3718  S5      : 0x3fc95ad8  S6      : 0x3fca4e18  S7      : 0x3fca52b4  
0x3fc95ad8 - $d
    at ??:??
S8      : 0x3fca51e0  S9      : 0x00015180  S10     : 0x00000001  S11     : 0x00000005  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001 - _vector_table
    at ??:??
MHARTID : 0x00000000  

Stack memory:
3fca35e0: 0x3fc99ee4 0x3fca366c 0x3fca361c 0x4038df50 0x00000004 0x3c15ae50 0x3c15ad04 0x00000030
0x3fc99ee4 - std::alloc::HOOK
    at ??:??
0x4038df50 - __assert_func
    at /home/gustavo/embedded_rust/rust-esp32-c3-demo/.embuild/espressif/esp-idf/v5.2.3/components/newlib/assert.c:34
0x3c15ae50 - $d
    at ??:??
3fca3600: 0x37303234 0x33383637 0x00000000 0x3fc95064 0x3fca3600 0x3fc95080 0x3fca35fc 0x726f6261
3fca3620: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x37373032 0x20333836
3fca3640: 0x63206e6f 0x2065726f 0x00000030 0x02040000 0x00000001 0x3c15ae50 0x3c1457f8 0x42077686
0x3c15ae50 - $d
    at ??:??
0x3c1457f8 - $d
    at ??:??
0x42077686 - core::result::Result<T,E>::map_err
    at /home/gustavo/.rustup/toolchains/nightly-2025-01-03-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/result.rs:854
3fca3660: 0x3fc95ae8 0x3c1349b0 0x3fc92c50 0x4204e4c4 0x3fca3650 0x3fca3654 0x00000000 0x420630ac
0x3fc95ae8 - $d
    at ??:??
0x3c1349b0 - anon.8bcaa858b0ab3dff9c641046dbf9d90a.91.llvm.15851069076412651452
    at ??:??
0x3fc92c50 - std::panicking::panic_count::LOCAL_PANIC_COUNT::{{closure}}::VAL
    at ??:??
0x4204e4c4 - std::sys::thread_local::key::racy::LazyKey::force
    at /home/gustavo/.rustup/toolchains/nightly-2025-01-03-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/sys/thread_local/key/racy.rs:39
0x420630ac - core::ptr::write
    at /home/gustavo/.rustup/toolchains/nightly-2025-01-03-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:1587
3fca3680: 0x00000001 0x3c15ae50 0x3fcb85ac 0x42066ffa 0x00000011 0x3c15ae50 0x3c1457f8 0x4207318e
0x3c15ae50 - $d
    at ??:??
0x42066ffa - std::thread::local::LocalKey<T>::try_with
    at /home/gustavo/.rustup/toolchains/nightly-2025-01-03-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/std/src/thread/local.rs:307
0x3c15ae50 - $d
[... 51 lines cut ...]
    at /home/gustavo/embedded_rust/ieee2030_5_emb_lib/src/drlc_ieee2030_5.rs:628
3fca3840: 0x8000000d 0x00000009 0x80000000 0x3fcb76bd 0x00000015 0x00000001 0x0000002a 0x00000000
3fca3860: 0x00000001 0x00000000 0x00000000 0x00000008 0x00000000 0x00000000 0x00000000 0x00000042
3fca3880: 0x3fcb7650 0x0000002b 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca38a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca38c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca38e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3900: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3920: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3940: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3960: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca3980: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca39a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca39c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 000000000

Rebooting...`

It seems to be a simple timeout on the get request.
Any thoughts on this?

Thanks in advance,
Best regards,
Gustavo

gustavowd, 2025-04-04:
Removing the panic, I got this:

I (344642) rust_esp32_c3_demo: Checking for Demand Response Programs...
I (344785) esp_idf_svc::http::client: connection is not in initial phase
I (344793) esp_idf_svc::http::client: connection is not in initial phase
E (344798) esp-tls-mbedtls: read error :-0x004C:
E (344799) transport_base: esp_tls_conn_read error, errno=Socket is not connected
I (344804) esp_idf_svc::http::client: connection is not in initial phase
E (344815) esp-tls-mbedtls: read error :-0x004C:
E (344816) transport_base: esp_tls_conn_read error, errno=Socket is not connected
I (344825) rust_esp32_c3_demo: Error: Response Get Error