esp-rs/esp-idf-svc #268: Crate should allow reading/writting values to default NVS storage but doesn't.

punkto, 2023-06-19:
Hi,

I'm trying to write some values to the default NVS partition but I'm not having luck. I'm not sure if the problem is me not configuring the NVS properly or something in the crate.

I'm using the latest crate versions
```
esp-idf-sys = { version = "0.33.0", features = ["binstart"] }
esp-idf-hal = "0.41.1"
esp-idf-svc = "0.46.0"
```

and a ESP32-C3-DevKitM-1. I have a simple program that tries to read and write an u8 to a namespace. It seems to get the partition right, but never writes the variable. The code is the following:
```rust
use esp_idf_hal::delay::FreeRtos;
use esp_idf_hal::gpio::*;
use esp_idf_hal::peripherals::Peripherals;

use esp_idf_svc::nvs::*;

fn main() -> anyhow::Result<()> {
    esp_idf_sys::link_patches();

    let nvs_default_partition: EspNvsPartition<NvsDefault> =
        EspDefaultNvsPartition::take().unwrap();

    let test_namespace = "test_ns";
    let nvs = match EspNvs::new(nvs_default_partition, test_namespace, true) {
        Ok(nvs) => {
            println!("Got namespace {:?} from default partition", test_namespace);
            nvs
        }
        Err(e) => panic!("Could't get namespace {:?}", e),
    };

    let peripherals = Peripherals::take().unwrap();
    let mut led = PinDriver::output(peripherals.pins.gpio4)?;

    loop {
        let tag_test_u8 = "test_u8";

        let is_tag_present = nvs.contains(tag_test_u8).unwrap();

        match is_tag_present {
            true => println!("NVS already contains tag {:?}", tag_test_u8),
            false => println!("NVS does not contain tag {:?}", tag_test_u8),
        };

        let new_value: u8 = if is_tag_present {
            let v: u8 = nvs.get_u8(tag_test_u8).unwrap().unwrap();
            println!("{:?} = {:?}", tag_test_u8, v);
            v + 1
        } else {
            0
        };

        match nvs.set_u8(tag_test_u8, new_value) {
            Ok(_) => println!("Tag updated"),
            Err(e) => println!("Tag not updated {:?}", e),
            // returns this -> Tag not updated EspError(4359) -> ESP_ERR_NVS_INVALID_HANDLE (0x1107): Handle has been closed or is NULL
            // https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/error-codes.html
        };

        led.set_high()?;
        // we are sleeping here to make sure the watchdog isn't triggered
        FreeRtos::delay_ms(1000);

        led.set_low()?;
        FreeRtos::delay_ms(1000);
    }
}
```

I have created a Wokwi project with the code: https://wokwi.com/projects/367961293345482753

The output is the following:
```
I (30) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (30) boot: compile time 12:10:40
I (30) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (41) boot.esp32c3: SPI Speed      : 80MHz
I (45) boot.esp32c3: SPI Mode       : DIO
I (50) boot.esp32c3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  3 factory          factory app      00 00 00010000 0014d000
I (101) boot:  4 ota_0            OTA app          00 10 00160000 0014d000
I (108) boot:  5 ota_1            OTA app          00 11 002b0000 0014d000
I (116) boot: End of partition table
E (120) esp_image: image at 0x160000 has invalid magic byte (nothing flashed here?)
E (129) boot: OTA app partition slot 0 is not bootable
I (134) boot_comm: chip revision: 3, min. application chip revision: 0
I (142) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=25560h (152928) map
I (173) esp_image: segment 1: paddr=00035588 vaddr=3fc8a600 size=00d00h (  3328) load
I (174) esp_image: segment 2: paddr=00036290 vaddr=40380000 size=09d88h ( 40328) load
I (186) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=4aa18h (305688) map
I (233) esp_image: segment 4: paddr=0008aa40 vaddr=40389d88 size=00740h (  1856) load
I (237) boot: Loaded app from partition at offset 0x10000
I (237) boot: Disabling RNG early entropy source...
I (253) cpu_start: Pro cpu up.
I (262) cpu_start: Pro cpu start user code
I (262) cpu_start: cpu freq: 160000000
I (262) cpu_start: Application information:
I (265) cpu_start: Project name:     libespidf
I (270) cpu_start: App version:      1
I (275) cpu_start: Compile time:     Jun 19 2023 13:39:36
I (281) cpu_start: ELF file SHA256:  0000000000000000...
I (287) cpu_start: ESP-IDF:          3cec3a0-dirty
I (292) cpu_start: Min chip rev:     v0.3
I (297) cpu_start: Max chip rev:     v0.99 
I (302) cpu_start: Chip rev:         v0.3
I (306) heap_init: Initializing. RAM available for dynamic allocation:
I (314) heap_init: At 3FC8C270 len 000504A0 (321 KiB): DRAM
I (320) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (326) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (333) spi_flash: detected chip: generic
I (338) spi_flash: flash io: dio
I (342) sleep: Configure to isolate all GPIO pins in sleep state
I (348) sleep: Enable automatic switching of GPIO sleep configuration
I (355) cpu_start: Starting scheduler.
Got namespace "test_ns" from default partition
NVS does not contain tag "test_u8"
Tag updated
NVS does not contain tag "test_u8"
Tag updated
NVS does not contain tag "test_u8"
Tag updated
NVS does not contain tag "test_u8"
Tag updated
NVS does not contain tag "test_u8"
Tag updated
NVS does not contain tag "test_u8"
Tag updated
NVS does not contain tag "test_u8"
Tag updated
NVS does not contain tag "test_u8"
Tag updated

```

punkto, 2023-06-19:
Hi Sergio, thanks for your support.

Please consider mostly the same code for the real hardware in: https://github.com/punkto/esp_rust_nvs_test/tree/main

It gives me the same result. I'm uploading with cargo run, that does the following:
```
espflash --monitor --speed=2000000 --partition-table=partitions.csv target/riscv32imc-esp-espidf/debug/nvs-test
```

Vollbrecht, 2023-06-19:
the problem here lies with the contains method you are using. I tested your code on an esp32c3 with default partition table and your code, but don't do the contain check and it works as expected. ( EG save the count even after reboot)

punkto, 2023-06-19:
Yes, I can reproduce what you are saying. So, if contains does not do what I expected... what does it do ? :sweat_smile: 

(thanks @Vollbrecht !)

ivmarkov, 2023-06-19:
> Yes, I can reproduce what you are saying (with my partitions table with two OTA). So, if contains does not do what I expected... what does it do ? 😅
> 
> (thanks @Vollbrecht !)

I admit it is a bit confusing, but... **don't** use `contains`. You are mixing up two APIs:
* `contains/remove/len/get_raw/set_raw`
  * This API is the "native" Rust API that operates in terms of `u8` slices
* `get_XXX/set_XXX/xxx_len`
  * This API is compatible with the native ESP IDF C API. Useful when you want to share an NVS storage that is used together with ESP-IDF-based C code. This API was added later.

Once again, I get it that it is confusing, and we should've probably added the second API on a different struct or suchlike, but we did not foresee the confusion. We still might change that.

punkto, 2023-06-19:
I get it now. Thanks all. You can close this issue if you want.

BTW, I have tried a version without `contains` in Wokwi and seems to work (same project -> https://wokwi.com/projects/367961293345482753).

ivmarkov, 2023-06-19:
I'll close. Feel free to open a new issue that asks for cleaner separation of the two APIs.

closed 2023-06-19 by ivmarkov