esp-rs/esp-idf-svc #270: Return Err instead of panic when EspNvs::set_str called with no CString compatible strings

punkto, 2023-06-22:
being "with no CString compatible strings" strings with nulls inside (allowed in rust but not in c).

The fix would be around line https://github.com/esp-rs/esp-idf-svc/blob/master/src/nvs.rs#L455 , changing the unwrap for an error raise.

In order to test this, you could call set_str with a value like "hi\0", that currently creates a panic.

If you want, I can send a PR for this. I'm not a Rust guru but it seems easy enough (famous last words :D).

Vollbrecht, 2023-06-26:
@punkto since https://github.com/esp-rs/esp-idf-svc/commit/51fce8e7f0fe5b63ccb3397a41d350fb7da06e92 landed this issue can be closed ?

ivmarkov, 2023-06-26:
What is still outstanding is to merge (a newer version of) the `nvs_read_write.rs` example which is part of [this PR](https://github.com/esp-rs/esp-idf-svc/issues/270).

But I would agree, the example is a separate topic from this issue and in fact should not be related to it.

punkto, 2023-07-01:
You can close it if you like. I'll open another for the example (#273 )

closed 2023-07-01 by ivmarkov