esp-rs/esp-idf-svc #272: Add function to disconnect from MQTT broker

crespum, 2023-06-30:
Rust implementation can't disconnect from a broker in a clean way.

ivmarkov, 2023-06-30:
Please elaborate why the existing MQTT client destruction logic is not good enough.

crespum, 2023-06-30:
In my application I publish some data and send the ESP32 to deep sleep mode right after that. If I want to do it right, I should disconnect from MQTT to make sure I can connect back after waking up.

As far as I've seen in the code, there's only a call to `esp_mqtt_client_destroy()` ([1](https://github.com/esp-rs/esp-idf-svc/blob/master/src/mqtt/client.rs#L573), [2](https://github.com/espressif/esp-mqtt/blob/master/mqtt_client.c#L875C11-L875C34)), which does not mean there's a clean disconnection.

ivmarkov, 2023-06-30:
Deep sleep means that your memory and the state of the CPU is completely gone. Once it awakes from deep sleep, it will essentially start over from the beginning of your `main` function. So if you want to leave your app in clean state, you should actually drop your mqtt client and it's connection (if you are using one). Ditto for the wifi, and for the http server (if you are using one).

ivmarkov, 2023-06-30:
Also: are you sure that the call to `destroy` is not enough for a clean disconnect, and why?

crespum, 2023-06-30:
Correct me if I'm wrong, but even if I drop the MQTT client object, forcing a destruction, the server side might not be aware of this event (depending on the implementation, until a certain timeout).

For instance, to avoid that issue, there's a method for [WiFi disconnection](https://github.com/esp-rs/esp-idf-svc/blob/master/src/wifi.rs#L564) exposed.

ivmarkov, 2023-07-01:
Since you are looking at the C side of things, here's the chain of calls:
`MqttClient::drop` -> `esp_mqtt_client_destroy` -> `esp_mqtt_client_stop` -> `send_disconnect_msg`

Therefore, I'm wondering - are you experiencing an actual problem, or are you just looking at the C code and trying to figure out whether something would break if you don't do anything before going to deep sleep (or just `drop` the mqttclient, as I suggested - but even this might not be necessary)?

Feel free to reopen with additional details in case you experience an actual issue.

closed 2023-07-01 by ivmarkov