esp-rs/esp-idf-svc #275: Can't compile mDNS when using IDF 5.1

lschnartzke, 2023-07-14:
Hi,
I'm currently trying to compile `esp-idf-svc` with mDNS support as I'd like to use it in my project.

I'm unable to compile this crate with mdns enabled.
So far, I've tried to use 
```
[package.metadata.esp-idf-sys]
esp_idf_components = ["mdns"]
```
which fails with "Unable to resolve component mdns".

I also modified the `.cargo/config.toml` file to include 
`rustflags = [ "--cfg",  "espidf_time64", "--cfg", "esp_idf_comp_mdns_enabled"]`
which provides we with error messages related to missing bindings to the mdns C code.
(e.g. ` error[E0425]: cannot find function `mdns_query_srv` in this scope`, originating in the esp-idf-svc crate src/mdns.rs).

I also tried adding `CONFIG_ESP_IDF_COMP_MDNS_ENABLED=y` to `sdkconfig.defaults`, but no luck so far. 

The project has been created using the esp-idf template (with std).

Am I doing something wrong or is using mDNS with IDF v5.1 not (yet) possible?

Additional files:
`sdkconfig.defaults`:
```
# Rust often needs a bit of an extra main task stack size compared to C (the default is 3K)
CONFIG_ESP_MAIN_TASK_STACK_SIZE=7000

# Use this to set FreeRTOS kernel tick frequency to 1000 Hz (100 Hz by default).
# This allows to use 1 ms granuality for thread sleeps (10 ms by default).
#CONFIG_FREERTOS_HZ=1000

# Workaround for https://github.com/espressif/esp-idf/issues/7631
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=n
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=n

CONFIG_ESP_IDF_COMP_MDNS_ENABLED=y
```

`Cargo.toml`:
```
[package]
...
edition = "2021"
resolver = "2"
rust-version = "1.66"

[package.metadata.esp-idf-sys]
esp_idf_components = ["mdns"]

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]

default = ["all", "hal", "esp-idf-sys/native"]

pio = ["esp-idf-sys/pio"]
all = ["std", "nightly", "experimental", "embassy"]
hal = ["esp-idf-hal", "embedded-svc", "esp-idf-svc"]
std = ["alloc", "esp-idf-sys/std", "esp-idf-sys/binstart", "embedded-svc?/std", "esp-idf-hal?/std", "esp-idf-svc?/std"]
alloc = ["embedded-svc?/alloc", "esp-idf-hal?/alloc", "esp-idf-svc?/alloc"]
nightly = ["embedded-svc?/nightly", "esp-idf-svc?/nightly"] # Future: "esp-idf-hal?/nightly"
experimental = ["embedded-svc?/experimental", "esp-idf-svc?/experimental"]
embassy = ["esp-idf-hal?/embassy-sync", "esp-idf-hal?/critical-section", "esp-idf-hal?/edge-executor", "esp-idf-svc?/embassy-time-driver", "esp-idf-svc?/embassy-time-isr-queue"]

[dependencies]
anyhow = "1"
thiserror = "1"
log = { version = "0.4.17", default-features = false }
esp-idf-sys = { version = "0.33", default-features = false }
esp-idf-hal = { version = "0.41", optional = true, default-features = false }
esp-idf-svc = { version = "0.46", optional = true, default-features = false }
embedded-svc = { version = "0.25", optional = true, default-features = false }

[build-dependencies]
embuild = "0.31.2"
```

`.cargo/config.toml`
```
[build]
target = "xtensa-esp32-espidf"

[target.xtensa-esp32-espidf]
linker = "ldproxy"
# runner = "espflash --monitor" # Select this runner for espflash v1.x.x
runner = "espflash flash --monitor" # Select this runner for espflash v2.x.x
rustflags = [ "--cfg",  "espidf_time64", ] # Extending time_t for ESP IDF 5: https://github.com/esp-rs/rust/issues/110

[unstable]
build-std = ["std", "panic_abort"]

[env]
# Note: these variables are not used when using pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "release/v5.1"
ESP_IDF_COMP_MDNS_ENABLED = "1"

```

ivmarkov, 2023-07-14:
Using mDNS with ESP IDF > 4.4 is not possible, because it got moved to a separate repo. I haven't got yet to see how to restore it.

Perhaps you can ask in the Matrix channel?

lschnartzke, 2023-07-14:
Ok, that's unfortunate. I guess I'll downgrade for now, thank you for the fast reply though.

Should I close this issue or leave it open until there's a way to use mDNS again?

ivmarkov, 2023-07-14:
Leave it open.

FrozenDroid, 2023-08-09:
Workaround seems easy enough, just put
```toml
[package.metadata.esp-idf-sys]
extra_components = [
    { component_dirs = [ "esp-protocols/components/mdns" ] }
]
```
in your `Cargo.toml` and clone the [esp-protocols repo](https://github.com/espressif/esp-protocols) to your root.

N3xed, 2023-08-09:
See esp-rs/esp-idf-sys#222.

ivmarkov, 2023-09-21:
Closing, as support for the ESP-IDF component Manager was merged in mainline, and the next release of this crate should support the mDNS component for ESP IDF 5, as long as you include it in your project.

closed 2023-09-21 by ivmarkov

benhansen-io, 2023-11-16:
For anyone still coming here from Google, the updated Cargo.toml snipped to add that worked for me was:

```
[[package.metadata.esp-idf-sys.extra_components]]
remote_component = { name = "espressif/mdns", version = "1.2" }
```

And the idf.py command to add the external component was:

```
idf.py add-dependency "espressif/mdns^1.2.2"
```