esp-rs/esp-idf-svc #277: Consider releasing patch release with fix for `home_chan_dwell_time`

faern, 2023-07-23:
Building `esp-idf-svc` against ESP-IDF 4.4, 5.1 and master is broken due to https://github.com/espressif/esp-idf/issues/11921 / https://github.com/esp-rs/esp-idf-template/issues/139.

This has been fixed in commit https://github.com/esp-rs/esp-idf-svc/commit/42e60e8c3fcca670632b9653c790fd8efe475aac, but not yet released on crates.io. Could you please consider making a patch release? Currently nothing builds unless you replace your `esp-idf-svc` dependencies with the git `master` version.

I'm creating this issue to be able to track a release. But also because there are zero hits in the issue tracker for "home_chan_dwell_time". So this can be a landing for everyone who hits this error.

```
missing field `home_chan_dwell_time` in initializer of `esp_idf_sys::wifi_scan_config_t`
```

MihaMarkic, 2023-07-23:
>  unless you replace your esp-idf-svc dependencies with the git master version

So, what exactly is the way to do it (consider myself as Rust newbie)? Get sources (including patch) from git and use them instead of crate?

MihaMarkic, 2023-07-23:
Just in case anybody else was wondering, placing this in Cargo.toml seems to work:
```toml
[patch.crates-io]
esp-idf-svc = { git = "https://github.com/esp-rs/esp-idf-svc" }
```

flyaruu, 2023-07-30:
Indeed, a patch release would be really welcome.

ivmarkov, 2023-07-30:
`0.46.1` released.

closed 2023-07-30 by ivmarkov