esp-rs/esp-idf-svc #278: Allow more control for OTA update

svenstaro, 2023-07-24:
The current OTA update API doesn't leave much control for the user. `complete()` currently always calls
```rust
esp!(unsafe { esp_ota_end(self.update_handle) })?;
esp!(unsafe { esp_ota_set_boot_partition(self.update_partition) })?;
```
and that ensures that the flashed OTA partition will always boot when the device next restarts. This might be fine in some cases, but in others, we might want more control. For instance, write an OTA image and then later on, after some other external conditions are met, set the new image as the boot image and reboot the box.

I propose an API change to give the user more control but I'd like to coordinate this first as it'll be a breaking change and I might even have to touch the trait.

I propose `complete()` to simply call
```rust
esp!(unsafe { esp_ota_end(self.update_handle) })?;
```
and that's that. Then, I'll build a new function
```rust
pub fn activate(partition: Option<esp_partition_t>) {
  if let Some(partition) = partition {
    esp!(unsafe { esp_ota_set_boot_partition(partition) })?;
  } else {
    let partition = esp_ota_get_next_update_partition(ptr::null());
    esp!(unsafe { esp_ota_set_boot_partition(partition) })?;
  }
}
```

The implementation should be pretty straightforward.

This will definitely be confusing for users as while this is not an API breakage, it will be a behavior breakage as now you're expected to call `activate()` manually after `complete()`.

Open question: What to do with the trait in `embedded-svc`?

Happy about feedback.

ivmarkov, 2023-07-30:
> The current OTA update API doesn't leave much control for the user. `complete()` currently always calls
>
> ```rust
> esp!(unsafe { esp_ota_end(self.update_handle) })?;
> esp!(unsafe { [...]

I would rather keep the current semantics of `complete` and would additionally introduce a new function - say - `finish` - that only does what you want. So in a way, `complete` would do what `finish` does, plus automatically set the `finish`-ed partition as a boot partition.

> Then, I'll build a new function
>
> ```rust
> pub fn activate(partition: Option<esp_partition_t>) {
>   if let Some(partition) = partition {
>     esp!(unsafe { esp_ota_set_boot_partition(partition) })?;
>   } [...]

This API is a bit hacky, as it takes a raw `esp_partition_t` struct. What the safe API does expose is the notion of a `Slot`. So ideally, the API should take a `Slot` reference. 

Second, to operate with slots, you might need an additional API, which would list all slots.

Third, passing `None` and then assuming that the slot you just wrote to is precisely the slot that `esp_ota_get_next_update_partition` would return does not sound right to me.

kyrias, 2023-08-07:
> This API is a bit hacky, as it takes a raw `esp_partition_t` struct. What the safe API does expose is the notion of a `Slot`. So ideally, the API should take a `Slot` reference.

Instead of explicitly having to pass around `Slot`s for this, what do you think about your suggested `EspOtaUpdate::finish` method returning a new struct like this instead?

```rust
struct EspOtaUpdateFinished {
    partition: *const esp_partition_t,
}
```

This type could then have a single method called `activate` which only acts on its private field.  That way the user would not need to manually figure out which `Slot` is/was used, nor would we need to change the API to let the user either pass in the `Slot` to use when starting the update or return the `Slot` used from `EspOtaUpdate`.

svenstaro, 2023-09-06:
@ivmarkov what do you think about the suggestion by @kyrias?

ivmarkov, 2023-09-07:
We need to check if passing `esp_partition_t` raw pointers around is actually safe. I'm not sure until when that pointer would be valid. And the structure does not have any lifetime attached.

kyrias, 2023-09-11:
@ivmarkov Pointers to `esp_partition_t` returned by the ESP-IDF partitions API are valid for the lifetime of the program.

ivmarkov, 2023-09-21:
* Do you need to use/call into the `EspOta` service while those "external conditions"  are being met? Hopefully not?
* Are you OK with the `EspOtaUpdateFinished` being life-timed (so that I can keep the `EspOta` instance locked, so that you cannot initiate and then abort in the middle another update)?

If you are OK with the above two restriction, I'll implement it roughly the way you suggest, yes.

ivmarkov, 2023-09-21:
Keep in mind the OTA API in current mainline has slightly changed, (so that I can now actually implement the above tricks with lifetimes).

kyrias, 2023-09-21:
> * Do you need to use/call into the `EspOta` service while those "external conditions"  are being met? Hopefully not?
> 

Nope.

> * Are you OK with the `EspOtaUpdateFinished` being life-timed (so that I can keep the `EspOta` instance locked, so that you cannot initiate and then abort in the middle another update)?

That's exactly what we want, yes.

ivmarkov, 2023-09-21:
OK, [done](https://github.com/esp-rs/esp-idf-svc/commit/f63722781fe12c513afa223b99a2c807498e9a0e).

Feel free to reopen if somehow I misunderstood your constraints.

(BTW: CI is failing for a completely unrelated issue.)

closed 2023-09-21 by ivmarkov

kyrias, 2023-09-22:
@ivmarkov I just tried to switch over to the new API but it turns out that it's harder to use than I was expecting.

So to give a bit more info about our setup, we're doing OTA updates over Bluetooth and the OTA process is controlled through message passing.  I wanted to use the new API in a state machine structured roughly like this:

```rust

enum OtaState<'a> {
    Uninitialized,
    Initialized(EspOtaUpdate<'a>),
    Finished(EspOtaUpdateFinished<'a>),
}
let mut ota = EspOta::new().unwrap();
let mut state = OtaState::Uninitialized;

loop {
    match state {
        OtaState::Uninitialized => match command_rx.recv().unwrap {
            Command::Initialize => state = OtaState::Initialized(ota.initiate_update().unwrap()),
            Command::FactoryReset => ota.factory_reset().unwrap(),
        }
        ...
    }
}
```

Which doesn't work since the initialize branch borrows the `ota` which leads to `match state` using that borrow, and which means that `ota.factory_reset()` is then theoretically a second use of the borrow even though execution can never that state.  I guess we could unroll the state machine to be a sequence of loops instead, but I think it'd be nicer if it was possible to have this be an enum-backed state machine instead.

And so I'm now thinking that it might be a more ergonomic API if `EspOtaUpdate` and `EspOtaUpdateFinished` took ownership of the `EspOta` and then returned it when they're consumed.  That means that the entire state machine could be modeled as an enum since the `EspOta` would not need to remain at a fixed memory location for the duration of the OTA process.  What do you think?

ivmarkov, 2023-09-22:
@kyrias This had come once in the past (not sure if from your org?) and then I did a compromise (that was before GATs got stabilized in Rust so at that time we could not return `EspOtaUpdate<'a>` anyway).

Now I really feel uneasy doing this once again. 

For one, I'm not convinced that would be a better overall API **for the general case**. 
What happens if `EspOtaUpdate::finish` fails for some reason? You'll not get back your `EspOta` instance, as `finish` is `finish(self) -> Result<EspOta, EspError>`. To remedy this, you need `finish(self) -> Result<EspOta, (EspOta, EspError)>`. Not the end of the world, but definitely more un-ergonomic for those users which are not arranging the code your way.

So rather than doing an enum, I would suggest you do nested loops.

kyrias, 2023-09-22:
> For one, I'm not convinced that would be a better overall API **for the general case**.
> What happens if `EspOtaUpdate::finish` fails for some reason? You'll not get back your `EspOta` instance, as [...]

I'd instead have recommended a new error type (`EspOtaUpdateFinishedError` or similar) that is infallibly convertible to `EspError` but also allows for taking the inner `EspOta`' akin to how e.g. `std::sync::PoisonError` works. 

I would personally guess that wanting to explicitly track the state using an enum would be more commonly wanted than not since that's the more idiomatic way of representing that kind of state in Rust, but it's fine, the underlying code is more or less the same just longer and a bit harder to read.

ivmarkov, 2023-09-22:
> I would personally guess that wanting to explicitly track the state using an enum would be more commonly wanted than not since that's the more idiomatic way of representing that kind of state in Rust, but it's fine, the underlying code is more or less the same just longer and a bit harder to read.

You might or might not have a point here. I guess it is a matter of personal preference, and as they say, beauty is in the eyes of the beholder. :)

For one, I'm firmly in the "avoid explicit state machines and do it with sequential code and nested loops instead", as I find this much easier to read and follow, as you are just using standard control-flow constructs - `if`, `loop` etc. 

Not that I've successfully avoided those everywhere - while my programs tend to be a bunch of loops (sometimes nested) and all running in parallel as "tasks", at the top level these need to talk to each other, which often comes in the shape of a "broadcast queue" of sorts, whose messages suspiciously look like simple state enums. But that's at the top level.

Now, this sequential code sometimes does not come for free, in that you might need a bunch of threads to model the loops, which on embedded is not necessarily ideal. Which is where `async` helps, as it is essentially the compiler way to turn your readable (again, beauty is in the eyes of the beholder), sequential code into efficient enum-based state machines that all run off a single stack/thread.

Anyway. Just a different POV.

dberlin, 2023-09-27:
Just to share a view here:

For reference.I've attached a bluetooth OTA that was working fine in the old world (slot based) and is now much harder to make work.  
[bluetooth_ota.rs.txt](https://github.com/esp-rs/esp-idf-svc/files/12741473/bluetooth_ota.rs.txt)

I've also created a gist with the diff of the old/new here:
https://gist.github.com/dberlin/159ca432123870b46b3713519af8b97f

The only difference between the old and new code is that the old code used get_update on the updater, and the new code has to store the update struct given by initiate_update.
It's about a 15 line code diff from old to new.

Otherwise, it's a simple updater.  It dose nothing crazy.  The main goal is to not have to buffer the entire update in memory, etc.
It would obviously work in a synchronous loop that did nothing but receive packets, and the lifetime of the updater pieces were obviously scoped to that single function.

This code, which does not work as-is, for a number of reasons:

1. The lifetime of what is returned by initiate_update is anonymous, despite it really having a static lifetime.
This causes rust to be very unsure how long it will ever live, and makes it hard to store and use in a struct.
It does not understand that that the initiate_update return lives longer than process_rx in this code.
You can't work around this by making just the current updater a global static  because the types are not marked Send/Sync.
It could probably be worked around by forcing BluetoothOTA struct to be a global static (since i think my unsafe marking of send/sync will override the fields not all being send/sync), but IMHO, it seems better to properly mark the lifetime of the initiate_update result as static.

2. The EspOtaUpdate struct, despite being just some pointers, isn't marked copy.  This means that attempts to use it in an Option, as you see here, are difficult because unwrapping the option requires moving the struct, which rules out using mutable references as you normally would.
IE self.current_updater.unwrap().complete() is invalid if self is a mutable reference, because it requires moving the value out of the option.
Any attempt to use the option value (matching, etc) will fail in the same way, ie  "if let Some(updater) = self.current_updater"
This can be mostly worked around with takes around the updater use, but it's still ugly for a structure that should (imho), be Copy. 

3. There is a similar issue with the EspOtaUpdateFinished struct.

It would be really nice (and makes this code work as is) if 

1. The lifetime of the result of initiate_update was static instead of anonymous. Working around this requires hacks like global structs that are force-marked send/sync.
As mentioned, right now you can't even make a global static to store the current updater to work around it because the underlying types are not send/sync.  You have to use interior mutability tricks.

2. It would be really nice to mark the structs as copy - any safety issues that exist do not come from them being copy or not :)
That avoids having to do takes around them, which will likely confuse a lot of people.

There are other ways to solve all of this, of course.

In practice, it all stems from the fact that EspOta used to encapsulate the entire OTA process fairly directly, so as you can see you could just call get_update to see whether you had a current slot or not. 
That made lifetime management easier, and you didn't have to store anything but an opaque updater instance.
That seems the opposite direction you want to go, but i mention it for completeness of cataloging solutions.

For now i have local patches to esp-idf-svc to mark the initiate_update result as static lifetime, and make the OtaUpdate* structs copy, which makes this updater compile/work.

kyrias, 2023-09-28:
@ivmarkov One thing I just realized, I think that `EspOtaUpdate` needs to have a `Drop` implementation that aborts the OTA update since otherwise the OTA update handle is never freed when dropped.

ivmarkov, 2023-09-28:
Indeed. Would you open a PR, or shall I do it?

kyrias, 2023-09-28:
I should have the time to do it later today, just wanted to check that you agree that it would be reasonable first.  :)

ivmarkov, 2023-09-28:
Of course it would be reasonable. :) I only hope it would work without major changes, as we have a few methods that take `self`. But I think they are not moving anything, so that should be OK...