esp-rs/esp-idf-svc #279: `EspMdns::query` crashes when no services are found

lschnartzke, 2023-07-27:
When running mDNS queries that return no results, the `EspMdns::query` function crashes the microcontroller. As far as I can tell, this is due to the   https://github.com/esp-rs/esp-idf-svc/blob/269657361af4677dc2d4a05c3312da2a1f6ebc5a/src/mdns.rs#L347

for which `result.is_null()` still returns true, if `mdns_query` does not find any services. This causes a panic by the
 https://github.com/esp-rs/esp-idf-svc/blob/269657361af4677dc2d4a05c3312da2a1f6ebc5a/src/mdns.rs#L363
(`thread 'main' panicked at 'unsafe precondition(s) violated: NonNull::new_unchecked requires that the pointer is non-null'`).

I fixed this with a simple
```rust
if result.is_null() {
            Ok(0)
        } else {
            Ok(copy_query_results(
                unsafe { Box::from_raw(result) },
                results,
            ))
        }
```

If there's nothing more to do, I can open a pull request for this.

The function also crashes when passing `None` to the `name` parameter of the function, though this might also be caused by simultaneously using `Type::Any`, just fyi.

Side question: Is there any more elegant way to work with mdns? At the moment I have to create a `Vec` of manually created `QueryResults` to pass as the results array, which is kind of annoying.

ivmarkov, 2023-07-30:
> ```rust
> if result.is_null() {
>             Ok(0)
>         } else {
>             Ok(copy_query_results(
>                 unsafe { Box::from_raw(result) },
>                 results,
>             ))
>         }
> ```
> 

If you could open a PR, that would be great, yes.

> Side question: Is there any more elegant way to work with mdns? At the moment I have to create a `Vec` of manually created `QueryResults` to pass as the results array, which is kind of annoying.

Not really. We do try to avoid the `alloc` module in the Rust wrappers (ESP IDF does allocate though), hence our APIs are slightly more inconvenient to work with.

ivmarkov, 2023-07-30:
Actually, I've implement a simpler and more correct fix. The whole `Box` thing is completely unnecessary in the first place.

closed 2023-07-30 by ivmarkov