esp-rs/esp-idf-svc #281: Pin thread to specific core

reinismu, 2023-08-01:
Is there a way with STD Rust to pin thread to specific core?

As I understand threading is made based on top of FreeRTOS. There is https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/freertos_idf.html#_CPPv429xTaskCreateStaticPinnedToCore14TaskFunction_tPCKcK8uint32_tPCv11UBaseType_tPC11StackType_tPC12StaticTask_tK10BaseType_t function.

Any way to accomplish it in a idiomatic way?

Vollbrecht, 2023-08-01:
you can use https://github.com/esp-rs/esp-idf-hal/blob/master/src/task.rs#L165 ThreadSpawnConfiguration struct with its pin_to_core option and then just use standard std::thread mechanism on top of it

reinismu, 2023-08-01:
> you can use https://github.com/esp-rs/esp-idf-hal/blob/master/src/task.rs#L165 ThreadSpawnConfiguration struct with its pin_to_core option and then just use standard std::thread mechanism on top of it.
> 
> * Creating object
> * use set() method on it
> * use std::thread mechanism

Awesome thanks! Are there any examples in the wild that I could check?

closed 2023-08-02 by reinismu