esp-rs/esp-idf-svc #293: AsyncEspTls panics with BorrowMutError

torkleyy, 2023-08-29:
## Code

```rust
pub struct AsyncTls(pub AsyncEspTls<AsyncTcp>);

impl AsyncRead for AsyncTls { ... }
impl AsyncWrite for AsyncTls { ... }

pub async fn connect_async_tls(
    hostname: &str,
    port: u16,
    cfg: &esp_idf_svc::tls::Config<'_>,
) -> Result<AsyncTls> {
    let mut tls = AsyncEspTls::adopt(AsyncTcp(
        Async::<TcpStream>::connect((hostname, port).to_socket_addrs()?.next().unwrap()).await?,
    ))
    .map_err(|e| anyhow!("failed to create AsyncEspTls: {e}"))?;
    tls.negotiate(hostname, cfg).await?;

    Ok(AsyncTls(tls))
}

async fn task() -> anyhow::Result<()> {
    info!("Connecting tls...");
    let mut tls = connect_async_tls(
        "example.com",
        443,
        &tls::Config {
            ca_cert: Some(X509::der(CA_CERT)),
            common_name: Some("example.com"),
            ..Default::default()
        },
    )
    .await?;
    info!("Connected tls");
    tls.write_all(b"GET / HTTP/1.1\r\nHost: example.com\r\nConnection: close\r\n\r\n")
        .await?;
    info!("Wrote tls");
    sleep(Duration::from_secs(1));
    let mut buf = [0; 1024];
    tls.read(&mut buf).await?;
    let s = String::from_utf8_lossy(&buf);
    info!("Response:\n{s}");

    Ok(())
}

fn main() {
    // ...

    esp_idf_sys::esp!(unsafe {
        esp_idf_sys::esp_vfs_eventfd_register(&esp_idf_sys::esp_vfs_eventfd_config_t {
            max_fds: 5,
            ..Default::default()
        })
    })?;

    async_io::block_on(async_executor::LocalExecutor::new().run(task()))?;
}

```

## Output

```
I (10799) eval_esp_rust: setting eventfd config
I (10799) eval_esp_rust: starting executor
I (10809) eval_esp_rust: Connecting tls...
W (10989) esp-tls: Failed to open new connection in specified timeout
thread 'main' panicked at 'already borrowed: BorrowMutError', S:\esp-idf-svc\src\tls.rs:556:56
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

abort() was called at PC 0x40112496 on core 0
0x40112496 - panic_abort::__rust_start_panic::abort
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\panic_abort\src\lib.rs:42

Backtrace: 0x40082c96:0x3ffb1d40 0x400891f5:0x3ffb1d60 0x4008fbf2:0x3ffb1d80 0x40112496:0x3ffb1df0 0x4011248a:0x3ffb1e10 0x400f4fe2:0x3ffb1e30 0x400f4f94:0x3ffb1e80 0x4010381b:0x3ffb1ef0 0x401036e0:0x3ffb1f20 0x400f4d22:0x3ffb1f40 0x4012239c:0x3ffb1f70 0x401241ff:0x3ffb1fb0 0x400d7941:0x3ffb2010 0x400d79c5:0x3ffb2040 0x400d955e:0x3ffb2070 0x400da305:0x3ffb2250 0x400d8b60:0x3ffb3d60 0x401a660b:0x3ffb4a80 0x400db850:0x3ffb4aa0 0x401012f2:0x3ffb4ac0 0x400db840:0x3ffb4af0 0x400d8c7f:0x3ffb4b20 0x400f21b3:0x3ffb4b40 0x401ac347:0x3ffb4b60        
0x40082c96 - panic_abort
    at S:\eval-esp-rust\target\xtensa-esp32-espidf\debug\build\esp-idf-sys-d5dfe0e6af70683e\out\build\C:/Users/me/.espressif/esp-idf/v5.1.1/components/esp_system\panic.c:452
0x3ffb1d40 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400891f5 - esp_system_abort
    at S:\eval-esp-rust\target\xtensa-esp32-espidf\debug\build\esp-idf-sys-d5dfe0e6af70683e\out\build\C:/Users/me/.espressif/esp-idf/v5.1.1/components/esp_system/port\esp_system_chip.c:84
0x3ffb1d60 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x4008fbf2 - abort
    at S:\eval-esp-rust\target\xtensa-esp32-espidf\debug\build\esp-idf-sys-d5dfe0e6af70683e\out\build\C:/Users/me/.espressif/esp-idf/v5.1.1/components/newlib\abort.c:38   
0x3ffb1d80 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x40112496 - panic_abort::__rust_start_panic::abort
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\panic_abort\src\lib.rs:42
0x3ffb1df0 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x4011248a - __rust_start_panic
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\panic_abort\src\lib.rs:37
0x3ffb1e10 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400f4fe2 - rust_panic
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\panicking.rs:755
0x3ffb1e30 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400f4f94 - std::panicking::rust_panic_with_hook
    at ??:??
0x3ffb1e80 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x4010381b - std::panicking::begin_panic_handler::{{closure}}
    at ??:??
0x3ffb1ef0 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x401036e0 - std::sys_common::backtrace::__rust_end_short_backtrace
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\sys_common\backtrace.rs:151
0x3ffb1f20 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400f4d22 - rust_begin_unwind
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\panicking.rs:593
0x3ffb1f40 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x4012239c - core::panicking::panic_fmt
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\panicking.rs:67
0x3ffb1f70 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x401241ff - core::result::unwrap_failed
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\result.rs:1651
0x3ffb1fb0 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400d7941 - core::result::Result<T,E>::expect
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\result.rs:1033
0x3ffb2010 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400d79c5 - esp_idf_svc::tls::esptls::AsyncEspTls<S>::wait::{{closure}}::{{closure}}
    at S:\esp-idf-svc\src\tls.rs:556
0x3ffb2040 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400d955e - esp_idf_svc::tls::esptls::AsyncEspTls<S>::wait::{{closure}}
    at S:\esp-idf-svc\src\tls.rs:557
0x3ffb2070 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400da305 - async_executor::Executor::run::{{closure}}
    at C:\Users\me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\async-executor-1.5.1\src\lib.rs:243
0x3ffb2250 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400d8b60 - eval_esp_rust::app
    at S:\eval-esp-rust\src\main.rs:127
0x3ffb3d60 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x401a660b - core::ops::function::FnOnce::call_once
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\ops\function.rs:250
0x3ffb4a80 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400db850 - std::rt::lang_start::{{closure}}
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\rt.rs:166
0x3ffb4aa0 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x401012f2 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\ops\function.rs:284
0x3ffb4ac0 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400db840 - std::rt::lang_start
    at C:\Users\me\.rustup\toolchains\esp\lib\rustlib\src\rust\library\std\src\rt.rs:165
0x3ffb4af0 - g_rom_spiflash_dummy_len_plus
    at ??:??
0x400d8c7f - main
    at ??:??
```

## Versions

* esp idf: v5.1.1
* rust: rustc 1.71.0-nightly (4ca000ac8 2023-07-13) (1.71.0.1)

---

I think the error is caused because of "Failed to open new connection in specified timeout", although I'm not sure why that even happens. In any case, the panic is a bug.

I can upload a full example to a repo if needed.

ivmarkov, 2023-08-29:
Please upload a full example. In any case, the error is NOT because of "failed to open new connection..." (which is a separate issue altogether), but because of 
"thread 'main' panicked at 'already borrowed: BorrowMutError', S:\esp-idf-svc\src\tls.rs:556:56"

... which should not happen, and which might be a bug, but to confirm or deny, I need access to the source code, including your implementation of `Socket` and `PollableSocket`.

Also this:
```rust
async_io::block_on(async_executor::LocalExecutor::new().run(task()))?;
```

... is weird. Use one (async_io::block_on) or the other (async_executor::LocalExecutor), but not both. As in simply `async_io::block_on(task())`.

If that helps, please find below my testing code:

```rust
pub struct EspTlsSocket(Option<async_io::Async<TcpStream>>);

impl EspTlsSocket {
    pub const fn new(socket: async_io::Async<TcpStream>) -> Self {
        Self(Some(socket))
    }

    pub fn handle(&self) -> i32 {
        self.0.as_ref().unwrap().as_raw_fd()
    }

    pub fn poll_readable(&self, ctx: &mut core::task::Context) -> core::task::Poll<Result<(), esp_idf_svc::sys::EspError>> {
        self.0.as_ref().unwrap().poll_readable(ctx).map_err(|_| EspError::from_infallible::<{ esp_idf_svc::sys::ESP_FAIL }>())
    }

    pub fn poll_writeable(&self, ctx: &mut core::task::Context) -> core::task::Poll<Result<(), esp_idf_svc::sys::EspError>> {
        self.0.as_ref().unwrap().poll_writable(ctx).map_err(|_| EspError::from_infallible::<{ esp_idf_svc::sys::ESP_FAIL }>())
    }

    fn release(&mut self) -> Result<(), esp_idf_svc::sys::EspError> {
        let socket = self.0.take().unwrap();
        socket.into_inner().unwrap().into_raw_fd();

        Ok(())
    }
}

impl esp_idf_svc::tls::Socket for EspTlsSocket {
    fn handle(&self) -> i32 {
        EspTlsSocket::handle(self)
    }

    fn release(&mut self) -> Result<(), esp_idf_svc::sys::EspError> {
        EspTlsSocket::release(self)
    }
}

impl esp_idf_svc::tls::PollableSocket for EspTlsSocket {
    fn poll_readable(&self, ctx: &mut core::task::Context) -> core::task::Poll<Result<(), esp_idf_svc::sys::EspError>> {
        EspTlsSocket::poll_readable(self, ctx)
    }

    fn poll_writable(&self, ctx: &mut core::task::Context) -> core::task::Poll<Result<(), esp_idf_svc::sys::EspError>> {
        EspTlsSocket::poll_writeable(self, ctx)
    }
}

fn test_https_client() -> anyhow::Result<()> {
    async fn test() -> anyhow::Result<()> {
        let addr = "google.com:443".to_socket_addrs()?.next().unwrap();
        let socket = Async::<TcpStream>::connect(addr).await?;

        let mut tls = esp_idf_svc::tls::AsyncEspTls::adopt(EspTlsSocket::new(socket))?; 

        tls.negotiate(
            "google.com", 
            &esp_idf_svc::tls::Config {
                use_crt_bundle_attach: true,
                ..Default::default()
            },
        )
        .await
        .unwrap();

        tls.write_all(b"GET / HTTP/1.0\r\n\r\n").await.unwrap();

        let mut body = [0_u8; 3048];

        let read = tls.read_fully(&mut body).await.unwrap();

        info!(
            "Body (truncated to 3K):\n{:?}",
            String::from_utf8_lossy(&body[..read]).into_owned()
        );

        Ok(())
    }

    smol::block_on(test())
}
```

ivmarkov, 2023-08-29:
`read_fully` and `write_all` I've removed in the meantime, but their implementation is pretty clear.

torkleyy, 2023-08-29:
> In any case, the error is NOT because of "failed to open new connection..." 

Actually, I was able to avoid the panic by specifying `timeout_ms: 2000` in the `esp_tls::tls::Config` (previously, it was 0 which will always lead to an immediate timeout). Of course it is not the reason for the panic, but it is only triggered in this case.

I will compare my `*Socket` impls with yours and see if I can pinpoint the issue.

torkleyy, 2023-08-29:
@ivmarkov Repo for reproduction here:

https://github.com/torkleyy/repro-async-tls

ivmarkov, 2023-08-29:
@torkleyy I just fixed something. Can you try again? Chance that fixing it is not very high, but let's see.

ivmarkov, 2023-08-29:
Btw and probably unrelated, but your `AsyncTls` is unnecessarily complex:
* You implement on it `AsyncRead` and `AsyncWrite` which is not necessary
* `poll_readable` and `poll_writable` are implemented with pinning and calling `poll` on `readable` and `writable` which is an unnecessary overhead, given that `Async` has native `poll_readable` and `poll_writable` methods that you can just call - as I do

torkleyy, 2023-08-30:
FYI, I've confirmed that `BorrowMutError` no longer occurs with your latest fix. However, I would like to check what exactly happens on timeout, because as far as I could see no error was returned.

torkleyy, 2023-08-30:
Okay, so the sync connect just returns 0 as if the connect was successful:

https://github.com/espressif/esp-idf/blob/82cceabc6e6a0a2d8c40e2249bc59917cc5e577a/components/esp-tls/esp_tls.c#L534-L538

The `connect_sync` is called because I did not change my `Config::non_blocking` to `true` (should we check this somewhere for the `AsyncEspTls`?). I'm not sure if it's right that we call `connect_sync` when we really want non-blocking communication.

In any case, the combination of `timeout_ms` = 0, `non_blocking` = `false` and `AsyncEspTls::negotiate` causes the app to hang indefinitely.

ivmarkov, 2023-08-30:
I've pushed a number of enhancements and a few bugfixes. Read on.

> Okay, so the sync connect just returns 0 as if the connect would block:
> 
> https://github.com/espressif/esp-idf/blob/82cceabc6e6a0a2d8c40e2249bc59917cc5e577a/components/esp-tls/esp_tls.c#L534-L538
> 
> The `connect_sync` is called because I did not change my `Config::non_blocking` to `true` (should we check this somewhere for the `AsyncEspTls`?). I'm not sure if it's right that we call `connect_sync` when we really want non-blocking communication.

`connect_sync` was called by mistake. `connect_async` is now called.

> In any case, the combination of `timeout_ms` = 0, `non_blocking` = `false` and `AsyncEspTls::negotiate` causes the app to hang indefinitely.

* `timeout_ms` now has a default value of 4 seconds. In any case, it is now only used if the user requests blocking connection (non_blocking = false)
* When `AsyncEspTls` is used, the value of `non_blocking` in the config does not matter either and is disregarded

torkleyy, 2023-08-30:
That makes sense.

I checked out the changes, it doesn't seem to work for me though. No matter the timeout value, `AsyncEspTls::negotiate` never becomes ready.

torkleyy, 2023-08-30:
@ivmarkov I cannot reproduce the problem with the dependency versions you are using.

ivmarkov, 2023-08-30:
> @ivmarkov I cannot reproduce the problem with the dependency versions you are using.

There are no other dependencies that actually work and im aware of. If you've done something on your own I suggest you abandon it and wait until the new `smol-rs` crates are finally released. Unfortunately we've hit a bad moment with upstreaming the esp idf support as the smol crates are in the midst of a major version release.