esp-rs/esp-idf-svc #299: no method named `ws_handler` found in EspHttpServer type

fabracht, 2023-09-20:
I'm trying to write a WebSocket server using esp-idf-svc 0.46.2.
I created a project using the esp-idf template and followed the "Guess" websocket example, but I can't use the server Web socket functionality. 

For reference:
Here's how I create the server
```rust
fn start_http_server() -> Result<EspHttpServer, Box<dyn std::error::Error>> {
    let conf = HttpConfiguration {
        stack_size: 10240,
        ..Default::default()
    };
    let mut server = EspHttpServer::new(&conf)?;

    server.fn_handler("/", embedded_svc::http::Method::Get, |req| {
        req.into_ok_response()?.write(INDEX_HTML.as_bytes())?;
        Ok(())
    })?;
    server.fn_handler("/*", embedded_svc::http::Method::Get, |req| {
        req.into_status_response(404)?;
        Ok(())
    })?;
    server.ws_handler("/ws", move |ws| {});
    info!("Started http server");
    Ok(server)
}
```

This is my Cargo.toml
```toml
[package]
name = "esp-websocket"
version = "0.1.0"
authors = ["Author"]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]

default = ["std", "hal", "esp-idf-sys/native"]

pio = ["esp-idf-sys/pio"]
all = ["std", "nightly", "experimental", "embassy"]
hal = ["esp-idf-hal", "embedded-svc", "esp-idf-svc"]
std = [
    "alloc",
    "esp-idf-sys/std",
    "esp-idf-sys/binstart",
    "embedded-svc?/std",
    "esp-idf-hal?/std",
    "esp-idf-svc?/std",
]
alloc = ["embedded-svc?/alloc", "esp-idf-hal?/alloc", "esp-idf-svc?/alloc"]
nightly = [
    "embedded-svc?/nightly",
    "esp-idf-svc?/nightly",
] # Future: "esp-idf-hal?/nightly"
experimental = ["embedded-svc?/experimental", "esp-idf-svc?/experimental"]
embassy = [
    "esp-idf-hal?/embassy-sync",
    "esp-idf-hal?/critical-section",
    "esp-idf-hal?/edge-executor",
    "esp-idf-svc?/embassy-time-driver",
    "esp-idf-svc?/embassy-time-isr-queue",
]

[dependencies]
log = { version = "0.4.17", default-features = false }
esp-idf-sys = { version = "0.33", default-features = false }
esp-idf-hal = { version = "0.41", optional = true, default-features = false, features = [
    "esp-idf-sys",
] }
esp-idf-svc = { version = "0.46", optional = true, default-features = true }
embedded-svc = { version = "0.25", optional = true, default-features = false }
embedded-hal = { version = "0.2.7", optional = false, default-features = false }
[build-dependencies]
embuild = "0.31.2"
```

Is there a feature I have to enable somewhere?

ivmarkov, 2023-09-20:
https://github.com/ivmarkov/ruwm/blob/master/sdkconfig.defaults#L8

closed 2023-09-20 by ivmarkov

RoyalFoxy, 2024-08-17:
@fabracht have you fixed this? I'm currently also having this issue

I found out that there is a module called `ws` in the server.rs file, the same file the EspHttpServer is coming from but that module is only enabled if the custom compiler flag `esp_idf_httpd_ws_support` is set but I couldn't get it to work anyways

RoyalFoxy, 2024-08-17:
I could now get it to work by adding `CONFIG_HTTPD_WS_SUPPORT=y` to my `sdkconfig.defaults` but rust-analyzer still seems to think it's not available...

ivmarkov, 2024-08-18:
> I could now get it to work by adding `CONFIG_HTTPD_WS_SUPPORT=y` to my `sdkconfig.defaults` but rust-analyzer still seems to think it's not available...

You have to inspect in `vscode` how `cargo` is ran by `rust-analyzer`. Perhaps it is running a slightly different command from just `cargo check`.

For me, `rust-analyzer` is configured to do exactly `cargo check` (modulo output and `--all-features`) in the crate root and thus grabs whatever you have set in `sdkconfig.defaults`.

RoyalFoxy, 2024-08-19:
@ivmarkov Thank you for your quick response! Could you maybe show me what settings you have applied exactly? I tried setting these but it doesn't seem to work
```json
{
  "rust-analyzer.rustc.source": "discover",
  "rust-analyzer.cargo.targetDir": true,
  "rust-analyzer.cargo.autoreload": true,
  "rust-analyzer.server.path": "/Users/royalfoxy/.rustup/toolchains/stable-aarch64-apple-darwin/bin/rust-analyzer",

  "rust-analyzer.cargo.allTargets": true, // <- New
  "rust-analyzer.check.allTargets": true, // <- New
  "rust-analyzer.cargo.features": "all", // <- New
  "rust-analyzer.check.features": "all" // <- New
}
```