esp-rs/esp-idf-svc #301: OTA update API changes are hard to upgrade to

dberlin, 2023-09-27:
The new OTA update API, which separates out initiating updates and gives more control over tracking their progress, is nice in theory.

But it seems fairly hard to upgrade to/use in the current state because of some (IMHO sharp edges) . It would be good to know if this is a bug or a feature :)

If it's a feature, i'll just make my own, if it's a bug, happy to file more detailed reports.

For reference.I've attached a bluetooth OTA that was working fine in the old world (slot based) and is now much harder to make work.  
[bluetooth_ota.rs.txt](https://github.com/esp-rs/esp-idf-svc/files/12741473/bluetooth_ota.rs.txt)

I've also created a gist with the diff of the old/new here:
https://gist.github.com/dberlin/159ca432123870b46b3713519af8b97f

The only difference between the old and new code is that the old code used get_update on the updater, and the new code has to store the update struct given by initiate_update.
It's about a 15 line code diff from old to new.

Otherwise, it's a simple updater.  It does nothing crazy.  The main goal is to not have to buffer the entire update in memory, etc.
It would obviously work in a synchronous loop that did nothing but receive packets, and the lifetime of the updater pieces were obviously scoped to that single function.

This code, which does not work as-is, for a number of reasons:

1. The lifetime of what is returned by initiate_update is anonymous, despite it really having a static lifetime.
This causes rust to be very unsure how long it will ever live, and makes it hard to store and use in a struct.
It does not understand that that the initiate_update return lives longer than process_rx in this code.
You can't work around this by making just the current updater a global static  because the types are not marked Send/Sync.
It could probably be worked around by forcing BluetoothOTA struct to be a global static (since i think my unsafe marking of send/sync will override the fields not all being send/sync), but IMHO, it seems better to properly mark the lifetime of the initiate_update result as static.

2. The EspOtaUpdate struct, despite being just some pointers, isn't marked copy.  This means that attempts to use it in an Option, as you see here, are difficult because unwrapping the option requires moving the struct, which rules out using mutable references as you normally would.
IE self.current_updater.unwrap().complete() is invalid if self is just a mutable reference, because it requires moving the value out of the option.
Any attempt to use the option value (matching, etc) will fail in the same way, ie  "if let Some(updater) = self.current_updater"
This can be mostly worked around with takes around the updater use, but it's still ugly for a structure that should (imho), be Copy. 

3. There is a similar issue with the EspOtaUpdateFinished struct.

It would be really nice (and makes this code work as is) if 

1. The lifetime of the result of initiate_update was static instead of anonymous. Working around this requires hacks like global structs that are force-marked send/sync.
As mentioned, right now you can't even make a global static to store the current updater to work around it because the underlying types are not send/sync.  You have to use interior mutability tricks.

2. It would be really nice to mark the structs as copy - any safety issues that exist do not come from them being copy or not :)
That avoids having to do takes around them, which seems like an antipattern for options.

There are other ways to solve all of this, of course.

In practice, it all stems from the fact that EspOta used to encapsulate the entire OTA process fairly directly, so as you can see you could just call get_update to see whether you had a current slot or not. 
That made lifetime management easier, and you didn't have to store anything but an opaque updater instance.
That seems the opposite direction you want to go (and i happen to liek the new way), but i mention it for completeness of cataloging solutions.

For now i have local patches to esp-idf-svc to mark the initiate_update result as static lifetime, and make the OtaUpdate* structs copy, which makes this updater compile/work.

Like I said, it would be good to get a view on the above - if what i'm hitting is sharp edges, great, happy to help file them down.
If it's intentional and you don't want updaters that look like this, i'll go my own way :)

ivmarkov, 2023-10-01:
@dberlin I need to dig deeper in your concrete use-case (and understand deeper from where your difficulties are coming), but let me clarify a couple of things: 
* The reason why `EspOtaUpdate` is lifetimed is so that when you get an `EspOtaUpdate` instance from `initiate_update`, the `EspOta` instance is essentially **locked** for the duration of the update, as the returned lifetime is the lifetime of the `&mut self` `EspOta` instance of `initiate_update`. This prevents you from having multiple active updates hanging around, which I'm unsure would work. 
* The reasoning why `EspOtaUpdate` is not `Copy` (or `Clone`) - is similar - at any point in time, you are supposed to have a **single** update going on. `Copy` (or `Clone`) would break that invariant.

ivmarkov, 2023-10-01:
A few more comments.

> 1. The lifetime of what is returned by initiate_update is anonymous, despite it really having a static lifetime.

I'm unsure how you are coming to the conclusion that "what is returned by `initiate_update`" has a static lifetime. The fact that there is nothing annotated with a lifetime inside it (just `PhantomData`) does not mean that what is returned has a static lifetime. At the very least, **a raw pointer** is returned, which means that we have to (a) think about the lifetime of the pointer itself and more importantly (b) think how it related to the parent object (`EspOta`) which returns that pointer and with the API. Or to give a few examples:
* I don't think it would be OK for `EspOtaUpdate` to outlive `EspOta`, which means that `EspOtaUpdate` should be bound by the liffetime of your mutable reference to `EspOta`
* I don't think it is OK to have multiple `EspOtaUpdate` instances around, so as per above, the reference to `EspOta` should be `&mut` and not `&`
 
>    This causes rust to be very unsure how long it will ever live, and makes it hard to store and use in a struct.

It lives as long as the `&mut ref` to `EspOta` lives, and your analysis about the Rust compiler being unsure is a bit misleading. The actual reason why you can't store it in the struct is because **in that very same struct** you have the `EspOta` update instance. So you are creating a self-referential struct, which is hard to impossible to do in Rust. Think what happens if you **move** this struct around. From the point of view of the Rust compiler, that would mean the `EspOta` instance inside will change its address, which - in turn - will invalidate the potential hidden address to `EspOta` that `EspOtaUpdate` *might* be keeping to `EspOta` (as its lifetime is essentially the lifetime of `&mut EspOta`). 

>    It does not understand that that the initiate_update return lives longer than process_rx in this code, and so the attempt to store the result in self.current_updater fails because it can't prove that it is longer than the function:
> 
> ```
> error: lifetime may not live long enough

This is also a bit imprecise. Really, the key is my previous paragraph that you are creating a self-referential struct. As to whether the new API is convenient or not, that's a whole other story. The old API allowed your pattern because `EspOtaUpdate` was returned by `&mut ref` and not as an instance, but it had other inconveniences, like `abort` and `complete` not taking `self` but `&mut self`. So you could - for example - call `complete` and then `abort`, which would return a runtime error.

The new API moves some of these constraints from runtime errors to compile time errors, which is definitely a benefit.

> You can't work around this by making just the current updater a global static because the types are not marked Send/Sync. 

We are digressing here because - again - see my reasoning for the new API above, but let me address this specific comment: that's because you are assuming that a static lifetime necessarily implies "global" which necessarily implies Send+Sync. The real story is finer grained that that:
* You can have global mut without Send+Sync but yeah, you need `unsafe` to use it
* You can have [static lifetime without global, which is safe](https://github.com/embassy-rs/static-cell)

> It could probably be worked around by forcing BluetoothOTA struct to be a global static (since i think my unsafe marking of send/sync will override the fields not all being send/sync), but IMHO, it seems better to properly mark the lifetime of the initiate_update result as static.

As per above, I'm not sure that would be a win.

I absolutely acknowledge a few things though: 
* the new API WILL require you to rethink your code
* the new API WILL be a bit harder to use for Rust newcomers, as they really need to think about lifetimes, mutable aliasing and so on
* the new API is (subjectively!) better as it pushes back some run-time errors to compile time. By the way, the new API is an API which used to be there before the current one and I removed at some point in time as I was not sure when Rust GATs would be stabilized. 9 months later GAT is in Rust stable though, hence the reason I restored it. (If that's not clear, it is not possible to implement the Ota `embedded-svc` traits around the new API without GATs.)

As for a concrete advise how to adjust your code to the new API, you need nested loops/ifs so to say, where you currently have a single explicit state machine.

> 2. The EspOtaUpdate struct, despite being just some pointers, isn't marked copy.  This means that attempts to use it in an Option, as you see here, are difficult because unwrapping the option requires moving the struct (since they can't be copiedd), which rules out using mutable references as you normally would.
>    IE self.current_updater.unwrap().complete() is invalid if self is just a mutable reference, because it requires moving the value out of the option.

Digressing here again (all this is Rust knowledge/tricks) but `self.current_updater.as_mut().unwrap()` would HAVE worked if `complete` was `complete(&mut self) and not `complete(self)` but then again, my earlier paragraph. :)

ivmarkov, 2023-12-21:
No feedback, closing.

closed 2023-12-21 by ivmarkov

dberlin, 2024-01-20:
FWIW - i gave no feedback because i gave up on using esp-idf-svc's OTA support.  The new API is too hard to use in practice in the  real world apps I have, so i forked the old API into my  projects.

I would argue very strongly the new API is not better enough to break everyone, and honestly, not better at all.

But you can feel free to ignore me - i'm also giving up on ESP rust for future projects.  The constant churn in the name of "better" that is happening is not useful and indicative of projects that have no useful feedback loop with their users.

Technically better is rarely a useful measure for products.  Does it help users achieve their goals?  

In this case, no - As you say,  it makes it harder for newcomers and harder for existing folks.

I'll check in a few years when maybe any of this is mildly stable and usable to users.

ostenning, 2024-03-28:
@ivmarkov I think an example would be helpful, or some documentation