esp-rs/esp-idf-svc #302: mqtt: build fails due to type and function signature mismatch

mbuesch, 2023-09-27:
Hi,

since yesterday the build fails in my project with these messages:

```
error[E0063]: missing field `outbox` in initializer of `esp_idf_sys::esp_mqtt_client_config_t`
   --> /.../rust/cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.46.2/src/mqtt/client.rs:212:26
    |
212 |         let mut c_conf = esp_mqtt_client_config_t {
    |                          ^^^^^^^^^^^^^^^^^^^^^^^^ missing `outbox`

error[E0425]: cannot find function `esp_mqtt_client_subscribe` in this scope
     --> /.../rust/cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.46.2/src/mqtt/client.rs:484:13
      |
484   |               esp_mqtt_client_subscribe(self.raw_client, c_topic.as_ptr(), qos as _)
      |               ^^^^^^^^^^^^^^^^^^^^^^^^^ help: a function with a similar name exists: `esp_mqtt_client_unsubscribe`
      |
     ::: /.../target/xtensa-esp32-espidf/debug/build/esp-idf-sys-af990a2ae26b7e94/out/bindings.rs:45553:5
      |
45553 | /     pub fn esp_mqtt_client_unsubscribe(
45554 | |         client: esp_mqtt_client_handle_t,
45555 | |         topic: *const ::core::ffi::c_char,
45556 | |     ) -> ::core::ffi::c_int;
      | |___________________________- similarly named function `esp_mqtt_client_unsubscribe` defined here

Some errors have detailed explanations: E0063, E0425.
For more information about an error, try `rustc --explain E0063`.
error: could not compile `esp-idf-svc` (lib) due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
```

I'm not sure why it suddenly started failing. I even tried to roll back some changes and Cargo.lock updates on my side, but that didn't help.
Any idea what's going on? Why are the mqtt types and functions suddenly different?

I don't even use mqtt in my project. Is there a workaround, e.g. by disabling mqtt completely?

mbuesch, 2023-09-27:
Ok, I just found out that this seems to originate from esp-idf 5.1.
If I switch my project back to 5.0 the problem goes away.

```
--- a/.cargo/config.toml
+++ b/.cargo/config.toml
 [env]
 ESP_IDF_SYS_ROOT_CRATE = { value = "main" }
-ESP_IDF_VERSION = { value = "branch:release/v5.1" }
+ESP_IDF_VERSION = { value = "branch:release/v5.0" }
```

However, 5.1 worked just fine before. I was on 5.1 for three weeks with no problems.
Was there an incompatible API change in a recent 5.1.x release?

ivmarkov, 2023-09-27:
Yes.
 It is compatible on C level but not on Rust level. Switch to the tagged esp idf releases instead, I.e. `v5.1.x`.

mbuesch, 2023-09-27:
Yeah, interesting. If I pin the project to the latest 5.1 release it also works again:
```
-ESP_IDF_VERSION = { value = "branch:release/v5.0" }
+ESP_IDF_VERSION = { value = "tag:v5.1.1" }
```
So there seems to be a breaking change in the branch that has not been released, yet.

ivmarkov, 2023-09-27:
`release/vx.y` should not really be used, as I recently found out. That's because it might have backwards incompatible changes (from. Rust POV)  with e.g. tag `v5.1.1` even if it's version is still listed as v5.1.1. Only when the new tag `v5.1.2` is tagged on this branch, the branch version itself is raised to `v5.1.2`.

closed 2023-09-27 by ivmarkov

mbuesch, 2023-09-27:
Hm yeah, we can close this issue for now.
But doesn't this require a change in esp-idf-svc once 5.1.2 is released?

ivmarkov, 2023-09-27:
The change is already incorporated. It is just that the version-based branching does not work for the `release/vx.y` branches that we should stop using.

ivmarkov, 2023-09-27:
Yes. 

Also that while esp idf patch level changes are supposed to be source-level backwards compatible (as in 5.1.1 vs 5.1.2) in C, they are unfortunately not always so in Rust.

The "backwards compatible" change which is not such in Rust is that they renamed the `subscribe` C fn to `subscribe_single` and they did `#define subscribe subscribe_single`. But Rust bindgen does not see this C macro, so from Rust POV the `subscribe` function simply disappeared.

ivmarkov, 2023-09-27:
And yes, we'll have to ask the esp idf C developers soon-ish if they can try to avoid some C backwards compatibility tricks that do not work for Rust. Like ab(using) the C preprocessor. Or using `static inline` in C headers.

@igrr JFYI.

mbuesch, 2023-09-27:
Got it.
Thanks a lot for explaining this in detail.

igrr, 2023-09-27:
cc @euripedesrocha