esp-rs/esp-idf-svc #305: W5500 never reports as up

sampaioletti, 2023-10-03:
When using the W5500 over SPI on an esp32-s3 It never reports is up even though it responds to pings and i'm able to establish a tcp connection
```rust
    let spi_driver = spi::SpiDriver::new(
        peripherals.spi3,
        pins.gpio12,
        pins.gpio13,
        pins.gpio11.into(),
        &DriverConfig {
            dma: spi::Dma::Auto(4096),
            ..Default::default()
        },
    )?;
   let eth_driver = EthDriver::new(
        spi_driver,
        pins.gpio9,
        Some(pins.gpio10),
        None::<AnyOutputPin>,
        SpiEthChipset::W5500,
        esp_idf_hal::units::Hertz(40 * 1000 * 1000),
        Some(&[0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED]),
        None,
        sys_loop.clone(),
    )?;
    let mut config = NetifConfiguration::eth_default_client();
    config.ip_configuration =
        ipv4::Configuration::Client(ipv4::ClientConfiguration::Fixed(ClientSettings {
            ip: ipv4::Ipv4Addr::new(192, 168, 0, 5),
            subnet: "0.0.0.0/24".parse().unwrap(),
            dns: None,
            secondary_dns: None,
        }));
    let netif = EspNetif::new_with_conf(&config)?;
    let eth = EspEth::wrap_all(eth_driver, netif)?;

    info!("Eth created");
    let mut eth = BlockingEth::wrap(eth, sys_loop.clone())?;
    info!("Starting eth...");

    eth.start()?;

    info!("Waiting for up...");

    eth.wait_netif_up()?; //hangs indefinietly

    let ip_info = eth.eth().netif().get_ip_info()?;

    info!(
        "Eth DHCP info: {} {} {:?}",
        eth.is_started()?,
        eth.is_up()?, //always false
        ip_info
    );
```
I've tried finding out if this is W5500 behavior or if I'm doing something wrong but haven't found anything yet.

Thank you for the help.

closed 2023-10-04 by sampaioletti

kali-muon, 2023-12-06:
> Dug into this further, and it appears to be caused by my static IP configuration. If I run it using the defaults (DHCP) it comes up after dhcp success...so this may be either a bug or intended behavior of the idf w5500 driver, i'll look into the source and see if it only reports up on dhcp success.
> 
> It took me a minute to figure some of this out (could just be me). would you like a PR for an eth-spi sample?

This is kind of off-topic: I'm actually trying to use the w5500 on the esp32s3 as well. The problem is, I cannot for the life of me get it to compile when I have `use esp_idf_svc::eth;` at the top of my `main.rs`. Initially, `esp_idf_svc::eth` failed to import because `esp_idf_eth_spi_ethernet_w5500` wasn't set, causing `mod eth` in this library to be deactivated. After adding that to my rustflags in `.cargo/config.toml` as `--cfg esp_idf_eth_spi_ethernet_w5500`, compilation got further but still failed. The error is as follows:
```
   Compiling embedded-svc v0.26.4
error[E0422]: cannot find struct, variant or union type `eth_w5500_config_t` in this scope
     --> /home/[REDACTED]/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.47.3/src/eth.rs:491:33
      |
491   |                 let w5500_cfg = eth_w5500_config_t {
      |                                 ^^^^^^^^^^^^^^^^^^ help: a struct with a similar name exists: `eth_mac_config_t`
      |
```
I would be eternally grateful if you could shine some light on how to get this to work.

ivmarkov, 2023-12-06:
> This is kind of off-topic: I'm actually trying to use the w5500 on the esp32s3 as well. The problem is, I cannot for the life of me get it to compile when I have `use esp_idf_svc::eth;` at the top [...]

You are holding it wrong. You are not supposed to add arbitrary rustc flags to your settings (only exception is the `espidf_time64` flag).

You need to modify your **`sdkconfig.defaults`** to enable *the relevant ESP IDF component(s)*. After that, the relevant rustc flags will be automatically set - for upstream crates - by `esp-idf-sys` build script itself. So that upstream crates can then disable pure Rust code that relies on the presence of these ESP IDF components, if these are not enabled/compiled-in.

Look [here](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/sdkconfig.defaults#L16).