esp-rs/esp-idf-svc #307: ESP32C6 Wifi APIs not available on v5.1/v5.1.1

sjames, 2023-10-18:
Hi,

I'm using ```cargo generate esp-rs/esp-idf-template cargo``` to generate a new esp32 project.  I'm using an esp32c6 module, so I use IDF v5.1.1 .

I find that the Rust Wifi APIs defined in src/wifi.rs are not available.  The wifi module is gated by the following configurations.

```
#[cfg(all(
    not(esp32h2),
    feature = "alloc",
    esp_idf_comp_esp_wifi_enabled,
    esp_idf_comp_esp_event_enabled,
))]
pub mod wifi;
```

I couldn't figure out how these configurations are set.  Does anyone have Wifi working with Rust on ESP32C6?

Regards,
Sojan

sjames, 2023-10-18:
Answering this myself.

It turns out that the esp-idf-template does not configure the MCU that needs to build, and the default mcu is the esp32h2 that does not have the Wifi.

I fixed this by setting the mcu in the .cargo/config.toml

```
[env]
# Note: these variables are not used when using pio builder (`cargo build --features pio`)
ESP_IDF_VERSION = "v5.1.1"
MCU = "esp32c6"

```

ivmarkov, 2023-10-18:
I'm in the process of updating the `esp-idf-template` crate to work with the newly released `esp-idf-*` crates.
Thanks for this feedback - makes sense! Just updated the `.cargo/config.toml` with the MCU parameter: https://github.com/esp-rs/esp-idf-template/pull/159/commits/6d71a635992bb8a5ffddc679c4f949e366c9f62d

A new `esp-idf-template` will be pushed till the end of the week.

ivmarkov, 2023-10-18:
Also this: https://github.com/esp-rs/esp-idf-template/pull/159/commits/01b55d3d2b1d9e20bde9626c96a6b0c56fe28700

ivmarkov, 2023-10-19:
To be fixed once this is merged: https://github.com/esp-rs/esp-idf-template/pull/159

closed 2023-10-19 by ivmarkov