esp-rs/esp-idf-svc #312: Fails to compile with certificate bundles disabled

thetek42, 2023-11-06:
When compiling a project with `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE=n`, `esp-idf-svc` fails to compile with the following error:

```
error[E0432]: unresolved imports `crate::sys::ESP_TLS_ERR_SSL_WANT_READ`, `crate::sys::ESP_TLS_ERR_SSL_WANT_WRITE`
   --> .cargo\git\checkouts\esp-idf-svc-a28457b0e32c6283\cdfc601\src\tls.rs:128:55
    |
128 |             self, EspError, ESP_ERR_NO_MEM, ESP_FAIL, ESP_TLS_ERR_SSL_WANT_READ,
    |                                                       ^^^^^^^^^^^^^^^^^^^^^^^^^ no `ESP_TLS_ERR_SSL_WANT_READ` in `sys`
129 |             ESP_TLS_ERR_SSL_WANT_WRITE, EWOULDBLOCK,
    |             ^^^^^^^^^^^^^^^^^^^^^^^^^^ no `ESP_TLS_ERR_SSL_WANT_WRITE` in `sys`
    |
help: a similar name exists in the module
    |
128 |             self, EspError, ESP_ERR_NO_MEM, ESP_FAIL, MBEDTLS_ERR_SSL_WANT_READ,
    |                                                       ~~~~~~~~~~~~~~~~~~~~~~~~~
help: a similar name exists in the module
    |
129 |             MBEDTLS_ERR_SSL_WANT_WRITE, EWOULDBLOCK,
    |             ~~~~~~~~~~~~~~~~~~~~~~~~~~

error[E0435]: attempt to use a non-constant value in a constant
   --> .cargo\git\checkouts\esp-idf-svc-a28457b0e32c6283\cdfc601\src\tls.rs:455:23
    |
454 |                 ESP_TLS_ERR_SSL_WANT_READ => Err(EspError::from_infallible::<
    |                 ------------------------- this would need to be a `const`
455 |                     { ESP_TLS_ERR_SSL_WANT_READ as i32 },
    |                       ^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0435]: attempt to use a non-constant value in a constant
   --> .cargo\git\checkouts\esp-idf-svc-a28457b0e32c6283\cdfc601\src\tls.rs:458:23
    |
457 |                 ESP_TLS_ERR_SSL_WANT_WRITE => Err(EspError::from_infallible::<
    |                 -------------------------- this would need to be a `const`
458 |                     { ESP_TLS_ERR_SSL_WANT_WRITE as i32 },
    |                       ^^^^^^^^^^^^^^^^^^^^^^^^^^
```

In esp-idf, these constants are defined [here](https://github.com/espressif/esp-idf/blob/b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e/components/esp-tls/esp_tls_errors.h#L63-L66).

Vollbrecht, 2023-11-06:
First of what are you trying to archive? And second what do you actually expected to happen here?

ivmarkov, 2023-11-06:
> I am trying to disable the [Certificate Bundle API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_crt_bundle.html), since it might not be needed for my use case.
> 
> I expect esp-idf-svc to compile properly without errors.

That's fair. I'll fix it shortly.

Vollbrecht, 2023-11-06:
the question is, do you mean **not using tls** itself in your project or **not using mbedtls** specific ? Our tls module probably should be disabled if we set something like CONFIG_MBEDTLS_TLS_DISABLED=y in sdkconfig ( this currently does not work). Or do you just want to use anothere impl like wolfssl ?

ivmarkov, 2023-11-06:
@Vollbrecht - there is (or there should be) **nothing** in the `tls` module which is mbedtls specific. Hence it should "just work" with wolfssl as well, or if both are disabled - just not be compiled at all.

Hence why I consider the current compilation issues a bug.

ivmarkov, 2023-11-06:
Also the reporter is disabling just the certificate bundle, which should not disable (I think) neither mbedtls, nor wolfssl.

thetek42, 2023-11-06:
Correct. The goal is not to disable TLS entirely; only certificate bundles should be disabled.

What I find peculiar is that the constants that cause this problem do not depend on `CONFIG_MBEDTLS_CERTIFICATE_BUNDLE` (see [this](https://github.com/espressif/esp-idf/blob/b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e/components/esp-tls/esp_tls_errors.h#L63-L66)).

ivmarkov, 2023-11-07:
@thetek42 Just to mention that I can repro the problem, but I'm a bit at a loss how exactly that could even happen... I'm beginning to suspect some sort of bindgen issue/bug here...

ivmarkov, 2023-11-07:
OK found the root cause - a bug in ESP IDF. Will follow up shortly with more details once I report it and implement a workaround...

Vollbrecht, 2023-11-07:
even if it is not the core problem, i was not able to disable the embedtls support via sdkconfigs in my short test, when i run with `cargo rustc -- --print cfg` it was always activated no matter what.

ivmarkov, 2023-11-07:
> even if it is not the core problem, i was not able to disable the embedtls support for anything via sdkconfigs in my short test, when i run with `cargo rustc -- --print cfg` it was always activated no matter what, so bindgen was generated for it and after that it got picked up by svc

Unless you show *exactly how* you disable mbedtls support, I can't comment what is going on.

ivmarkov, 2023-11-07:
And then again, *the user is NOT disabling mbedtls* - he is disabling the certificate bundle, which leads to compilation issues, while it should **not**.

ivmarkov, 2023-11-07:
@thetek42:
- ESP IDF issue: https://github.com/espressif/esp-idf/issues/12541
- Workaround (in `esp-idf-sys`, I'll release a ptach shortly): https://github.com/esp-rs/esp-idf-sys/commit/2067b8cc518339694e936fdb405ecd15a5e33b1d

ivmarkov, 2023-11-07:
New release of `esp-idf-sys` should be available shortly (Publish action ongoing).
Just do `cargo update` in the root of your binary crate.

closed 2023-11-07 by ivmarkov