esp-rs/esp-idf-svc #313: esp-idf-svc fails to compile in idf-rust:esp32s3_1.73.0.1 docker image

Joku1806, 2023-11-08:
Hello,

I followed the install instructions for the std toolchain from [here](https://esp-rs.github.io/book/introduction.html) and generated a project with cargo generate esp-rs/esp-idf-template cargo.

Locally, everything builds fine and a .embuild directory is created automatically, but in [this Docker Image](https://hub.docker.com/layers/espressif/idf-rust/esp32s3_1.73.0.1/images/sha256-43e6cf001b767190ed6fdc39735a972807639db9601a70b770243cb1ad50c027?context=explore) this does not happen.

I don't know why this happens, as far as I can tell from the layers all required dependencies are installed and I also sourced `$HOME/export-esp.sh` before building.

Here are the CI logs:

```
Running with gitlab-runner 16.5.0 (853330f9)
  on esp32 vxqhoGEM1, system ID: s_089face47da2
Preparing the "docker" executor 00:04
Using Docker executor with image espressif/idf-rust:esp32s3_1.73.0.1 ...
Authenticating with credentials from $DOCKER_AUTH_CONFIG
Pulling docker image espressif/idf-rust:esp32s3_1.73.0.1 ...
Using docker image sha256:6313f85adb406f1e7a99293fdc4f8e8ffbf2726b1708d9b133f39c378f0d6859 for espressif/idf-rust:esp32s3_1.73.0.1 with digest espressif/idf-rust@sha256:e5968bf4b848c73cdc0b6469d21e88cc5c2ac9dd7378c6161fa324225f8a069f ...
Preparing environment 00:02
Running on runner-vxqhogem1-project-30397-concurrent-0 via homebox...
Getting source from Git repository 00:02
Fetching changes with git depth set to 20...
Reinitialized existing Git repository in /builds/tkn/teaching/pj2023w/practical-network-coding-with-esp32/.git/
Checking out 101a7b08 as detached HEAD (ref is 8-create-test-infrastructure-and-run-tests-in-ci)...
Removing Cargo.lock
Removing target/
Skipping Git submodules setup
Executing "step_script" stage of the job script 00:35
Using docker image sha256:6313f85adb406f1e7a99293fdc4f8e8ffbf2726b1708d9b133f39c378f0d6859 for espressif/idf-rust:esp32s3_1.73.0.1 with digest espressif/idf-rust@sha256:e5968bf4b848c73cdc0b6469d21e88cc5c2ac9dd7378c6161fa324225f8a069f ...
$ source /home/esp/export-esp.sh
$ cargo build
    Updating crates.io index
 Downloading crates ...
  Downloaded log v0.4.20
  Downloaded thread_local v1.1.7
  Downloaded strum v0.24.1
  Downloaded glob v0.3.1
  Downloaded errno v0.3.6
  Downloaded pin-utils v0.1.0
  Downloaded embedded-hal-nb v1.0.0-rc.1
  Downloaded stable_deref_trait v1.2.0
  Downloaded futures-task v0.3.29
  Downloaded cexpr v0.6.0
  Downloaded peeking_take_while v0.1.2
  Downloaded const_format_proc_macros v0.2.32
  Downloaded toml_edit v0.20.7
  Downloaded ident_case v1.0.1
  Downloaded thiserror-impl v1.0.50
  Downloaded normpath v1.1.1
  Downloaded embassy-futures v0.1.0
  Downloaded thiserror v1.0.50
  Downloaded fs_at v0.1.10
  Downloaded itoa v1.0.9
  Downloaded rustc_version v0.4.0
  Downloaded defmt-macros v0.3.6
  Downloaded unicode-xid v0.2.4
  Downloaded uncased v0.9.9
  Downloaded memchr v2.6.4
  Downloaded enumset v1.1.3
  Downloaded hashbrown v0.14.2
  Downloaded which v4.4.2
  Downloaded strum_macros v0.24.3
  Downloaded futures-util v0.3.29
  Downloaded serde v1.0.192
  Downloaded winnow v0.5.19
  Downloaded rustix v0.38.21
  Downloaded bstr v1.7.0
  Downloaded linux-raw-sys v0.4.10
  Downloaded regex-automata v0.4.3
  Downloaded syn v2.0.39
[... 311 lines cut ...]
    13: _start
```

After the failed compilation, there is no `.embuild` directory:

```
esp@48abec0f69f5:~/practical-network-coding-with-esp32$ ls -hartl
total 104K
-rw-r--r-- 1 esp esp  521 Nov  8 00:14 .editorconfig
drwxr-xr-x 2 esp esp 4.0K Nov  8 00:15 src
-rw-r--r-- 1 esp esp  465 Nov  8 00:15 sdkconfig.defaults
-rw-r--r-- 1 esp esp   28 Nov  8 00:15 rust-toolchain.toml
-rw-r--r-- 1 esp esp   53 Nov  8 00:15 build.rs
-rw-r--r-- 1 esp esp  841 Nov  8 00:15 README.md
-rw-r--r-- 1 esp esp  262 Nov  8 00:15 .gitlab-ci.yml
-rw-r--r-- 1 esp esp   48 Nov  8 00:15 .gitignore
drwx------ 1 esp esp 4.0K Nov  8 00:15 ..
drwxr-xr-x 8 esp esp 4.0K Nov  8 00:15 .git
-rw-r--r-- 1 esp esp  47K Nov  8 00:16 Cargo.lock
-rw-r--r-- 1 esp esp  859 Nov  8 00:26 Cargo.toml
drwxr-xr-x 5 esp esp 4.0K Nov  8 00:27 .
drwxr-xr-x 3 esp esp 4.0K Nov  8 00:27 target
```

I also tried to manually add embuild using cargo add, but that did not help. Other than that, I am not sure what could be the problem.

ivmarkov, 2023-11-08:
Build fails with:
`Error: Unsupported target 'x86_64-unknown-linux-gnu'`,

I.e. you are trying to build `esp-idf-sys` for your host target, which is impossible. You should build with a target like `xtensa-esp32-espidf` or suchlike.

These targets are usually configured in `.cargo/config.toml` file inside your binary crate, but I actually don't see the `.cargo` directory in your project layout?

Joku1806, 2023-11-08:
Oh yeah, that seems to work.
I am very new to rust development and saw that `.cargo` is a dotfile and thought that everything build-related is defined in the `Cargo.toml`, so I immediately added it to the `.gitignore` without checking it first.

Sorry for the confusion, going to close this!

closed 2023-11-08 by Joku1806