esp-rs/esp-idf-svc #314: How to get a list of access points?

kekonn, 2023-11-11:
I'm new to ESP32's and as an exercise I am trying to simply print out a list of access points the wifi adapter can see.

[Code here](https://gist.github.com/kekonn/633eb88f40f2ce30a6d06dd4a0614d9a).

This is the output:

```
[2023-11-11T20:10:49Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
0x3fff0030 - g_ticks_per_us_app
    at ??:??
load:0x40078000,len:15576
0x40078000 - __udivmoddi4
    at ??:??
load:0x40080400,len:4
0x40080400 - _init
    at ??:??
ho 8 tail 4 room 4
load:0x40080404,len:3876
0x40080404 - _iram_text_start
    at ??:??
entry 0x4008064c
0x4008064c - _iram_text_start
    at ??:??
I (31) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (31) boot: compile time Jun  7 2023 07:48:23
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v3.1
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 003f0000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=35e78h (220792) map
I (186) esp_image: segment 1: paddr=00045ea0 vaddr=3ffb0000 size=035ach ( 13740) load
I (192) esp_image: segment 2: paddr=00049454 vaddr=40080000 size=06bc4h ( 27588) load
I (203) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=b3b4ch (736076) map
I (469) esp_image: segment 4: paddr=00103b74 vaddr=40086bc4 size=0da30h ( 55856) load
I (503) boot: Loaded app from partition at offset 0x10000
I (503) boot: Disabling RNG early entropy source...
I (515) cpu_start: Multicore app
I (515) cpu_start: Pro cpu up.
I (515) cpu_start: Starting app cpu, entry point is 0x4008212c
0x4008212c - call_start_cpu1
    at /mnt/backup/source/esp32/hello-esp/.embuild/espressif/esp-idf/v5.1.1/components/esp_system/port/cpu_start.c:154
I (0) cpu_start: App cpu up.
I (533) cpu_start: Pro cpu start user code
I (533) cpu_start: cpu freq: 160000000 Hz
I (533) cpu_start: Application information:
I (538) cpu_start: Project name:     libespidf
I (543) cpu_start: App version:      1
I (547) cpu_start: Compile time:     Nov 11 2023 16:28:11
I (553) cpu_start: ELF file SHA256:  0000000000000000...
I (559) cpu_start: ESP-IDF:          v5.1.1
I (564) cpu_start: Min chip rev:     v0.0
I (569) cpu_start: Max chip rev:     v3.99 
I (574) cpu_start: Chip rev:         v3.1
I (579) heap_init: Initializing. RAM available for dynamic allocation:
I (586) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (592) heap_init: At 3FFB7720 len 000288E0 (162 KiB): DRAM
I (598) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (604) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (611) heap_init: At 400945F4 len 0000BA0C (46 KiB): IRAM
I (618) spi_flash: detected chip: generic
I (622) spi_flash: flash io: dio
W (626) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (636) pcnt(legacy): legacy driver is deprecated, please migrate to `driver/pulse_cnt.h`
W (645) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (654) app_start: Starting scheduler on CPU0
I (659) app_start: Starting scheduler on CPU1
I (659) main_task: Started on CPU0
I (669) main_task: Calling app_main()
I (669) hello_esp: Setting up WiFi radio
I (699) wifi:wifi driver task: 3ffbf480, prio:23, stack:6656, core=0
I (699) wifi:wifi firmware version: ce9244d
I (699) wifi:wifi certification version: v7.0
I (699) wifi:config NVS flash: enabled
I (699) wifi:config nano formating: disabled
I (709) wifi:Init data frame dynamic rx buffer num: 32
I (709) wifi:Init management frame dynamic rx buffer num: 32
I (719) wifi:Init management short buffer num: 32
I (719) wifi:Init dynamic tx buffer num: 32
I (729) wifi:Init static rx buffer size: 1600
I (729) wifi:Init static rx buffer num: 10
I (729) wifi:Init dynamic rx buffer num: 32
I (739) wifi_init: rx ba win: 6
I (739) wifi_init: tcpip mbox: 32
I (739) wifi_init: udp mbox: 6
I (749) wifi_init: tcp mbox: 6
I (749) wifi_init: tcp tx win: 5744
I (759) wifi_init: tcp rx win: 5744
I (759) wifi_init: tcp mss: 1440
I (759) wifi_init: WiFi IRAM OP enabled
I (769) wifi_init: WiFi RX IRAM OP enabled
I (779) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (879) wifi:mode : softAP (64:b7:08:cc:bd:ad)
I (889) wifi:Total power save buffer number: 16
I (889) wifi:Init max length of beacon: 752/752
I (889) wifi:Init max length of beacon: 752/752
I (889) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.71.1
E (899) wifi:NAN WiFi stop
I (899) wifi:flush txq
I (909) wifi:stop sw txq
I (909) wifi:lmac stop hw txq
I (909) esp_idf_svc::wifi: EspWifi dropped
I (919) esp_idf_svc::netif: Dropped
I (919) esp_idf_svc::netif: Dropped
I (929) wifi:Deinit lldesc rx mblock:10
I (939) esp_idf_svc::nvs: NvsDefault dropped
I (939) esp_idf_svc::eventloop: System event loop dropped
Error: Error scanning for access points

Caused by:
    ESP_FAIL
I (939) main_task: Returned from app_main()
```

Can someone tell my why the Wifi is getting dropped and what I am doing wrong?

Vollbrecht, 2023-11-11:
you got an Error inside the wifi driver : `E (899) wifi:NAN WiFi stop` i am not sure about that but i think it is talking about the WiFi NAN (Neighbor Awareness Networking) protocol here. Do you have othere devices around you that might use this protocoll? To my understanding the esp's dont support this. All speculation here but it might be a client that try's to connect to the station you opend. 
To test it out if that is the case could you try just creating a raw [WfifiDriver](https://esp-rs.github.io/esp-idf-svc/esp_idf_svc/wifi/struct.WifiDriver.html) and only use sta mode , you can wrap it the same way in the BlockingWifi and then test it again?

kekonn, 2023-11-11:
>  Do you have other devices around you that might use this protocol?

I haven't even heard of that protocol before, so no idea. 

> To test it out if that is the case could you try just creating a raw [WfifiDriver](https://esp-rs.github.io/esp-idf-svc/esp_idf_svc/wifi/struct.WifiDriver.html) and only use sta mode

STA mode means connecting to an existing WiFi network right? I'm not trying to do that. I'm trying to print a list of visible wifi networks without doing any connection. Like the promiscuous mode from back in the Backtrack days.

Or is that not a supported use case?

Vollbrecht, 2023-11-11:
yes you dont need to connect to anything. what i want you to do is not opening an AP mode and STA mode at the same time. you even can start in None Mode.

Vollbrecht, 2023-11-11:
with the set_configuration method you can set it to None / Ap / sta /mixed . Please test it with None or Sta and run a scan

kekonn, 2023-11-11:
Setting it to None didn't work, but setting it to Client mode did the trick!

Thanks!

closed 2023-11-11 by kekonn

Vollbrecht, 2023-11-11:
In any case this error still is really wired. The AP should not crash if its get something send it's not compatible with ( if that's the case here) so we would still need to dick deeper.

Vollbrecht, 2023-11-11:
So it would be cool if you could sniff the packets you receive and check if you get something like a NAN package. I think this would be a bug in the upstream esp-idf impl

kekonn, 2023-11-11:
> In any case this error still is really wired. The AP should not crash if its get something send it's not compatible with ( if that's the case here) so we would still need to dick deeper.

If you want me to run some code for diagnost purposes, feel free to give me a snippet.

Vollbrecht, 2023-11-11:
If you could do one test where you configure the EspWifi ( so you have netif running) just in AP mode -  and just set a simple test ssid and pw and connect with your phone ore something and see if it still crashes?

Vollbrecht, 2023-11-11:
the only related problem on esp-idf i could find is https://github.com/espressif/esp-idf/issues/12473 though also a relative new issue

kekonn, 2023-11-11:
I used the following code:

```rust
fn wifi_ap() -> Result<()> {
    log::info!("Setting up WiFi radio");
    let sysloop = EspSystemEventLoop::take().with_context(|| "Error getting SysEventLoop handle")?;
    let peripherals = Peripherals::take()?;
    let nvs = EspDefaultNvsPartition::take().with_context(|| "Error retrieving default NVS partitions")?;
    let mut wifi = BlockingWifi::wrap(
        EspWifi::new(peripherals.modem, sysloop.clone(), Some(nvs))?,
        // WifiDriver::new(peripherals.modem, sysloop.clone(), Some(nvs))?, 
        sysloop)?;
    let wifi = wifi.wifi_mut();

    log::info!("Starting Wifi");

    wifi.set_configuration(&Configuration::AccessPoint( AccessPointConfiguration {
        ssid: "ESP32".into(),
        password: "AchtTekens".into(),
        auth_method: AuthMethod::WPA2Personal,
        ..Default::default()
    }))?;

    log::info!("Starting WiFi AP");
    wifi.start()?;
        
    loop {
        let netif = wifi.ap_netif();
        log::info!("AP Info: {} - {:?}", netif.get_name(), netif.get_ip_info()?);

        FreeRtos::delay_ms(1000);
    }
}
```

And that worked flawlessly. Could connect with my phone and it kept running:

```
I (26319) hello_esp: AP Info: ap1 - IpInfo { ip: 192.168.71.1, subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dns: Some(8.8.8.8), secondary_dns: Some(8.8.8.8) }
I (27319) hello_esp: AP Info: ap1 - IpInfo { ip: 192.168.71.1, subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dns: Some(8.8.8.8), secondary_dns: Some(8.8.8.8) }
I (27539) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (27539) wifi:station: ae:84:bc:02:88:d5 join, AID=1, bgn, 40U
I (27899) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.71.2
I (28279) wifi:<ba-add>idx:2 (ifx:1, ae:84:bc:02:88:d5), tid:0, ssn:17, winSize:64
I (28319) hello_esp: AP Info: ap1 - IpInfo { ip: 192.168.71.1, subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dns: Some(8.8.8.8), secondary_dns: Some(8.8.8.8) }
W (28569) wifi:[ADDBA]rx delba, code:1, delete tid:0
I (29319) hello_esp: AP Info: ap1 - IpInfo { ip: 192.168.71.1, subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dns: Some(8.8.8.8), secondary_dns: Some(8.8.8.8) }
I (30319) hello_esp: AP Info: ap1 - IpInfo { ip: 192.168.71.1, subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dns: Some(8.8.8.8), secondary_dns: Some(8.8.8.8) }
I (31319) hello_esp: AP Info: ap1 - IpInfo { ip: 192.168.71.1, subnet: Subnet { gateway: 192.168.71.1, mask: Mask(24) }, dns: Some(8.8.8.8), secondary_dns: Some(8.8.8.8) }
```

Vollbrecht, 2023-11-11:
hmm, so in the first case you used the mixed mode i think, have to look up the default. Could you try again here? Thanks for testing. Maybe it was a spurious error cause we really need such a packet in the time-frame you are opening the driver and wait for something.

kekonn, 2023-11-11:
From my testing so far it doesn't matter if I use `EspWifi` or `WifiDriver`. It does crash if I do not configure the wifi at all.

If I set it in mixed mode, it works fine. I think it's a driver initialisation error. You just have to set any configuration or it crashes.

Vollbrecht, 2023-11-11:
Ok ty for your testing. Did it always crash with the same error?

kekonn, 2023-11-11:
Always with `wifi:NAN WiFi stop`, yes.