esp-rs/esp-idf-svc #317: wifi connection is complete, the ping fails

MrDustZero, 2023-11-14:
code
`

    let mut esp_wifi = EspWifi::new(modem, sysloop.clone(), None)?;

    let mut wifi = BlockingWifi::wrap(&mut esp_wifi, sysloop)?;

    wifi.set_configuration(&Configuration::Client(ClientConfiguration::default()))?;

    info!("Starting wifi...");

    wifi.start()?;

    info!("Scanning...");

    let ap_infos = wifi.scan()?;

    let ours = ap_infos.into_iter().find(|a| a.ssid == ssid);

    let channel = if let Some(ours) = ours {
        info!(
            "Found configured access point {} on channel {}",
            ssid, ours.channel
        );
        Some(ours.channel)
    } else {
        info!(
            "Configured access point {} not found during scanning, will go with unknown channel",
            ssid
        );
        None
    };

    wifi.set_configuration(&Configuration::Mixed(
        ClientConfiguration {
            ssid: ssid.into(),
            password: pass.into(),
            channel,
            auth_method,
            ..Default::default()
        }
        ,
        AccessPointConfiguration {
            ssid: "aptest".into(),
            channel: channel.unwrap_or(1),
            ..Default::default()
        },
    ))?;

    info!("Connecting wifi...");

    wifi.connect()?;

    info!("Waiting for DHCP lease...");

    wifi.wait_netif_up()?;

    let ip_info = wifi.wifi().sta_netif().get_ip_info()?;

    info!("Wifi DHCP info: {:?}", ip_info);

    log::info!("About to do some pings for {:?}", std::net::Ipv4Addr::new(114,114,114,114));

    let ping_summary = esp_idf_svc::ping::EspPing::default().ping(std::net::Ipv4Addr::new(114,114,114,114), &Default::default())?;
    if ping_summary.transmitted != ping_summary.received {
        bail!("Pinging IP {} resulted in timeouts", std::net::Ipv4Addr::new(114,114,114,114));
    }

    log::info!("Pinging done");
`
[... 193 lines cut ...]
> I (15673) esp_idf_svc::ping: Ping end callback invoked
> I (15673) esp_idf_svc::ping: 5 packets transmitted, 0 received, time 4983ms
> I (15683) esp_idf_svc::ping: Ping session stopped
> I (15693) esp_idf_svc::ping: Ping session 0x40829d00 removed
> 0x40829d00 - _heap_start
>     at ??:??
> I (15693) wifi:state: run -> init (0)
> I (15703) wifi:ifidx:0, rssi:-41, nf:-93, phytype(0x3, CBW20-SGI), phymode(0x5, 11ax), max_rate:860, he:1
> I (15703) wifi:max ampdu length exponent:3(65535 bytes), mmss:0(no restriction)
> I (15713) wifi:pm stop, total sleep time: 0 us / 6172595 us
> 
> I (15723) wifi:<ba-del>idx:0, tid:6
> I (15723) wifi:new:<11,0>, old:<11,2>, ap:<11,2>, sta:<11,0>, prof:11
> I (15893) wifi:flush txq
> I (15893) wifi:stop sw txq
> I (15893) wifi:lmac stop hw txq
> I (15893) esp_idf_svc::wifi: EspWifi dropped
> I (15893) esp_idf_svc::netif: Dropped
> I (15893) esp_idf_svc::netif: Dropped
> I (15903) wifi:Deinit lldesc rx mblock:10
> I (15913) esp_idf_svc::eventloop: System event loop dropped
> Error: Pinging IP 114.114.114.114 resulted in timeouts
> I (15913) main_task: Returned from app_main()

 hi，esp-32-c6 After the wifi connection is complete, the ping fails,i need help  thanks!

ivmarkov, 2023-11-14:
Why are you assuming that `114.114.114.114` is pingable in principle?

closed 2023-11-14 by ivmarkov

MrDustZero, 2023-11-14:
@ivmarkov sorry, but I can't ping anything else, including the gateway，And tried multiple wifi
，Do I need to configure anything

MrDustZero, 2023-11-14:
I (16616) esp_idf_svc::ping: Ping timeout callback invoked
I (16616) esp_idf_svc::ping: From ??? icmp_seq=1 timeout       I think there's something wrong with that

ivmarkov, 2023-11-14:
Let me ask real simple: **can** you ping `114.114.114.114` **from your own PC**? From the terminal? Forget about the ESP. Ping it from your PC first!

MrDustZero, 2023-11-15:
ESP
> I (11583) esp_netif_handlers: sta ip: 192.168.3.71, mask: 255.255.255.0, gw: 192.168.3.1
> I (11583) wifi: Wifi DHCP info: IpInfo { ip: 192.168.3.71, subnet: Subnet { gateway: 192.168.3.1, mask: Mask(24) }, dns: Some(192.168.3.1), secondary_dns: Some(0.0.0.0) }
> I (11593) wifi: About to do some pings for 192.168.3.1
> I (11603) esp_idf_svc::ping: About to run a summary ping 192.168.3.1 with configuration Configuration { count: 5, interval: 1s, timeout: 1s, data_size: 56, tos: 0 }
> I (11613) esp_idf_svc::ping: Ping session established, got handle 0x408285dc
> 0x408285dc - _heap_start
>     at ??:??
> I (11623) esp_idf_svc::ping: Ping session started
> I (11623) esp_idf_svc::ping: Waiting for the ping session to complete
> I (12623) esp_idf_svc::ping: Ping timeout callback invoked
> I (12623) esp_idf_svc::ping: From ??? icmp_seq=1 timeout
> I (13623) esp_idf_svc::ping: Ping timeout callback invoked
> I (13623) esp_idf_svc::ping: From ??? icmp_seq=2 timeout

PC

> Pinging 192.168.3.1 with 32 bytes of data:
> Reply from 192.168.3.1: bytes=32 time=1ms TTL=64
> Reply from 192.168.3.1: bytes=32 time=1ms TTL=64
> Reply from 192.168.3.1: bytes=32 time=4ms TTL=64
> Reply from 192.168.3.1: bytes=32 time=3ms TTL=64
> 
> Ping statistics for 192.168.3.1:
>     Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
> Approximate round trip times in milli-seconds:
>     Minimum = 1ms, Maximum = 4ms, Average = 2ms

@ivmarkov 
see this ，wifi connection complete，ping gateway

I really don't know what the problem is
(ToT)

MrDustZero, 2023-11-16:
idf and rust all uninstall are installed resolve

slonkazoid, 2025-07-20:
for future strugglers: make sure you dont drop `EspWifi`