esp-rs/esp-idf-svc #320: How to get systime

zzn26, 2023-11-20:
How to check the system time and set the system time?If you have time, please help me with my questions.

ivmarkov, 2023-11-20:
For getting the system time, you either:
* Use the "standard" Rust [method](https://doc.rust-lang.org/std/time/struct.SystemTime.html)
* Or use the `EspSystemTime` structure from `esp-idf-svc`

I don't think there is a Rust API to set the system time, but you can use the unsafe APIs in `esp-idf-sys` / `libc`: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html

With that said - for questions - please join the Matrix chat: https://matrix.to/#/#esp-rs:matrix.org
It is not OK to open issues and ask questions there.

closed 2023-11-20 by ivmarkov

zzn26, 2023-11-23:
@ivmarkov Thank you very much for answering my questions.