esp-rs/esp-idf-svc #321: Can we build esp-idf-svc on Window ?

TuEmb, 2023-11-20:
Hi everyone,

I'm new with `esp-idf-svc`. I'm trying to use this to my ESP32C6 device. However, I'm running into an issue when building example code with command:

`cargo run --example wifi`

I'm concerned that if it support building on Window. Could you guys help me with this ?

[image]

ivmarkov, 2023-11-20:
Do what it says. Clone `esp-idf-svc` as `c:\esp`. Then build and run it as usual.

closed 2023-11-20 by ivmarkov

thetek42, 2023-11-23:
@TuEmb I came across the same issue when using Windows at work. I solved it by enabling long paths (see [this](https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry)) and then executing `set ESP_IDF_PATH_ISSUES=warn` in a terminal.

TuEmb, 2023-11-23:
@thetek42 Thanks so much. I worked for me. I can step over this issue 
Which the target are u using ? Currently, I'm using ESP32C6 and run command:

`cargo run --example wifi --target riscv32imac-unknown-none-elf`

Still having another issue. Could you help me with it ?
[image]

Thanks so much

thetek42, 2023-11-23:
Unfortunately I don't have an ESP32-C6 around, so I won't be able to help you that well. This is what I executed in order to get it running (*on a regular ESP32* - note the different target):

```
set ESP_IDF_PATH_ISSUES=warn
set ESP_IDF_TOOLS_INSTALL_DIR=global
set RUSTFLAGS=--cfg espidf_time64
set ESP_IDF_VERSION=v5.1.1
cargo +esp build --target xtensa-esp32-espidf -Zbuild-std=std,panic_abort -Zbuild-std-features=panic_immediate_abort --example wifi
espflash flash target\xtensa-esp32-espidf\debug\examples\wifi
```

Probably a bunch of these options are not even necessary, I just copied them from somewhere else. My guess is that the `-Zbuild-std` flags might fix the issue.

ivmarkov, 2023-11-23:
> @thetek42 Thanks so much. It worked for me. I can step over this issue Which the target are u using ? Currently, I'm using ESP32C6 and run command:
> 
> `cargo run --example wifi --target riscv32imac-unknown-none-elf`
> 
> Still having another issue. Could you help me with it ? [image]
> 
> Thanks so much

The correct target is `riscv32imac-esp-espidf`. Why are you not using the `esp-idf-template` project to generate all configuration and a sample binary crate?

TuEmb, 2023-11-24:
@thetek42 @ivmarkov Thanks for your support. I used `esp-idf-template` and run successful on my ESP32C6