esp-rs/esp-idf-svc #326: How to prevent intermixed logs

torkleyy, 2023-12-06:
Sometimes, logging from different tasks will get mixed up:

[image]

How can we prevent this? Is there a place where we could add a mutex?

torkleyy, 2023-12-06:
This happens specifically when configuring UART0 as stdin:

https://github.com/esp-rs/esp-idf-hal/issues/200

[image]

thetek42, 2023-12-08:
The intermixed logs are caused by `writeln!` (as used [here](https://github.com/esp-rs/esp-idf-svc/blob/master/src/log.rs#L192)) calling to `write()` multiple times when formatting is used. The following example illustrates this:

```rust
writeln!(
    EspStdout,
    "\x1b[0;{}m{} ({}) {}: {}\x1b[0m",
    color,
    Self::get_marker(record.metadata().level()),
    unsafe { esp_log_timestamp() },
    record.metadata().target(),
    record.args()
)
```

Will effectively execute this (pseudocode):

```
write("\x1b[0;");
write(color);
write("m");
write(Self::get_marker(record.metadata().level());
write(" ");
...
write(record.args);
write("\x1b[0m");
```

Stdout is locked for every single one of these individual `write()` calls, but it is not locked for the entire `writeln!`.

ivmarkov, 2023-12-08:
Problem is not easy to solve as we are - these days - doing cheap (as in no-alloc) logging which moreover is not truncating logs. To do this, we *have to* rely on `writeln!` as it streams the data to stdout rather than accumulating everything in a large buffer (a "string") and then outputting the accumulated string in a single call to the native ESP stdout C-level stream.

Putting a "mutex" "around" `writeln!` in Rust is only a partial solution, as the native C code won't know about the mutex, and you'll still be getting interleaved log output.

Also, and as you can see [here](https://github.com/espressif/esp-idf/blob/master/components/vfs/vfs_uart.c#L205), the rabbit hole is deeper than even that, because even if we were able to get our hands on the native C-level stdout mutex from newlib, we would still be getting interleaved logs if something is writing to `stdout`, and then another thing is hitting `stderr`, as long as one or the other thing does not "log" with a single call to e.g. C "fwrite" which resolves to "uart_write".

So yeah, I'm open to suggestions, but please think carefully for pros and cons, as I'm not seeing a perfect solution.

ivmarkov, 2023-12-08:
> Also, and as you can see [here](https://github.com/espressif/esp-idf/blob/master/components/vfs/vfs_uart.c#L205), the rabbit hole is deeper than even that, because even if we were able to get our [...]

I'm thinking that actually the problem of interleaved logs (or I should rather say - outputs) - where one is coming from, stdout, and another - from stderr - is probably not solved in ESP IDF either, as ESP IDF logging is using the `vprintf` function, which would only lock the `stdout` file handle, but won't lock or even realize, that the `stderr` handle outputs on the same UART.

So... forgetting about the above ^^^... and looking at the C/POSIX docus, maybe we just need to call `flockfile`/`funlockfile` before/after `writeln!`?

thetek42, 2023-12-11:
It looks like this solved the issue. Thanks!

closed 2023-12-11 by torkleyy