esp-rs/esp-idf-svc #328: http/server.rs: Response blocking when returning HandlerError

navaati, 2023-12-14:
When returning an error in the closure given to `EspHttpServer.fn_handler`, the logs will say `I (1494008) esp_idf_svc::http::server: About to handle internal error [<the error>], response not sent yet`, the client will get:
```
HTTP/1.1 500 Internal Error
Content-Type: text/html
Transfer-Encoding: chunked
```
but then no body will be returned and the client will stall and will have to be SIGTERM’d. Reading the source in server.rs, in particular the `EspHttpConnection::render_error` method, I see that the text of the method is supposed to be returned to the client, but apparently that doesn’t happen.

My version of esp-idf-svc is 0.47.2.

ivmarkov, 2023-12-21:
It is very difficult to assess what is going on without pasting a short code snippet. If you do that, I can look into it.

navaati, 2023-12-29:
Hi.

Yes, sorry, I needed to quickly create this issue so that I didn’t forget about it. Now I had time to extract an example:

```rust
use std::{thread::sleep, time::Duration};

use anyhow::{Result, Context};
use esp_idf_hal::{
    prelude::Peripherals,
    modem::Modem,
};
use esp_idf_svc::{
    nvs::EspDefaultNvsPartition,
    wifi::EspWifi,
    eventloop::EspSystemEventLoop,
};
use embedded_svc::{
    wifi::{ClientConfiguration, Configuration},
    http::Method,
};

fn main() -> Result<()> {
    // It is necessary to call this function once. Otherwise some patches to the runtime
    // implemented by esp-idf-sys might not link properly. See https://github.com/esp-rs/esp-idf-template/issues/71
    esp_idf_svc::sys::link_patches();

    // Bind the log crate to the ESP Logging facilities
    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take()?;

    let _wifi = init_wifi(peripherals.modem).context("Failed to connect to wifi")?;

    use esp_idf_svc::http::server::{Configuration as HttpServerConfig, EspHttpServer};
    let mut httpserver = EspHttpServer::new(&HttpServerConfig::default())?;
    httpserver.fn_handler("/", Method::Get, |request| {
        log::warn!("Called HTTP endpoint");

        let _x : u8 = "not an int".parse()?;

        let mut response = request.into_ok_response()?;
        response.write(b"OK")?;
        Ok(())
    })?;

    loop { sleep(Duration::from_secs(1)) }
}

fn init_wifi<'d>(modem: Modem) -> Result<EspWifi<'d>> {
    let mut wifi_driver = EspWifi::new(
        modem, EspSystemEventLoop::take()?,
        Some(EspDefaultNvsPartition::take()?),
    )?;

    wifi_driver.set_configuration(&Configuration::Client(ClientConfiguration{
        ssid: "YOUR_WIFI_SSID".into(),
        password: "YOUR_WIFI_PASSWORD".into(),
        ..Default::default()
    }))?;
    wifi_driver.start()?;
    wifi_driver.connect()?;

    while !wifi_driver.sta_netif().is_up()? {
        log::info!("Waiting for IP");
        sleep(Duration::from_secs(1));
    }
    log::info!("Connected to Wifi, IP: {}", wifi_driver.sta_netif().get_ip_info()?.ip);

    Ok(wifi_driver)
}
```

Libraries versions are:
```
esp-idf-svc = "0.47.3"
embedded-svc = "0.26.4"
esp-idf-hal = "0.42.5"
anyhow = "1.0.77"
```

Please tell me if you need more, like a full repo with Cargo.lock.

closed 2024-05-22 by ivmarkov

navaati, 2024-06-23:
Hi. Thank you very much for fixing !

I can confirm I get proper behaviour with version `esp-idf-svc = "0.49.0"`:

```console
$ http http://192.168.1.230/
HTTP/1.1 500 Internal Error
Content-Type: text/html
Transfer-Encoding: chunked

                    <!DOCTYPE html5>
                    <html>
                        <body style="font-family: Verdana, Sans;">
                            <h1>INTERNAL ERROR</h1>
                            <hr>
                            <pre>invalid digit found in string</pre>
                        <body>
                    </html>
```

Success :) !