esp-rs/esp-idf-svc #329: Some functions are missing in the binding file generated by esp-idf-sys

silent-rain, 2023-12-15:
Esp32s3 does not have many methods in the binding file 'bindings. rs' generated by' esp-idf-sys', but it does exist in the API doc. Why is this?

For example, esp_flash_get_size function， How can I configure to generate bindings for these functions?

Vollbrecht, 2023-12-15:
First of this is a esp-idf-sys issue not esp-idf-svc specific.

To understand why you can't see the method you need to look how the complete chain is generated.
Step by step
  1. understand where the functions come from - e.g in what C header files they are living. In your example they come from [esp_flash.h](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_flash/index.html?highlight=esp_flash_get_size#id1)
   2. Check if they are exported in the include site of esp-idf-sys [here](https://github.com/esp-rs/esp-idf-sys/blob/master/src/include/esp-idf/bindings.h) 
   
  3. In your case we currently dont expose the esp_flash.h directly only with the conditional
 ```C
 #ifdef ESP_IDF_COMP_SPI_FLASH_ENABLED
#include "esp_spi_flash.h"
#include "esp_partition.h"
#endif
```
 4. As of [recomendation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_flash/index.html?highlight=esp_flash_get_size#overview) of the esp-idf guide its best to use esp_partition.h for most of the access.

ivmarkov, 2023-12-15:
> Esp32s3 does not have many methods in the binding file 'bindings. rs' generated by' esp-idf-sys', but it does exist in the API doc. Why is this?
> 
> For example, esp_flash_get_size function， How can I configure to generate bindings for these functions?

(I've checked for esp32, not for esp32s3, but I don't think there is any difference between esp32 and esp32s3 w.r.t. flash functions.)

The TL;DR is:
* I see `esp_flash_get_size` in the generated bindings
* The component @Vollbrecht talks about from above is actually **enabled by default**, as long as you use the regular `cargo` build. Now, if you use the CMake-based build, you need to enable that component explicitly, but otherwise you should be good to go?

So... are you actually **sure** that you don't see that function?

Feel free to reopen if there's an additional detail I'm missing.

closed 2023-12-15 by ivmarkov

silent-rain, 2023-12-15:
> (I've checked for esp32, not for esp32s3, but I don't think there is any difference between esp32 and esp32s3 w.r.t. flash functions.)
>
> The TL;DR is:
>
> * I see `esp_flash_get_size` in the generated [...]

This function was not generated in esp32s3

Vollbrecht, 2023-12-15:
you guys first need to tell what specific function you are talking about, because there is a chance that there are multiple function with the same name here around. Also take note about this info regarding esp-idf 5+:

```
ROM SPI Flash

In versions before v5.0, ROM SPI flash functions were included via esp32**/rom/spi_flash.h. Thus, code written to support different ESP chips might be filled with ROM headers of different targets. Furthermore, not all of the APIs could be used on all ESP chips.

Now, the common APIs are extracted to esp_rom_spiflash.h. Although it is not a breaking change, you are strongly recommended to only use the functions from this header (i.e., prefixed with esp_rom_spiflash and included by esp_rom_spiflash.h) for better cross-compatibility between ESP chips.

To make ROM SPI flash APIs clearer, the following functions are also renamed:

    esp_rom_spiflash_lock() to esp_rom_spiflash_set_bp()
    esp_rom_spiflash_unlock() to esp_rom_spiflash_clear_bp()
```

ivmarkov, 2023-12-15:
> This function was not generated in esp32s3

OK, I'll try to check for esp32s3 on my machine shortly (with both ESP IDF 4.X and 5.X).

What ESP IDF specifically are you using?
Can you also confirm, that you are using the cargo build, and not the cmake build?

ivmarkov, 2023-12-15:
OK, so these functions are missing in the bindings from ESP IDF 5 onward.
However I don't think this is esp32s3 specific at all.

It is more about the fact that since ESP IDF 5 there is a new component, `esp_partition`, and the new `esp_partition.h` header **no longer** includes the `esp_flash` header (which is now in a separate component).

Anyway, I've pushed https://github.com/esp-rs/esp-idf-sys/commit/2785e382176da6382cdb5b4bb94aad8d6fdaf0da which should fix this. If the CI passes, please use `esp-idf-sys` from GitHub, until I branch a new release on crates.io.