esp-rs/esp-idf-svc #332: Cannot change network interface after creation

torkleyy, 2023-12-20:
Many methods of `EspNetif` are not exposed, such as setters and dhcp control. Without this, the ip configuration of an established connection cannot be changed.

ivmarkov, 2023-12-20:
If you have concrete methods that need to be exposed (as well as justification why just  re-creating the interface is not ideal), please go ahead and justify.

But in its current shape, this issue is hardly actionable.

torkleyy, 2023-12-21:
> as well as justification why just re-creating the interface is not ideal

You cannot just recreate the interface. You would at least have to rotate the key, because it has to be unique. And swapping the interface does not allow setting an IP for an existing connection:

```rs
    log::info!("Starting...");
    wifi.start()?;
    log::info!("Connecting...");
    wifi.connect()?;
    log::info!("Connected");
    wifi.wait_netif_up()?;
    log::info!("Ready");

    log::info!("Changing ip");

    wifi.wifi_mut().swap_netif_sta(EspNetif::new_with_conf(&NetifConfiguration {
        key: "test".into(),
        ip_configuration: ipv4::Configuration::Client(ipv4::ClientConfiguration::Fixed(ipv4::ClientSettings {
            ip: ipv4::Ipv4Addr::new(192, 168, 178, 244),
            subnet: ipv4::Subnet {
                gateway: ipv4::Ipv4Addr::new(192, 168, 178, 1),
                mask: ipv4::Mask(24),
            },
            dns: Some(ipv4::Ipv4Addr::new(1, 1, 1, 1)),
            secondary_dns: None,
        })),
        ..NetifConfiguration::wifi_default_client()
    }).unwrap()).unwrap();

    log::info!("Changed ip");
    wifi.wait_netif_up()?;

    log::info!("Netif up again");
```

Above code will simply timeout.

ivmarkov, 2023-12-21:
If you believe [this][(https://github.com/esp-rs/esp-idf-svc/blob/master/src/netif.rs#L234) is possible to do at runtime without destroying the `EspNetif` object, by all means demonstrate how, and I'll merge.

As for the the above code timing out, what is the root cause? You can't just say "you need to do B. because there is some bug with A, which I don't know the reason for".

torkleyy, 2023-12-21:
> As for the the above code timing out, what is the root cause? You can't just say "you need to do B. because there is some bug with A, which I don't know the reason for".

You're correct. I assumed you're not supposed to swap network interfaces during an active connection, but maybe it's also a bug. I have not investigated it further.

> If you believe [this](https://github.com/esp-rs/esp-idf-svc/blob/master/src/netif.rs#L234) is possible to do at runtime without destroying the EspNetif object, by all means demonstrate how, and I'll merge.

I can do that. The [esp-idf docs for `esp_netif_set_ip_info`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html#_CPPv421esp_netif_set_ip_infoP11esp_netif_tPK19esp_netif_ip_info_t) lead me to believe that it's possible.