esp-rs/esp-idf-svc #333: panic when reading wifi scan results with unkown? auth mode

empirephoenix, 2023-12-21:
When running the following snipped:

```
        for i in 1..11 {
            println!("Scanning channel {}", i);
            self.wifi_driver.start_scan(&ScanConfig{
                scan_type : ScanType::Passive(Duration::from_secs(1)),
                show_hidden: false,
                channel: Some(i),
                ..Default::default()
            }, true)?;
            let sr = self.wifi_driver.get_scan_result()?;
            for r in sr.iter(){
                println!("Found wifi {}", r.ssid);
            }
        }
```
I get the following output and panic: 

```
Scanning channel 1
I (1336) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.71.1
Found wifi FRITZ!Box 7590 WO
Scanning channel 2
Found wifi IB-JMD
Scanning channel 3
Scanning channel 4
Found wifi TP-Link_4A34
Scanning channel 5
Found wifi HUAWEI_H122_F5C8
Scanning channel 6
thread 'main' panicked at /home/empire/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-idf-svc-0.47.3/src/wifi.rs:156:18:
explicit panic
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

This is caused by this panic:
```
impl From<Newtype<wifi_auth_mode_t>> for AuthMethod {
    #[allow(non_upper_case_globals)]
    fn from(mode: Newtype<wifi_auth_mode_t>) -> Self {
        match mode.0 {
            [...] a bunch of other auth modes shortened
            wifi_auth_mode_t_WIFI_AUTH_WAPI_PSK => AuthMethod::WAPIPersonal,
            _ => panic!(),
        }
    }
}
```

A) Should this missing auth method be addded here?
B) If possible I would like to see a AuthMethod::Invalid or similar for these cases, as crashing a running application during a scan is something I would not expect to happen, after all crashing random devices with a wifi killer device is not something that I want.

I'm not really proficient with Rust, but I can try to do a pull request, especially for the B) case.

I have checked all ap's my laptop can find via "iwlist wlan0 scan" on channel 6, especially Cell 10 seems to be noteworthy for me due to it returning:  Authentication Suites (1) : unknown (18) which kinda fits.
                        

```
wlan0     Scan completed :
          Cell 02 - Address: 80:2A:A8:14:E1:AF
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
                    Quality=69/70  Signal level=-41 dBm  
                    Encryption key:on
                    ESSID:"C3MA"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
[... 254 lines cut ...]
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=00000003be7282bd
                    Extra: Last beacon: 6507ms ago
                    IE: Unknown: 000D566F6461666F6E652D31353144
                    IE: Unknown: 01088C129824B048606C
                    IE: Unknown: 030106
                    IE: Unknown: 050400020000
                    IE: Unknown: 0706455520010D14
                    IE: Unknown: 2A0100
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 3B025100
                    IE: Unknown: 2D1A2D0017FFFFFF00000000000000C2010100000000001004870100
                    IE: Unknown: 3D1606001700000000000000000000000000000000000000
                    IE: Unknown: 7F080500080200000040
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
                    IE: Unknown: DD220050F204104A00011010440001021057000101103C0001031049000600372A000120

```

If there is anything additionally I can submit to help, or I should try to add a unknown enum type

ivmarkov, 2023-12-21:
I think we should do the following:
* Wrap [this](https://github.com/esp-rs/embedded-svc/blob/master/src/wifi.rs#L140) in an `Option`
* Fix the Wifi code to return `None`. As in, change `impl From<Newtype<wifi_auth_mode_t>> for AuthMethod {` to 
`impl From<Newtype<wifi_auth_mode_t>> for Option<AuthMethod> {` and then where it currently says `panic`, return `None`
* Extend the `AuthMethod` enum with additional authentication options, and then map them in the `From` impl. Here's a list of additional authentications recognized by the ESP IDF in the meantime: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html

If you are willing to work on a PR, I can support with feedback/suggestions.

ivmarkov, 2023-12-21:
... and thank you for the analysis!

empirephoenix, 2023-12-21:
Is there a good reason against adding a AuthMethod.Unknown instead and use this instead of the panic? 
I kinda find it strange to get a Some(None) in case of a open wifi.
Also it would not cause less of an change for already existing code

Vollbrecht, 2023-12-21:
One thing to notice that between esp-idf v4.4.6 and v5.1.2 there are difference in the upstream defined types. Look the diff from [here](https://github.com/espressif/esp-idf/blob/v4.4.6/components/esp_wifi/include/esp_wifi_types.h#L60) and [here](https://github.com/espressif/esp-idf/blob/v5.1.2/components/esp_wifi/include/esp_wifi_types.h#L60)

ivmarkov, 2023-12-21:
> Is there a good reason against adding a AuthMethod.Unknown instead and use this instead of the panic? I kinda find it strange to get a Some(None) in case of a open wifi.

If the wifi is open, you'll get `Some(AuthMethod::None), not Option::None.

And then ask yourself the question what shall we do when you put `AuthMethod::Unknown` in a `Configuration` object? Hence why I prefer Option<AuthMethod> in the ap info struct.

empirephoenix, 2023-12-21:
Ah ok that makes sense, impossible to represent invalid invariant then

I think I want to do this in in multiple steps, first getting it to not crash and the new auth methods then after that.
https://github.com/esp-rs/esp-idf-svc/pull/335/files

Any ideas how to get rid of the now new unwraps? Or make the other From return Option<X> as well, as it cannot be converted in case of a unknown auth method?

empirephoenix, 2024-01-01:
Ok, I think I have a acceptable version now, sadly I did not manage to understand where the wifi_auth_mode_t is mapped so I did not manage to add the ~3 additional auth methods mentioned in the documentation

ivmarkov, 2024-01-25:
The suggested PR was merged, so let's close it.

closed 2024-01-25 by ivmarkov