esp-rs/esp-idf-svc #342: embassy timer impl is very inaccurate

t-moe, 2024-01-12:
Starting with a fresh (template-generated) project, and adding  
`embassy-time = { version = "0.1.5", features = ["generic-queue"] }` to my dependencies,
and with the following code:

```
use esp_idf_svc::hal::task::block_on;
use embassy_time::{Duration, Timer};

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    block_on(async {
        for _ in 0..10 {
            log::info!("tick");
            Timer::after(Duration::from_secs(1)).await;
        }
    });
}

```

I get the output:

```
I (571) esp_idf_embassy_time: tick
I (2151) esp_idf_embassy_time: tick
I (5311) esp_idf_embassy_time: tick
I (11621) esp_idf_embassy_time: tick
I (24251) esp_idf_embassy_time: tick
I (49501) esp_idf_embassy_time: tick
I (100011) esp_idf_embassy_time: tick
```

which is very inaccurate.

ivmarkov, 2024-01-12:
Should be fine now.

closed 2024-01-12 by ivmarkov

bugadani, 2024-01-13:
@ivmarkov I hate to bother you with this, but next time please put at least one character into the commit message that isn't the issue number you're fixing. It's pretty annoying to find what has changed :D

ivmarkov, 2024-01-13:
I will, but in the meantime... :P

[image: For_Daniel.png]