esp-rs/esp-idf-svc #343: How to use the mdns module?

abonshan, 2024-01-12:
I can't get the mdns module to work with the documentation or the stuff I found in some other issues. As in I can't import EspMdns and use it. I'm guessing that I have to enable it as an "remote-component" or "extra-component" but I found no list for which components are which or an explanation on how to enable them. I would be grateful if someone could kindly clear it up for me and others facing the same questions. Thanks in advance

Vollbrecht, 2024-01-13:
Since esp-idf version 5 espressif started to move things that were previously directly available into there separate components registry. So in your case you are seeing that the mdns module in our code is behind the following feature gate if you look [here](https://github.com/esp-rs/esp-idf-svc/blob/master/src/lib.rs#L65) .  If you see such a feature gate where you see *comp_espressif__<component-name> with a double underscore ( __ ) in between you can use the esp-idf-sys component feature to add it as a remote component. This will automatically make it available.

For more information how you can use this esp-idf-sys feature you can have a look [here](https://github.com/esp-rs/esp-idf-sys/blob/master/BUILD-OPTIONS.md#remote-components-idf-component-registry) 

To make it short you can add it via your Cargo.toml by adding
```toml
[[package.metadata.esp-idf-sys.extra_components]]
remote_component = { name = "espressif/mdns", version = "1.2" }
```

abonshan, 2024-01-13:
Thank you, that did the trick.

closed 2024-01-13 by abonshan