esp-rs/esp-idf-svc #344: Can't build esp-idf-svc

OrganomagnesiumHalide, 2024-01-14:
I generated a project using `cargo generate esp-rs/esp-idf-template` and ran `cargo build --release`. I received the following error:

```
feature_caps: unsafe { g_wifi_feature_caps },
    |                                    ^^^^^^^^^^^^^^^^^^^ not found in this scope
````

It's in `esp-idf-svc-0.47.3/src/wifi.rs:498:36` and has been documented a [few times](https://www.reddit.com/r/rust/comments/18q6qlv/cannot_compile_esp32_program/).

OrganomagnesiumHalide, 2024-01-14:
I tried but then I got 

```
error: failed to select a version for `embedded-svc`.
    ... required by package `esp-idf-svc v0.47.3 (https://github.com/esp-rs/esp-idf-svc#0cc687be)`
versions that meet the requirements `^0.26.4` (locked to 0.26.4) are: 0.26.4
the package `esp-idf-svc` depends on `embedded-svc`, with features: `asyncify` but `embedded-svc` does not have these features.
```

ivmarkov, 2024-01-16:
https://github.com/esp-rs/esp-idf-svc/issues/345#issuecomment-1893266254

closed 2024-01-16 by ivmarkov