esp-rs/esp-idf-svc #350: How can I open an async TCP connection (std)

coder0xff, 2024-01-21:
I'm aware of crates like [async-std] and [edge-net], but I'm trying to figure out what the best path forward for layer-4 async network connections is. My understanding of the workings of async on esp32 is limited still, and so far I've been using `esp_idf_svc::hal::task::block_on`. Is async-std fully compatible with esp32? Is edge-net a good choice for std-enabled projects? Is there an example of using a TCP connection with esp-idf-svc + std? Thanks.

Vollbrecht, 2024-01-22:
> Update: it looks like async-std is not at all compatible, as it relies on a Posix environment. 

esp-idf is posix compatible in many places. For example the pthread support is done through it, also other filesystem related stuff, systemtime and also networking. That support is how we got a simple std rust support in the first place. Networking stuff is done through lwip and it uses BSD-Sockets. You can find out more about it [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html?highlight=posix#lwip)

ivmarkov, 2024-01-22:
> I'm aware of crates like [async-std](https://crates.io/crates/async-std) and [edge-net](https://crates.io/crates/edge-net), but I'm trying to figure out what the best path forward for layer-4 async [...]

No. `async-std` is not compatible with ESP IDF and I fear it is also stagnating.

Your best bet is the `smol` crates or if you really insist - `tokio`.

> Is edge-net a good choice for std-enabled projects? 

If you use the GitHub version - yes. It has a lot of improvements (in fact, almost a complete redesign) compared to the last version published in crates.io.

> Is there an example of using a TCP connection with esp-idf-svc + std? Thanks.

I guess you meant in an async way... because sync std is how it works out of the box without any external crates - because as @Vollbrecht mentioned - ESP IDF is posix-compatible. Anyway:
https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/src/main.rs#L725

closed 2024-01-22 by ivmarkov