esp-rs/esp-idf-svc #351: Unable to build newly created esp-idf project

M4urici0GM, 2024-01-21:
Hi, i'm just starting out with esp, and i'm having a bad time building the project.
Whenever i try to build the project i get this error:

```
             feature_caps: unsafe { g_wifi_feature_caps },
    |                                    ^^^^^^^^^^^^^^^^^^^ not found in this scope
```
I've also tried patching my binary based on [this](https://github.com/esp-rs/esp-idf-svc/issues/345#issuecomment-1893266254) comment
But doing that gives me another error:
```
error: failed to select a version for `esp-idf-hal`.
    ... required by package `esp-idf-svc v0.47.3`
    ... which satisfies dependency `esp-idf-svc = "^0.47.3"` (locked to 0.47.3) of package `test2 v0.1.0 (/Users/mourice/repos/esp/test2)`
versions that meet the requirements `^0.42.5` (locked to 0.42.5) are: 0.42.5

the package `esp-idf-svc` depends on `esp-idf-hal`, with features: `esp-idf-sys` but `esp-idf-hal` does not have these features.
 It has a required dependency with that name, but only optional dependencies can be used as features.

failed to select a version for `esp-idf-hal` which could resolve this conflict
```

I've followed the guide to setup environment, and using the non-std version of the template, i was able to build it and run-it
But i'm not sure how to fix this one.

Vollbrecht, 2024-01-21:
duplicate of https://github.com/esp-rs/esp-idf-svc/issues/344 its already fixed in the current git master branch. Till we release a patched version please use the github version with patch.crates-io in your cargo.toml. you need that on both esp-idf-svc and embedded-svc

M4urici0GM, 2024-01-22:
i've already tried patching the library using `patch.crates-io`:
```
[patch.crates-io]
embedded-svc = { git = "https://github.com/esp-rs/embedded-svc" }
esp-idf-hal = { git = "https://github.com/esp-rs/esp-idf-hal" }
esp-idf-sys = { git = "https://github.com/esp-rs/esp-idf-sys" }
```

Now i get
```
error: failed to select a version for `esp-idf-hal`.
    ... required by package `esp-idf-svc v0.47.3`
    ... which satisfies dependency `esp-idf-svc = "^0.47.3"` (locked to 0.47.3) of package `test2 v0.1.0 (/Users/mourice/repos/esp/test2)`
versions that meet the requirements `^0.42.5` (locked to 0.42.5) are: 0.42.5

the package `esp-idf-svc` depends on `esp-idf-hal`, with features: `esp-idf-sys` but `esp-idf-hal` does not have these features.
 It has a required dependency with that name, but only optional dependencies can be used as features.

failed to select a version for `esp-idf-hal` which could resolve this conflict
```

Vollbrecht, 2024-01-22:
That's because you are not patching esp-idf-svc. All versions must align to current githead to be usable. One more question. What esp-idf version are you using / e.g what ESP_IDF_VERSION did you set in your .cargo/config.toml ? In most cases its best to only use taged version aka "v5.1.2" or "v4.4.6" and not the release "release/X" versions since they are not stable

M4urici0GM, 2024-01-22:
Oh, nice, i was able to build.
On the .cargo/config its the 5.1.1
Really appreciate the help :shipit:

closed 2024-01-22 by M4urici0GM

ryankopf, 2024-03-15:
Experiencing the same issue here, changing esp-idf-svc  = "=0.47.3" to esp-idf-svc  = "=0.48.1" worked, but it required changing a few other things:
Added heapless     = "=0.8.0" to cargo
In wifi module:
let mut ssid_heapless = heapless::String::<32>::new();
ssid_heapless.push_str(ssid).expect("SSID exceeds heapless String capacity");
In main.rs function:
esp_idf_svc::http::Method::Get