esp-rs/esp-idf-svc #357: implementation of `esp_idf_svc::eventloop::EspEventDeserializer` is not general enough compiler error

cs-clarence, 2024-02-01:
### [Minimal repro here](https://github.com/cs-clarence/async-wifi-tokio-repro)

Caling async blocks calling async functions from `AsyncWifi` is compile error when passed to `tokio::spawn`.

The compile error:
```
error: implementation of `esp_idf_svc::eventloop::EspEventDeserializer` is not general enough
  --> src/main.rs:20:5
   |
20 | /     tokio::task::spawn(async move {
21 | |         wifi.connect().await.expect("Failed to start wifi");
22 | |     });
   | |______^ implementation of `esp_idf_svc::eventloop::EspEventDeserializer` is not general enough
   |
   = note: `esp_idf_svc::eventloop::EspEventDeserializer` would have to be implemented for the type `esp_idf_svc::wifi::WifiEvent<'0>`, for any lifetime `'0`...
   = note: ...but `esp_idf_svc::eventloop::EspEventDeserializer` is actually implemented for the type `esp_idf_svc::wifi::WifiEvent<'1>`, for some specific lifetime `'1`
```
I also tried `start` and it's the same error.

Calling `scan` causes a different error:
```
error: higher-ranked lifetime error
  --> src/main.rs:29:5
   |
29 | /     tokio::task::spawn(async move {
30 | |         wifi.scan().await.expect("Failed to scan");
31 | |     });
   | |______^
   |
   = note: could not prove `{async block@src/main.rs:29:24: 31:6}: std::marker::Send`
```

It's not just limited to `tokio::spawn`, any function that expects a `Send` future also causes the error.

```rs
/// Just to check if it's not just the [`tokio::spawn`] that's causing the compile error
fn send_future<F: Future<Output = ()> + Send + 'static>(
    f: F,
) -> impl Future<Output = ()> + Send + 'static {
    f
}
send_future(async move {
    wifi.start().await.expect("Failed to start wifi");
});

```
Same error:
```
error: implementation of `esp_idf_svc::eventloop::EspEventDeserializer` is not general enough
  --> src/main.rs:28:5
   |
28 | /     send_future(async move {
29 | |         wifi.start().await.expect("Failed to start wifi");
30 | |     });
   | |______^ implementation of `esp_idf_svc::eventloop::EspEventDeserializer` is not general enough
   |
   = note: `esp_idf_svc::eventloop::EspEventDeserializer` would have to be implemented for the type `esp_idf_svc::wifi::WifiEvent<'0>`, for any lifetime `'0`...
   = note: ...but `esp_idf_svc::eventloop::EspEventDeserializer` is actually implemented for the type `esp_idf_svc::wifi::WifiEvent<'1>`, for some specific lifetime `'1`
```

ivmarkov, 2024-02-01:
... or did it work before, with the prevous release of `esp-idf-svc`?

cs-clarence, 2024-02-01:
@ivmarkov I was using the master branch before which had [this commit](https://github.com/esp-rs/esp-idf-svc/commit/3ebaf6337dc00d8a4fa9a4f02640508c22a0faac). It didn't have that problem then I updated to 0.48.0 which broke my build.

ivmarkov, 2024-02-01:
You are right, and I stand correct. `AsyncWifi` **was** `Send` before, and something did regress. I think it should be relatively easy to fix.

ivmarkov, 2024-02-01:
@cs-clarence **Wild guess**, but I think it is because of [this signature](https://github.com/esp-rs/esp-idf-svc/blob/master/src/eventloop.rs#L440)

The new event loop subscriptions do not copy/clone data. They return a reference of the data which is inside the queue, but the data is no longer `Send`. I think we *might* be able to recover some `Send`-ness, but let me think...

ivmarkov, 2024-02-01:
[Done!](https://github.com/esp-rs/esp-idf-svc/commit/2996516330a26a41ede09c58424a7478d7d646ca)

Can you temporarily switch to the GIT branch? We'll be releasing this fix and a few others shortly, in an incremental release...

Please reopen / ping me here if it still does not work, but it really should, as there is now a new [example](https://github.com/esp-rs/esp-idf-svc/blob/master/examples/wifi_async_send.rs) tracking that `AsyncWifi` is `Send`... :-)

closed 2024-02-01 by ivmarkov

cs-clarence, 2024-02-01:
@ivmarkov it compiles now, thanks