esp-rs/esp-idf-svc #361: macOS `std` Development Environment Setup Errors

apollolabsdev, 2024-02-12:
This is an issue I faced setting up an environment for `std` development on a macOS.

This is on a Macbook Air with an M2 chip and Ventura 13.0.

The steps from the Rust ESP book [here](https://esp-rs.github.io/book/installation/riscv.html) and [here](https://esp-rs.github.io/book/installation/std-requirements.html) were followed. In addition to the linked steps for macOS highlighted on the Espressif website [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#for-macos-users).

I used `cargo-generate` to create a new project then ran `cargo build` without any modifications. The below error resulted. From the look of it, it appeared like the ESP-IDF necessary components were not being found. I verified that the environment variables were set up as required by the instructions and checked the relevant files. However, after much tinkering the code would only build successfully if I setup `ESP_IDF_TOOLS_INSTALL_DIR = "global"` under `[env]` in `config.toml`.

```
error: failed to run custom build command for `esp-idf-sys v0.34.0`

Caused by:
  process didn't exit successfully: `/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/target/debug/build/esp-idf-sys-eed4e9c9a8eb67e0/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=EXTRA-COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENT_MANAGER
  IDF_PYTHON_ENV_PATH=/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/python_env/idf5.1_py3.12_env
  ESP_IDF_VERSION=5.1
  PATH=/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/tools/ninja/1.10.2/:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/python_env/idf5.1_py3.12_env/bin:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/esp-idf/v5.1.2/tools:$PATH
  IDF_DEACTIVATE_FILE_PATH=/var/folders/0d/5g4dzlsj6gs05rq71bc62s_80000gn/T/tmp4qlowc_zidf_47983
  Current system platform: macos-arm64
  Skipping riscv32-esp-elf@esp-12.2.0_20230208 (already installed)
  Skipping cmake@3.24.0 (already installed)
  Skipping ninja@1.10.2 (already installed)
  IDF_PYTHON_ENV_PATH=/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/python_env/idf5.1_py3.12_env
  ESP_IDF_VERSION=5.1
  PATH=/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/tools/ninja/1.10.2/:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/python_env/idf5.1_py3.12_env/bin:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/esp-idf/v5.1.2/tools:$PATH
  IDF_DEACTIVATE_FILE_PATH=/var/folders/0d/5g4dzlsj6gs05rq71bc62s_80000gn/T/tmpvzl22kl1idf_47983
  cargo:rerun-if-changed=/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/sdkconfig.defaults
  CMAKE_PREFIX_PATH_riscv32imc_zicsr_zifencei-esp-espidf = None
  CMAKE_PREFIX_PATH_riscv32imc_zicsr_zifencei_esp_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_riscv32imc_zicsr_zifencei-esp-espidf = None
  CMAKE_riscv32imc_zicsr_zifencei_esp_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: cd "/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d495f87e907f1a6/out/build" && CMAKE_PREFIX_PATH="" EXTRA_COMPONENT_DIRS="" IDF_COMPONENT_MANAGER="1" IDF_PATH="/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/esp-idf/v5.1.2" IDF_TARGET="esp32c3" IDF_TOOLS_PATH="/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif" PATH="/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/python_env/idf5.1_py3.12_env/bin:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/esp-idf/v5.1.2/tools:/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/tools/ninja/1.10.2/:/opt/homebrew/opt/llvm/bin:/Users/omarhiari/bin:/Library/Frameworks/Python.framework/Versions/3.12/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/System/Cryptexes/App/usr/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/TeX/texbin:/opt/homebrew/opt/llvm/bin:/Users/omarhiari/bin:/Library/Frameworks/Python.framework/Versions/3.12/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/Users/omarhiari/.cargo/bin" PROJECT_DIR="/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping" SDKCONFIG_DEFAULTS="/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d495f87e907f1a6/out/gen-sdkconfig.defaults;/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/sdkconfig.defaults" "cmake" "/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d495f87e907f1a6/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/esp-idf/v5.1.2/tools/cmake/toolchain-esp32c3.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/python_env/idf5.1_py3.12_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d495f87e907f1a6/out" "-DCMAKE_C_FLAGS= -march=rv32imc_zicsr_zifencei -ffunction-sections -fdata-sections -march=rv32imc_zicsr_zifencei -mabi=ilp32 -mcmodel=medany" "-DCMAKE_CXX_FLAGS= -march=rv32imc_zicsr_zifencei -ffunction-sections -fdata-sections -march=rv32imc_zicsr_zifencei -mabi=ilp32 -mcmodel=medany" "-DCMAKE_ASM_FLAGS=  -ffunction-sections -fdata-sections -march=rv32imc_zicsr_zifencei -mabi=ilp32 -mcmodel=medany"
  -- Building ESP-IDF components for target esp32c3
  -- Checking Python dependencies...
  Python requirements are satisfied.
  Constraint file: /Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/espidf.constraints.v5.1.txt
  Requirement files:
   - /Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/esp-idf/v5.1.2/tools/requirements/requirements.core.txt
  Python being checked: /Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/python_env/idf5.1_py3.12_env/bin/python
  -- Configuring incomplete, errors occurred!
  See also "/Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/target/riscv32imc-esp-espidf/debug/build/esp-idf-sys-6d495f87e907f1a6/out/build/CMakeFiles/CMakeOutput.log".

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32c3",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1.2",
[... 50 lines cut ...]
    CMake Error at
    /Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/esp-idf/v5.1.2/components/mqtt/CMakeLists.txt:3
    (message):

      Missing esp-mqtt submodule.  Please run 'git submodule update --init
      --recursive' in ESP-IDF directory to fix this.

    Call Stack (most recent call first):

[... cut ...]
    /Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/esp-idf/v5.1.2/tools/cmake/build.cmake:574 (__component_get_requirements)
    /Users/omarhiari/Documents/GitHub/ESP32C3/std_examples/ping/.embuild/espressif/esp-idf/v5.1.2/tools/cmake/project.cmake:547 (idf_build_process)
    CMakeLists.txt:28 (project)

  thread 'main' panicked at /Users/omarhiari/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.50/src/lib.rs:1098:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Vollbrecht, 2024-02-12:
You should not need to set ESP_IDF_TOOLS_INSTALL_DIR . The first thing you should try is to `rm -rf .embuild` inside your project dir as a failed pull of esp-idf from github could lead to such a thing you were observing. The same logic apply if you use `ESP_IDF_TOOLS_INSTALL_DIR` and point it to a now project independent global install dir. If you run `cargo b` it will install esp-idf for you automatically, eithere in the local project dir inside .embuild or inside a global dir if you are using that env var. The `ESP_IDF_VERSION` env var is needed and sets what version of esp-idf is gonna be installed/used by your project.

apollolabsdev, 2024-02-12:
Yes, I forgot to mention that though, I did delete `.embuild` and run `cargo clean` several times before running `cargo build` again. All ending up with the same result. `ESP_IDF_VERSION` is already set under `[env]` from the original `cargo generate` project but it won't work until `ESP_IDF_TOOLS_INSTALL_DIR` is set.

apollolabsdev, 2024-02-21:
Has anything changed recently on the `cargo generate` template? This issue has appeared all over again regardless of the `ESP_IDF_TOOLS_INSTALL_DIR` option I used. I delete `.embuild` and run a `cargo clean` before each build everytime I do a change.

jukrieger, 2024-04-02:
@apollolabsdev Currently, you need to set the following option in your `.cargo/config` file, inside the `[env]` block.

`CRATE_CC_NO_DEFAULTS="1"`

apollolabsdev, 2024-04-03:
Awesome, it works! 
I was getting by using the 4.4 version of the ESP-IDF as I didn't face the same issue there.