esp-rs/esp-idf-svc #362: mqtt connect failed cause “Guru Meditation Error: Core  0 panic'ed (Instruction access fault). Exception was unhandled.”

cvbni, 2024-02-15:
`I (3638) stdc6: Wifi connected
I (3648) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4628) esp_netif_handlers: sta ip: 192.168.1.2, mask: 255.255.255.0, gw: 192.168.1.1
I (4628) stdc6: Wifi created & netif up
I (4628) stdc6: Starting UART1 loopback
I (4628) uart: queue free spaces: 10
I (4638) stdc6: MQTT client created
I (4638) stdc6: About to start the MQTT client
I (4648) stdc6: MQTT Listening for messages
I (4648) stdc6: [Queue] Event: BeforeConnect
I (4658) stdc6: [Queue] Event: Others , count 1
E (5018) esp-tls: [sock=54] connect() error: Host is unreachable
E (5018) transport_base: Failed to open a new connection: 32772
E (5018) mqtt_client: Error transport connect
I (5028) stdc6: [Queue] Event: Error(EspError(-1))
I (5028) stdc6: [Queue] Event: Others , count 2
I (5038) stdc6: [Queue] Event: Disconnected
I (5038) stdc6: [Queue] Event: Others , count 3
E (5048) mqtt_client: Client has not connected
E (5048) mqtt_client: Client has not connected
Guru Meditation Error: Core  0 panic'ed (Instruction access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x00000000  RA      : 0x00000000  SP      : 0x408325f0  GP      : 0x408142a0
0x00000000 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
0x00000000 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
0x408325f0 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0x408142a0 - sha3_families
    at ??:??
TP      : 0x407f2494  T0      : 0x40022494  T1      : 0xffffffe0  T2      : 0x00000000
0x407f2494 - ??
    at ??:??
0x40022494 - ??
    at ??:??
0xffffffe0 - s_rtc_timer_retain_mem
    at ??:??
0x00000000 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
S0/FP   : 0x00000000  S1      : 0x00000000  A0      : 0x00000001  A1      : 0x4082c1fc
0x00000000 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
0x00000000 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
0x00000001 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
0x4082c1fc - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
A2      : 0x4082c1fc  A3      : 0x00000001  A4      : 0x4081bdc4  A5      : 0x20001000
0x4082c1fc - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0x00000001 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
0x4081bdc4 - uxCriticalNesting
    at ??:??
0x20001000 - ??
    at ??:??
A6      : 0x00000001  A7      : 0x0000000a  S2      : 0x00000000  S3      : 0x00000000
0x00000001 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
0x0000000a - alloc::task::raw_waker
    at ??:??
0x00000000 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
0x00000000 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
0x00000000 - alloc::sync::Arc<T,A>::as_ptr
[... 579 lines cut ...]
0x408329d8 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0x0000000c - alloc::task::<impl core::convert::From<alloc::sync::Arc<W>> for core::task::wake::Waker>::from
    at ??:??

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
0x4001975a - ??
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
0x4086c410 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
load:0x4086e610,len:0x2d68
0x4086e610 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
load:0x40875720,len:0x1800
0x40875720 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??`

ivmarkov, 2024-02-15:
Please post somewhere an as small as possible project that reproduces the issue, thanks.

cvbni, 2024-02-15:
the cargo.toml
`[package]
name = "stdc6"
version = "0.1.0"
authors = ["cvbnix"]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native", "binstart"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly", "esp-idf-hal/nightly"]
experimental = ["esp-idf-svc/experimental", "embedded-svc/experimental"]
embassy = [
    "esp-idf-svc/embassy-sync",
    "esp-idf-svc/critical-section",
    "esp-idf-svc/embassy-time-driver",
]

# Propagated esp-idf-hal features
critical-section = ["esp-idf-hal/critical-section"]
wake-from-isr = ["esp-idf-hal/wake-from-isr"]
embassy-sync = ["esp-idf-hal/embassy-sync"]

# The next are propagated from esp-idf-sys via esp-idf-hal
native = ["esp-idf-hal/native"]
alloc_handler = ["esp-idf-hal/alloc_handler"]
panic_handler = ["esp-idf-hal/panic_handler"]
binstart = ["esp-idf-hal/binstart"]
libstart = ["esp-idf-hal/libstart"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.48", default-features = false }
heapless = { version = "0.8", default-features = false }
num_enum = { version = "0.7", default-features = false }
enumset = { version = "1", default-features = false }
uncased = { version = "0.9.7", default-features = false }
embedded-hal-async = { version = "1", default-features = false }
embedded-svc = { version = "0.27", default-features = false }
esp-idf-hal = { version = "0.43", default-features = false }
embassy-time-driver = { version = "0.1", optional = true, features = [
    "tick-hz-1_000_000",
] }
embassy-futures = "0.1.1"
[build-dependencies]
embuild = "0.31.3"

[dev-dependencies]
anyhow = "1"
esp-idf-sys = { version = "0.34", features = ["binstart"] }
futures = "0.3"
serde = { version = "1", default-features = false, features = ["derive"] }
serde_json = { version = "1", default-features = false, features = ["alloc"] }
postcard = "1"

async-io = "2"

`
[... 155 lines cut ...]
    .await;
    match res {
        Either3::First(res) => res,
        Either3::Second(res) => res,
        Either3::Third(res) => res,
    }
}

fn mqtt_create(
    url: &str,
    client_id: &str,
    username: &str,
    password: &str,
) -> Result<(EspAsyncMqttClient, EspAsyncMqttConnection), EspError> {
    let (mqtt_client, mqtt_conn) = EspAsyncMqttClient::new(
        url,
        &MqttClientConfiguration {
            client_id: Some(client_id),
            username: Some(username),
            password: Some(password),
            ..Default::default()
        },
    )?;
    Ok((mqtt_client, mqtt_conn))
}`

ivmarkov, 2024-02-15:
Can you try to reduce it a bit further by removing the UART code?

cvbni, 2024-02-15:
> Can you try to reduce it a bit further by removing the UART code?
all the same
```
I (6224) stdc6: MQTT Listening for messages
I (6224) stdc6: [Queue] Event: BeforeConnect
I (6234) stdc6: [Queue] Event: Others , count 1
E (7524) esp-tls: [sock=54] connect() error: Host is unreachable
E (7524) transport_base: Failed to open a new connection: 32772
E (7524) mqtt_client: Error transport connect
I (7534) stdc6: [Queue] Event: Error(EspError(-1))
I (7534) stdc6: [Queue] Event: Others , count 2
I (7544) stdc6: [Queue] Event: Disconnected
I (7544) stdc6: [Queue] Event: Others , count 3
E (7554) mqtt_client: Client has not connected
E (7564) mqtt_client: Client has not connected
Guru Meditation Error: Core  0 panic'ed (Instruction access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x00000000  RA      : 0x00000000  SP      : 0x408307e0  GP      : 0x40813580
0x00000000 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
0x00000000 - alloc::sync::Arc<T,A>::as_ptr
    at ??:??
0x408307e0 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0x40813580 - coex_schm_ble_default_bt_a2dp_wifi_connecting
    at ??:??
TP      : 0x407f167c  T0      : 0x40022494  T1      : 0xffffffe0  T2      : 0x00000000
```
Dispose all of the Uart code:

```
use core::pin::pin;
use core::time::Duration;
use embassy_futures::select::{select3, Either3}; //select,Either,
use esp_idf_svc::eventloop::EspSystemEventLoop;
use esp_idf_svc::hal::gpio;
use esp_idf_svc::hal::peripherals::Peripherals; //+uart
use esp_idf_svc::hal::prelude::*;
use esp_idf_svc::hal::uart::*;
use esp_idf_svc::mqtt::client::*;
use esp_idf_svc::nvs::EspDefaultNvsPartition;
use esp_idf_svc::sys::EspError;
use esp_idf_svc::timer::{EspAsyncTimer, EspTimerService}; //EspTaskTimerService,
use esp_idf_svc::wifi::*;
use log::*;
const READ_BUF_SIZE: usize = 256;
const SSID: &str = "CMCC-Hnpu";
const PASSWORD: &str = "1002002200";
const MQTT_URL: &str = "mqtt://cvbni.dynv6.net:60432";
const MQTT_CLIENT_ID: &str = "WiFi-mqtt-demo";
const TOPIC_U: &str = "jwbu/";
const TOPIC_D: &str = "jwbd/";
const MQTT_USERNAME: &str = "*******";
const MQTT_PASSWORD: &str = "*************";
const UART_BAUDRATE: u32 =115200;

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();
    let sys_loop = EspSystemEventLoop::take().unwrap();
    let timer_service = EspTimerService::new().unwrap();
    let nvs = EspDefaultNvsPartition::take().unwrap();
    let peripherals = Peripherals::take().unwrap(); //mod

    esp_idf_svc::hal::task::block_on(async {
        //---------------------wifi---------------
        //let _wifi = wifi_create(&sys_loop, &timer_service, &nvs).await?;
        let mut esp_wifi = EspWifi::new(peripherals.modem, sys_loop.clone(), Some(nvs.clone()))?;
        let mut wifi = AsyncWifi::wrap(&mut esp_wifi, sys_loop.clone(), timer_service.clone())?;
[... 120 lines cut ...]
        Either3::Second(res) => res,
        Either3::Third(res) => res,
    }
}

fn mqtt_create(
    url: &str,
    client_id: &str,
    username: &str,
    password: &str,
) -> Result<(EspAsyncMqttClient, EspAsyncMqttConnection), EspError> {
    let (mqtt_client, mqtt_conn) = EspAsyncMqttClient::new(
        url,
        &MqttClientConfiguration {
            client_id: Some(client_id),
            username: Some(username),
            password: Some(password),
            ..Default::default()
        },
    )?;
    Ok((mqtt_client, mqtt_conn))
}
```
const MQTT_URL: &str = "mqtt://cvbni.dynv6.net:60432";
when I replace above  that is ipv6 URL to an available ipv4 URL,then this program will just work ok

cvbni, 2024-02-15:
Also with this app, when I turn off the WiFi AP and then run the ESP32C6 board, it will also keep restarting again and again

Vollbrecht, 2024-02-20:
The panic itself might be duo to the missing drop impl. Can you retry with the current git master branch and see if your problem still exists in its current form?

cvbni, 2024-02-20:
all is ok,thanks
no any "Guru Meditation Error"

```
I (6515) wifi:pm stop, total sleep time: 927942 us / 2936460 us

I (6525) wifi:<ba-del>idx:0, tid:0
I (6525) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
I (6575) wifi:flush txq
I (6575) wifi:stop sw txq
I (6575) wifi:lmac stop hw txq
I (6575) esp_idf_svc::wifi: EspWifi dropped
I (6575) esp_idf_svc::netif: Dropped
I (6575) esp_idf_svc::netif: Dropped
I (6585) wifi:Deinit lldesc rx mblock:10
I (6595) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (6595) esp_idf_svc::nvs: EspNvs dropped
I (6595) esp_idf_svc::nvs: NvsDefault dropped
I (6605) esp_idf_svc::eventloop: System event loop dropped
thread 'main' panicked at src\main.rs:48:6:
called `Result::unwrap()` on an `Err` value: EspError(-1)
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

abort() was called at PC 0x4202fff9 on core 0
0x4202fff9 - panic_abort::__rust_start_panic::abort
    at C:\Users\cvbnix\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\panic_abort\src\lib.rs:43
Core  0 register dump:
MEPC    : 0x40801634  RA      : 0x40806c5c  SP      : 0x4081ffa0  GP      : 0x408142a0
0x40801634 - panic_abort
    at D:\e\stdc6\target\riscv32imac-esp-espidf\debug\build\esp-idf-sys-43185ca0e978a7fd\out\build\D:/Dev/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system\panic.c:452
0x40806c5c - __ubsan_include
    at D:\e\stdc6\target\riscv32imac-esp-espidf\debug\build\esp-idf-sys-43185ca0e978a7fd\out\build\D:/Dev/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system\ubsan.c:313
0x4081ffa0 - _ZN12_GLOBAL__N_110eh_globalsE
    at ??:??
0x408142a0 - sha3_families
    at ??:??
TP      : 0x407dfbb4  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
```
the err is here,the ".unwrap()" get it
```

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();
    esp_idf_svc::hal::task::block_on(async {
        run(
            &format!("{}{}", TOPIC_U, MQTT_CLIENT_ID),
            &format!("{}{}", TOPIC_D, MQTT_CLIENT_ID),
            MQTT_URL,
            MQTT_CLIENT_ID,
            MQTT_USERNAME,
            MQTT_PASSWORD,
        )
        .await
    })
    .unwrap();
}
```
cargo.toml
```
#esp-idf-svc = { version = "0.48", default-features = false }
esp-idf-svc = { git  = "https://github.com/esp-rs/esp-idf-svc", default-features = false }
```

closed 2024-02-20 by cvbni

casimcdaniels, 2024-03-06:
What exactly did you do to fix your issue @cvbni ? I'm getting the same issue, even with your code and updating to master. Just trying to run `examples/mqtt_sync` and I hit a similar stack trace:

```txt
I (39) pp: pp rom version: 9387209
I (39) net80211: net80211 rom version: 9387209
I (59) wifi:wifi driver task: 3fc9e97c, prio:23, stack:6656, core=0
I (59) system_api: Base MAC address is not set
I (69) system_api: read default base MAC address from EFUSE
I (79) wifi:wifi firmware version: 1fd20f4
I (89) wifi:wifi certification version: v7.0
I (89) wifi:config NVS flash: enabled
I (99) wifi:config nano formating: disabled
I (109) wifi:Init data frame dynamic rx buffer num: 32
I (119) wifi:Init static rx mgmt buffer num: 10
I (129) wifi:Init management short buffer num: 32
I (129) wifi:Init dynamic tx buffer num: 32
I (139) wifi:Init static tx FG buffer num: 2
I (149) wifi:Init static rx buffer size: 1600
I (159) wifi:Init static rx buffer num: 10
I (169) wifi:Init dynamic rx buffer num: 32
I (179) wifi_init: rx ba win: 6
I (179) wifi_init: tcpip mbox: 32
I (189) wifi_init: udp mbox: 6
I (199) wifi_init: tcp mbox: 6
I (199) wifi_init: tcp tx win: 5760
I (209) wifi_init: tcp rx win: 5760
I (219) wifi_init: tcp mss: 1440
I (229) wifi_init: WiFi IRAM OP enabled
I (239) wifi_init: WiFi RX IRAM OP enabled
I (279) phy_init: phy_version 1150,7c3c08f,Jan 24 2024,17:32:21
W (289) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (349) wifi:mode : sta (24:0a:c4:00:01:10)
I (349) wifi:enable tsf
I (359) pls_work: Wifi started
I (2779) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2779) wifi:state: init -> auth (b0)
I (2789) wifi:state: auth -> assoc (0)
I (2789) wifi:state: assoc -> run (10)
I (2789) wifi:connected with Wokwi-GUEST, aid = 1, channel 6, BW20, bssid = 42:13:37:55:aa:01
I (2809) wifi:security: Open Auth, phy: bg, rssi: 16
I (2839) wifi:pm start, type: 1

I (2839) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (2839) pls_work: Wifi connected
I (2879) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (3839) esp_netif_handlers: sta ip: 10.13.37.2, mask: 255.255.255.0, gw: 10.13.37.1
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40380214  RA      : 0x4038020a  SP      : 0x3fc9b990  GP      : 0x3fc8fe00  
TP      : 0x3fc6657c  T0      : 0x3fc956fc  T1      : 0x40383476  T2      : 0x00000000  
S0/FP   : 0x3fc9af98  S1      : 0x80000003  A0      : 0x3fc9b990  A1      : 0x3fc927f8  
A2      : 0x00000000  A3      : 0x00000004  A4      : 0x3fc956ec  A5      : 0x600c2000  
A6      : 0xa0000000  A7      : 0x0000000a  S2      : 0x00000080  S3      : 0x00000001  
S4      : 0x00000000  S5      : 0x3fc9afa8  S6      : 0x00000002  S7      : 0x00000000  
S8      : 0x3fca7a7c  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000001  T6      : 0x00000000  
MSTATUS : 0x00000000  MTVEC   : 0x40380001  MCAUSE  : 0x00000002  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fc9b990: 0x40388360 0x4038603a 0x3fc9ba10 0x3fc8fe00 0x3fc6657c 0x4005890e 0x4200e340 0xffffffff
3fc9b9b0: 0x3fc9af98 0x00000000 0x00000001 0x00000001 0x00000000 0x00000004 0x3fc956ec 0x600c2000
3fc9b9d0: 0xa0000000 0x0000000a 0x00000000 0x00000000 0x00000000 0x3fc9afa8 0x00000002 0x00000000
3fc9b9f0: 0x3fca7a7c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ba10: 0x00000025 0x00000002 0x3c0fd1c8 0x00000000 0x00000000 0x3fca7898 0x3fc9ba60 0x4200ac7e
3fc9ba30: 0xffffffff 0x00000000 0xffffffff 0x00000000 0x3fc9baf4 0x00000001 0x3fc9af98 0x403861a0
3fc9ba50: 0x00000000 0x00000000 0x3fc9ad24 0x420d94c2 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ba70: 0x00000101 0x3c0fd1c8 0x00000000 0x3fca7898 0x00000000 0x00000000 0x00000000 0x00000000
3fc9ba90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9ad24 0x420d94d4
3fc9bab0: 0x00000000 0x00000000 0x00000000 0x403882f8 0x00000000 0x00000000 0x00000000 0x00000000
3fc9bad0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9baf0: 0x00000154 0x3fc9b990 0x3fc96134 0x3fc93078 0x3fc93078 0x3fc9baf4 0x3fc93070 0x00000005
3fc9bb10: 0x3fc9ad6c 0x3fc9ad6c 0x3fc9baf4 0x00000000 0x00000014 0x3fc9aff0 0x5f737973 0x00747665
3fc9bb30: 0x00000000 0x00000000 0x00000000 0x3fc9bae0 0x00000014 0x00000000 0x3fca7544 0x420540e6
3fc9bb50: 0x00000000 0x3fc96ae8 0x3fc96b50 0x3fc96bb8 0x00000000 0x00000000 0x00000001 0x00000000
3fc9bb70: 0x00000000 0x00000000 0x420cc238 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9bb90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9bbb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9bbd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9bbf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9bc10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9bc30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3f000000 0x0000000c 0x00000004
3fc9bc50: 0x00000001 0x3fc9bc44 0x0000000c 0x3fc9bc6c 0x00000000 0x3fc9bc54 0x00000054 0x00000000
[... cut ...]
3fc9bcd0: 0x00000000 0x00000000 0x3fc9bcdc 0xffffffff 0x3fc9bcdc 0x3fc9bcdc 0x00000000 0x3fc9bcf0
3fc9bcf0: 0xffffffff 0x3fc9bcf0 0x3fc9bcf0 0x00000001 0x00000001 0x00000000 0x0000ffff 0x00000000
3fc9bd10: 0xb33fffff 0x00000000 0x00000028 0x3c0fbb84 0x3fc9c848 0x3fc9cf5c 0x00000002 0x3fc9becc
3fc9bd30: 0x3fc9becc 0x00000001 0x3fc9bda0 0x3fc9bda0 0x00000001 0x00000054 0x00000000 0x3fc9bd48
3fc9bd50: 0x00000000 0x00000000 0x00000000 0x3fc9bd60 0xffffffff 0x3fc9bd60 0x3fc9bd60 0x00000000
3fc9bd70: 0x3fc9bd74 0xffffffff 0x3fc9bd74 0x3fc9bd74 0x00000001 0x00000001 0x00000000 0x0000ffff

```

cvbni, 2024-03-06:
@casimcdaniels  You should open a new issue,
your device is caused by wifi,but my device is caused by mqtt

> I (2839) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
> I (2839) pls_work: Wifi connected
> I (2879) wifi:AP's beacon interval = 102400 us, DTIM period = 2
> I [...]