esp-rs/esp-idf-svc #364: Exception when using MQTT and disconnecting WiFi

crespum, 2024-02-15:
Hello,

I've modified the `mqtt_client.rs` example as follows to test if the MQTT client is being properly dropped after WiFi disconnection:

```rust
//! MQTT blocking client example which subscribes to an internet MQTT server and then sends
//! and receives events in its own topic.

use core::time::Duration;

use esp_idf_svc::eventloop::EspSystemEventLoop;
use esp_idf_svc::hal::peripherals::Peripherals;
use esp_idf_svc::mqtt::client::*;
use esp_idf_svc::nvs::EspDefaultNvsPartition;
use esp_idf_svc::sys::EspError;
use esp_idf_svc::wifi::*;

use log::*;

const SSID: &str = env!("WIFI_SSID");
const PASSWORD: &str = env!("WIFI_PASS");

const MQTT_URL: &str = "mqtt://broker.emqx.io:1883";
const MQTT_CLIENT_ID: &str = "esp-mqtt-demo";
const MQTT_TOPIC: &str = "esp-mqtt-demo";

fn main() {
    esp_idf_svc::sys::link_patches();
    esp_idf_svc::log::EspLogger::initialize_default();

    let sys_loop = EspSystemEventLoop::take().unwrap();
    let nvs = EspDefaultNvsPartition::take().unwrap();

    let mut _wifi = wifi_create(&sys_loop, &nvs).unwrap();

    {
        let mut client = mqtt_create(MQTT_URL, MQTT_CLIENT_ID).unwrap();
        run(&mut client, MQTT_TOPIC).unwrap();
    }

    std::thread::sleep(Duration::from_millis(500));
    _wifi.disconnect().unwrap();

    loop {
        std::thread::sleep(Duration::from_millis(100));
    }
}

fn run(client: &mut EspMqttClient<'_>, topic: &str) -> Result<(), EspError> {
    let payload = "Hello from esp-mqtt-demo!";

    client
        .enqueue(topic, QoS::AtMostOnce, false, payload.as_bytes())
        .unwrap();
    Ok(())
}

fn mqtt_create(url: &str, client_id: &str) -> Result<EspMqttClient<'static>, EspError> {
    let mqtt_client = EspMqttClient::new_cb(
        url,
        &MqttClientConfiguration {
            client_id: Some(client_id),
            ..Default::default()
        },
        move |message_event| match message_event.payload() {
            EventPayload::Error(e) => error!("MQTT error {:?}", e),
            e => warn!("MQTT event {:?}", e),
        },
    )?;

    Ok(mqtt_client)
}

fn wifi_create(
    sys_loop: &EspSystemEventLoop,
    nvs: &EspDefaultNvsPartition,
) -> Result<EspWifi<'static>, EspError> {
    let peripherals = Peripherals::take()?;

    let mut esp_wifi = EspWifi::new(peripherals.modem, sys_loop.clone(), Some(nvs.clone()))?;
    let mut wifi = BlockingWifi::wrap(&mut esp_wifi, sys_loop.clone())?;

    wifi.set_configuration(&Configuration::Client(ClientConfiguration {
        ssid: SSID.try_into().unwrap(),
        password: PASSWORD.try_into().unwrap(),
        ..Default::default()
    }))?;

    wifi.start()?;
    info!("Wifi started");

    wifi.connect()?;
    info!("Wifi connected");

    wifi.wait_netif_up()?;
    info!("Wifi netif up");

    Ok(esp_wifi)
}
```

However, disconnection results in an exception:
```
[...]
I (3346) mqtt_client: Wifi connected
I (3366) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (5576) wifi:<ba-add>idx:0 (ifx:0, e0:63:da:71:5e:4b), tid:0, ssn:1, winSize:64
I (6346) esp_netif_handlers: sta ip: 192.168.1.168, mask: 255.255.255.0, gw: 192.168.1.1
I (6346) mqtt_client: Wifi netif up
W (6356) mqtt_client: MQTT event BeforeConnect
W (6746) mqtt_client: MQTT event Connected(false)
I (7256) wifi:state: run -> init (0)
I (7256) wifi:pm stop, total sleep time: 2877000 us / 3917581 us

I (7256) wifi:<ba-del>idx:0, tid:0
I (7256) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
E (7266) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (7266) mqtt_client: Poll read error: 119, aborting connection
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
```

My best guess is that something is being used (maybe the callback) after being drop?? 

To be honest, I'm not sure if it's a bug or something missing in my code. Any help is welcome.

MabezDev, 2024-02-15:
Might be related to #362?

ivmarkov, 2024-02-15:
It looks like a bug, because whatever happens, the code should not crash (that is, given enough memory).

Note that there is no such thing as "MQTT client should be dropped on Wifi disconnect", as it of course cannot drop itself. It should just continue trying to connect, until eventually the network connectivity is restored.

ivmarkov, 2024-02-15:
Update: Actually, looks just like a forgotten `panic!` / `todo!` somewhere...

ivmarkov, 2024-02-15:
[Probably this one](https://github.com/esp-rs/esp-idf-svc/blob/master/src/mqtt/client.rs#L975).

crespum, 2024-02-19:
I don't see any reference to that line in the [coredump](https://github.com/esp-rs/esp-idf-svc/files/14331850/coredump.txt). Is there a way to extract a _cleaner_ coredump?

BTW, the version we are using now (0.46) doesn't have this behavior.

MabezDev, 2024-02-19:
@crespum are you using panic-immediate-abort? If so, temporarily remove it to actually see panic messages.

crespum, 2024-02-19:
I'm not sure if this is what you mean. I've enabled these two options:

```
CONFIG_ESP_SYSTEM_PANIC=y
CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT=y
```

But the [coredump](https://github.com/esp-rs/esp-idf-svc/files/14332250/coredump_2.txt) is similar.

MabezDev, 2024-02-19:
I'm referring to this https://gitlab.com/scrobotics/optical-makerspace/dark-sky-meter-fw/-/blob/master/.cargo/config.toml?ref_type=heads#L10

Vollbrecht, 2024-02-19:
panic-abort =/= panic_immediate_abort . The later one is a build-std-feature. Though one thing you can try is removing "  "-C",
   "default-linker-libraries"", because if you **don't** use panic_immediate_abort "default-linker-libraries" is not needed.

crespum, 2024-02-19:
I'm running an example from this repo, so none of those options [are enabled](https://github.com/esp-rs/esp-idf-svc/blob/master/.cargo/config.toml).

MabezDev, 2024-02-19:
> panic-abort =/= panic_immediate_abort . The later one is a build-std-feature. Though one thing you can try is removing " "-C", "default-linker-libraries"", because if you **don't** use panic_immediate_abort "default-linker-libraries" is not needed.

I'll defer to your judgement as I'm not too familiar with these features, but https://github.com/rust-lang/rust/blob/master/library/panic_abort/src/lib.rs looks like panic-abort also won't print the message?

crespum, 2024-02-19:
I'm using an ESP32-C3 and rustc 1.77.0-nightly (bf8716f1c 2023-12-24)

crespum, 2024-02-19:
I'm not running the original example but a modified version of it ([check the code in my first comment](https://github.com/esp-rs/esp-idf-svc/issues/364#issue-2136145504)).

Vollbrecht, 2024-02-19:
How did you force the disconnect on the esp? did you restart your wifi router?

Vollbrecht, 2024-02-19:
I found the problem will update with what is happening and providing a patch shortly.

Vollbrecht, 2024-02-19:
Your problem was not the WIFI disconnect in this case.
What happend:
1) creating a instance of mqttClient
2) rust client: ( creating a Boxed callback that is tied to the mqttClient struct)
3) c mqtt hidden thread is spawend and sending stuff via the event system to the callback
4) droping the mqttClient ( freeing the Boxed callback)
5) your thread sleep gives a direct chance to run the hidden thread again before the wifi disconnect
6) hidden thread sends something to the freed callback -> BOOOOOOOOOOOOOOOOOOOOOOM

Vollbrecht, 2024-02-19:
you can test out https://github.com/esp-rs/esp-idf-svc/pull/369 and check if is now working for you

crespum, 2024-02-20:
Brilliant. I've tested it on both the modified example and the original project. No issues!

PR #369 (2024-02-20): Add Drop implementation for EspMqttClient
fixes #364
closed 2024-02-20 by ivmarkov