esp-rs/esp-idf-svc #366: Rust analyzer not fully functional in esp-rs/esp-idf projects?

weiying-chen, 2024-02-16:
I generated an esp-rs/esp-idf project using the official [template](https://github.com/esp-rs/esp-idf-template).

Then I modified it like this:

```
fn main() {
    use esp_idf_svc::{
        hal::prelude::Peripherals,
        sys::link_patches,
    };

    esp_idf_svc::log::EspLogger::initialize_default();

    let peripherals = Peripherals::take()?;
}
```
There were no warnings or errors in my text editor ([Helix[](https://helix-editor.com/)). Is there something in a esp-rs/esp-if that's making rust-analyzer not catch there common issues? The Rust compiler did catch the errors, though:

```
sys::link_patches,
  |         ^^^^^^^^^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

13 |     let peripherals = Peripherals::take()?;
   |                                          ^ cannot use the `?` operator in a function that returns `()`
```

Note: if I go to a non-esp-rs/esp-if project, I do see these warnings and errors in my text editor.

Vollbrecht, 2024-02-16:
rust-analyzer works perfectly fine. Though the problem itself is more often that RA doesn't pick up the correct cargo configs. So it's more a matter of telling RA directly the same configs, that are included inside the `.cargo/config.toml` file, like target, feature flags etc. How you can configure RA is dependent on the integration in your editor/ide. Check how you can configure it on your env and look here for the possible [settings](https://rust-analyzer.github.io/manual.html#configuration). 
You can try out the extaArgs + extraEnv and the target ones. Note that there are settings for .cargo and for .check as they are for different RA invocations you may need to set both configs.

weiying-chen, 2024-02-17:
This is my Helix log when I open a Rust project that has esp-idf-svc as a dependency:

```
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "2024-02-17T11:37:28.907967Z ERROR rust_analyzer::main_loop: FetchBuildDataError:\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "error[E0531]: cannot find unit struct, unit variant or constant `SIGABRT` in crate `libc`\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "   --> /home/weiying-chen/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/test/src/test_result.rs:102:24\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "    |\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "102 |             Some(libc::SIGABRT) => TestResult::TrFailed,\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "    |                        ^^^^^^^ not found in `libc`\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "error: aborting due to 1 previous error\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "For more information about this error, try `rustc --explain E0531`.\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "error: could not compile `test` (lib) due to 2 previous errors\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "\n"
2024-02-17T19:37:28.908 helix_lsp::transport [ERROR] rust-analyzer err <- "\n"
2024-02-17T19:37:28.914 helix_term::application [WARN] unhandled window/showMessage: ShowMessageParams { typ: Warning, message: "Failed to run build scripts of some packages." }
2024-02-17T19:37:29.579 helix_term::application [WARN] unhandled window/showMessage: ShowMessageParams { typ: Warning, message: "Failed to run build scripts of some packages." }
```

I think the issue is Helix, not esp-idf-svc?

weiying-chen, 2024-02-18:
Thanks to this [thread](https://github.com/rust-lang/rust-analyzer/issues/16552), I found out rust-analyzer works perfectly in esp-idf projects if you use rustc 1.77.0-nightly (2a3e63551 2023-12-30). The issue is rustc 1.78.0-nightly (ee9c7c940 2024-02-14).

juliankrieger, 2024-04-12:
@weiying-chen I can't really confirm this. I've been having Problems with RA showing errors on exported crates which require ESP components, for example `eventloop`: 

[image]

My toolchains are as following:

I am on rustc.177.0-nighty, my project however is using the esp toolchain: 
[image]

weiying-chen, 2024-04-12:
@juliankrieger So your point is that RA has issues work 1.77 too?

julienvincent, 2024-04-19:
I have been running into the same issue. I upgraded my nightly to 1.79-0-nightly and now rust-analyzer is broken in all my esp projects.

Vollbrecht, 2024-06-21:
@weiying-chen your issue comes from the fact that rust-analyzer by default try's to build `cargo test` with the rust "libtest" framework. This can be mitigated by adding 
```toml

[[bin]]
name = "{{project-name}}"
harness = false # do not use the built in cargo test harness -> resolve rust-analyzer errors
``` 
to your Cargo.toml. 

rust-analyzer should work fine with that option. I am closing this, feel free to reopen it if that doesn't solve this particular problem or create a new issue. Thanks!

closed 2024-06-21 by Vollbrecht

faulesocke, 2024-07-14:
Even with that, rust analyzer seems to not work correctly in my library crate. Running `cargo clippy --all-targets` now complains:

```
error[E0601]: `main` function not found in crate `esp_os`
   --> src/lib.rs:247:2
    |
247 | }
    |  ^ consider adding a `main` function to `src/lib.rs`
```

I added this to my Cargo.toml:
```toml
[lib]
harness = false
```

But whats more annoying is that I'm currently trying to write an example, so I added
```toml
[[example]]
name = "basic"
harness = false
```

But I still don't see error messages reliably in my `examples/basic.rs` file (maybe due to the bogus main function error?)

Can this be resolved on the esp-rs side or has this to be fixed in rustc itself?